micro-ROS/micro_ros_arduino #684: micro-ros_subscriber_twist example Arduino .ino project fails to compile

igillespie, 2022-01-12:
## Issue template

- Hardware description: Arduino DUE
- Installation type: micro-ros-arduino
- Version or commit hash: ros2-galactic

#### Steps to reproduce the issue
Install micro_ros_arduino-2.0.2-galactic into Arduino 1.8.19 IDE's libraries directory. 
Go to File > Examples > micro_ros_arduino > micro-ros_subscriber_twist
Try to compile the project.

#### Expected behavior
Compilation finishes successfully.

#### Actual behavior
I see this in the log:

Arduino: 1.8.19 (Mac OS X), Board: "Arduino Due (Programming Port)"

Library micro_ros_arduino-2.0.2-galactic has been declared precompiled:
Using precompiled library in /Users/iang/Documents/Arduino/libraries/micro_ros_arduino-2.0.2-galactic/src/cortex-m3
The platform does not support 'compiler.libraries.ldflags' for precompiled libraries.
sketch/micro-ros_subscriber_twist.ino.cpp.o: In function `set_microros_transports':
/Users/iang/Documents/Arduino/libraries/micro_ros_arduino-2.0.2-galactic/src/micro_ros_arduino.h:32: undefined reference to `rmw_uros_set_custom_transport'
sketch/micro-ros_subscriber_twist.ino.cpp.o: In function `setup':
/Users/iang/Documents/Arduino/libraries/micro_ros_arduino-2.0.2-galactic/examples/micro-ros_subscriber_twist/micro-ros_subscriber_twist.ino:45: undefined reference to `rcutils_get_default_allocator'
/Users/iang/Documents/Arduino/libraries/micro_ros_arduino-2.0.2-galactic/examples/micro-ros_subscriber_twist/micro-ros_subscriber_twist.ino:48: undefined reference to `rclc_support_init'
/Users/iang/Documents/Arduino/libraries/micro_ros_arduino-2.0.2-galactic/examples/micro-ros_subscriber_twist/micro-ros_subscriber_twist.ino:51: undefined reference to `rclc_node_init_default'
/Users/iang/Documents/Arduino/libraries/micro_ros_arduino-2.0.2-galactic/examples/micro-ros_subscriber_twist/micro-ros_subscriber_twist.ino:54: undefined reference to `rosidl_typesupport_c__get_message_type_support_handle__geometry_msgs__msg__Twist'
/Users/iang/Documents/Arduino/libraries/micro_ros_arduino-2.0.2-galactic/examples/micro-ros_subscriber_twist/micro-ros_subscriber_twist.ino:54: undefined reference to `rclc_subscription_init_default'
/Users/iang/Documents/Arduino/libraries/micro_ros_arduino-2.0.2-galactic/examples/micro-ros_subscriber_twist/micro-ros_subscriber_twist.ino:61: undefined reference to `rclc_executor_init'
/Users/iang/Documents/Arduino/libraries/micro_ros_arduino-2.0.2-galactic/examples/micro-ros_subscriber_twist/micro-ros_subscriber_twist.ino:62: undefined reference to `rclc_executor_add_subscription'
sketch/micro-ros_subscriber_twist.ino.cpp.o: In function `loop':
/Users/iang/Documents/Arduino/libraries/micro_ros_arduino-2.0.2-galactic/examples/micro-ros_subscriber_twist/micro-ros_subscriber_twist.ino:68: undefined reference to `rclc_executor_spin_some'
collect2: error: ld returned 1 exit status
exit status 1
Error compiling for board Arduino Due (Programming Port).

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

#### Additional information
I have run installed the SAMD platforms.txt file. I have also tried this using Arduino IDE on Linux Ubuntu 20.04 where I see the same error message.

Thanks for any help or insight you can provide!

pablogs9, 2022-01-12:
Can you check which platform.txt are your board using? For sure it is a matter of modifying this file because this patching process is just to handle `ldflags`.

Could you identify which file should be modified so we can update the README.md?

[This link](https://arduino.github.io/arduino-cli/0.20/platform-specification/#platformlocaltxt) can provide some tips.

CC: @lukicdarkoo was the original contributor of the Arduino DUE port, maybe he can provide some help

igillespie, 2022-01-12:
Thanks for the help. Here is what the platforms.txt I'm using looks like. Please let me know if you have any suggestions on what I might try.

# Arduino SAM Core and platform.
# ------------------------------
#
# For more info:
# https://github.com/arduino/Arduino/wiki/Arduino-IDE-1.5-3rd-party-Hardware-specification

name=Arduino ARM (32-bits) Boards
version=1.6.12

# SAM3 compile variables
# ----------------------

compiler.warning_flags=-w
compiler.warning_flags.none=-w
compiler.warning_flags.default=
compiler.warning_flags.more=-Wall
compiler.warning_flags.all=-Wall -Wextra

compiler.path={runtime.tools.arm-none-eabi-gcc-4.8.3-2014q1.path}/bin/
compiler.c.cmd=arm-none-eabi-gcc
compiler.c.flags=-c -g -Os {compiler.warning_flags} -std=gnu11 -ffunction-sections -fdata-sections -nostdlib --param max-inline-insns-single=500 -Dprintf=iprintf -MMD
compiler.c.elf.cmd=arm-none-eabi-gcc
compiler.c.elf.flags=-Os -Wl,--gc-sections
compiler.S.cmd=arm-none-eabi-gcc
compiler.S.flags=-c -g -x assembler-with-cpp -MMD
compiler.cpp.cmd=arm-none-eabi-g++
compiler.cpp.flags=-c -g -Os {compiler.warning_flags} -std=gnu++11 -ffunction-sections -fdata-sections -nostdlib -fno-threadsafe-statics --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -MMD
compiler.ar.cmd=arm-none-eabi-ar
compiler.ar.flags=rcs
compiler.objcopy.cmd=arm-none-eabi-objcopy
compiler.objcopy.eep.flags=-O ihex -j .eeprom --set-section-flags=.eeprom=alloc,load --no-change-warnings --change-section-lma .eeprom=0
compiler.elf2hex.flags=-O binary
compiler.elf2hex.cmd=arm-none-eabi-objcopy
compiler.ldflags=
compiler.size.cmd=arm-none-eabi-size
compiler.define=-DARDUINO=
compiler.combine.flags=-u _sbrk -u link -u _close -u _fstat -u _isatty -u _lseek -u _read -u _write -u _exit -u kill -u _getpid

# This can be overridden in boards.txt
build.extra_flags=

# These can be overridden in platform.local.txt
compiler.c.extra_flags=
compiler.c.elf.extra_flags=
compiler.cpp.extra_flags=
compiler.S.extra_flags=
compiler.ar.extra_flags=
compiler.elf2hex.extra_flags=
compiler.libraries.ldflags=

compiler.libsam.c.flags="-I{build.system.path}/libsam" "-I{build.system.path}/CMSIS/CMSIS/Include/" "-I{build.system.path}/CMSIS/Device/ATMEL/"

# USB Flags
# ---------
build.usb_flags=-DUSB_VID={build.vid} -DUSB_PID={build.pid} -DUSBCON '-DUSB_MANUFACTURER={build.usb_manufacturer}' '-DUSB_PRODUCT={build.usb_product}'

# Default usb manufacturer will be replaced at compile time using
# numeric vendor ID if available or by board's specific value.
build.usb_manufacturer="Unknown"

# SAM3 compile patterns
# ---------------------

## Compile c files
recipe.c.o.pattern="{compiler.path}{compiler.c.cmd}" {compiler.c.flags} -mcpu={build.mcu} -mthumb -DF_CPU={build.f_cpu} -DARDUINO={runtime.ide.version} -DARDUINO_{build.board} -DARDUINO_ARCH_{build.arch} {compiler.c.extra_flags} {build.extra_flags} {compiler.libsam.c.flags} {includes} "{source_file}" -o "{object_file}"

## Compile c++ files
recipe.cpp.o.pattern="{compiler.path}{compiler.cpp.cmd}" {compiler.cpp.flags} -mcpu={build.mcu} -mthumb -DF_CPU={build.f_cpu} -DARDUINO={runtime.ide.version} -DARDUINO_{build.board} -DARDUINO_ARCH_{build.arch} {compiler.cpp.extra_flags} {build.extra_flags} {compiler.libsam.c.flags} {includes} "{source_file}" -o "{object_file}"

## Compile S files
recipe.S.o.pattern="{compiler.path}{compiler.S.cmd}" {compiler.S.flags} -mcpu={build.mcu} -mthumb -DF_CPU={build.f_cpu} -DARDUINO={runtime.ide.version} -DARDUINO_{build.board} -DARDUINO_ARCH_{build.arch} {compiler.S.extra_flags} {build.extra_flags} {compiler.libsam.c.flags} {includes} "{source_file}" -o "{object_file}"

## Create archives
# archive_file_path is needed for backwards compatibility with IDE 1.6.5 or older, IDE 1.6.6 or newer overrides this value
archive_file_path={build.path}/{archive_file}
recipe.ar.pattern="{compiler.path}{compiler.ar.cmd}" {compiler.ar.flags} {compiler.ar.extra_flags} "{archive_file_path}" "{object_file}"

## Combine gc-sections, archives, and objects
recipe.c.combine.pattern="{compiler.path}{compiler.c.elf.cmd}" -mcpu={build.mcu} -mthumb {compiler.c.elf.flags} "-T{build.variant.path}/{build.ldscript}" "-Wl,-Map,{build.path}/{build.project_name}.map" {compiler.c.elf.extra_flags} -o "{build.path}/{build.project_name}.elf" "-L{build.path}" -Wl,--cref -Wl,--check-sections -Wl,--gc-sections -Wl,--entry=Reset_Handler -Wl,--unresolved-symbols=report-all -Wl,--warn-common -Wl,--warn-section-align -Wl,--start-group {compiler.combine.flags} {object_files} {compiler.libraries.ldflags} "{build.variant.path}/{build.variant_system_lib}" "{build.path}/{archive_file}" -Wl,--end-group -lm -lgcc

## Create output (.bin file)
recipe.objcopy.bin.pattern="{compiler.path}{compiler.elf2hex.cmd}" {compiler.elf2hex.flags} {compiler.elf2hex.extra_flags} "{build.path}/{build.project_name}.elf" "{build.path}/{build.project_name}.bin"

## Save hex
recipe.output.tmp_file={build.project_name}.bin
recipe.output.save_file={build.project_name}.{build.variant}.bin

## Compute size
recipe.size.pattern="{compiler.path}{compiler.size.cmd}" -A "{build.path}/{build.project_name}.elf"
recipe.size.regex=\.text\s+([0-9]+).*

# SAM3 Uploader tools
# -------------------

# BOSSA
tools.bossac.path={runtime.tools.bossac.path}
tools.bossac.cmd=bossac
tools.bossac.cmd.windows=bossac.exe

tools.bossac.upload.params.verbose=-i -d
tools.bossac.upload.params.quiet=
tools.bossac.upload.params.verify=-v
tools.bossac.upload.pattern="{path}/{cmd}" {upload.verbose} --port={serial.port.file} -U {upload.native_usb} -e -w {upload.verify} -b "{build.path}/{build.project_name}.bin" -R

tools.bossac_remote.upload.pattern=/usr/bin/run-bossac {upload.verbose} --port=ttyATH0 -U {upload.native_usb} -e -w -v -b /tmp/sketch.bin -R

lukicdarkoo, 2022-01-12:
It looks related to:
https://github.com/arduino/ArduinoCore-sam/pull/115

igillespie, 2022-01-13:
Interesting. It isn't clear to me if there is a solution. Sorry if I'm missing something obvious. Any ideas on what I can do to get it to build?

igillespie, 2022-01-13:
Ok, so I was installing the platforms.txt file in the wrong place on my Mac.

For anyone who has this issue in the future, on my Mac, I had to put the modified file here: 
~/<username>/Library/Arduino15/packages/arduino/hardware/sam (after I installed the Due board via the Arduino Boards Manager).

This made this warning, "The platform does not support 'compiler.libraries.ldflags' for precompiled libraries." go away. However, I still can't compile. I have these warnings like from the original post:

`sketch/micro-ros_subscriber_twist.ino.cpp.o: In function `set_microros_transports':
/Users/iang/Documents/Arduino/libraries/micro_ros_arduino-2.0.2-galactic/src/micro_ros_arduino.h:32: undefined reference to `rmw_uros_set_custom_transport'
sketch/micro-ros_subscriber_twist.ino.cpp.o: In function `setup':
/Users/iang/Documents/Arduino/libraries/micro_ros_arduino-2.0.2-galactic/examples/micro-ros_subscriber_twist/micro-ros_subscriber_twist.ino:45: undefined reference to `rcutils_get_default_allocator'
/Users/iang/Documents/Arduino/libraries/micro_ros_arduino-2.0.2-galactic/examples/micro-ros_subscriber_twist/micro-ros_subscriber_twist.ino:48: undefined reference to `rclc_support_init'
/Users/iang/Documents/Arduino/libraries/micro_ros_arduino-2.0.2-galactic/examples/micro-ros_subscriber_twist/micro-ros_subscriber_twist.ino:51: undefined reference to `rclc_node_init_default'
/Users/iang/Documents/Arduino/libraries/micro_ros_arduino-2.0.2-galactic/examples/micro-ros_subscriber_twist/micro-ros_subscriber_twist.ino:54: undefined reference to `rosidl_typesupport_c__get_message_type_support_handle__geometry_msgs__msg__Twist'
/Users/iang/Documents/Arduino/libraries/micro_ros_arduino-2.0.2-galactic/examples/micro-ros_subscriber_twist/micro-ros_subscriber_twist.ino:54: undefined reference to `rclc_subscription_init_default'
/Users/iang/Documents/Arduino/libraries/micro_ros_arduino-2.0.2-galactic/examples/micro-ros_subscriber_twist/micro-ros_subscriber_twist.ino:61: undefined reference to `rclc_executor_init'
/Users/iang/Documents/Arduino/libraries/micro_ros_arduino-2.0.2-galactic/examples/micro-ros_subscriber_twist/micro-ros_subscriber_twist.ino:62: undefined reference to `rclc_executor_add_subscription'
sketch/micro-ros_subscriber_twist.ino.cpp.o: In function `loop':
/Users/iang/Documents/Arduino/libraries/micro_ros_arduino-2.0.2-galactic/examples/micro-ros_subscriber_twist/micro-ros_subscriber_twist.ino:68: undefined reference to `rclc_executor_spin_some'
collect2: error: ld returned 1 exit status
exit status 1`

igillespie, 2022-01-13:
I tried one last thing. I deleted my micro_ros_arduino library and re-installed from master [here](https://github.com/micro-ROS/micro_ros_arduino). Then I opened the original example and it compiled! YAY!

Thank you for the help everyone. I'm closing this.

closed 2022-01-13 by igillespie