micro-ROS/micro_ros_arduino #704: Performance with joy controller and twist messages to cmd_vel

igillespie, 2022-01-17:
## Issue template

- Hardware description: Arduino UDE
- Installation type: micro_ros_arduino
- Version or commit hash: galactic

This isn't so much as a bug report, but more of a performance issue I'm trying to resolve.

I have a Jetson Nano talking to an Arduino DUE. The Arduino DUE is running micro_ros_arduino and subscribes to twist messages from cmd_vel topic.

If I use teleop_twist_keyboard the twist messages are sent to the Arduino (and my robot wheels move) as expected. This works. But when I try to control through the joy stick, the micro_ros_agent can't seem to keep up with the all of the messages and I end up getting incredible lag between what I do on the joystick and what the motors on the robot do. I tried changing the /joy config 
so that autorepeat_rate is 0.0 and deadline is .2. I also set publisher depth to 10.

Anyone have any ideas on how I might reduce the lag time in my system?

Thanks for any help :-)

pablogs9, 2022-01-18:
Which are the rates of `teleop_twist_keyboard` and your joystick?

igillespie, 2022-01-18:
Thanks for the reply!

I'm launching teleop_twist_keyboard as a standalone node like this and just using the defaults:

`ros2 run teleop_twist_keyboard teleop_twist_keyboard`

When I try to get the params from that node, the system doesn't give me anything (no idea why). When I ask for the teleop_twist_joy_node params I get this:

`/teleop_twist_joy_node:
  ros__parameters:
    axis_angular:
      pitch: -1
      roll: -1
      yaw: 2
    axis_linear:
      x: 5
      y: -1
      z: -1
    enable_button: 7
    enable_turbo_button: 6
    qos_overrides:
      /parameter_events:
        publisher:
          depth: 1000
          durability: volatile
          history: keep_last
          reliability: reliable
    require_enable_button: true
    scale_angular:
      pitch: 0.0
      roll: 0.0
      yaw: 0.5
    scale_angular_turbo:
      pitch: 0.0
      roll: 0.0
      yaw: 4.0
    scale_linear:
      x: 0.5
      y: 0.0
      z: 0.0
    scale_linear_turbo:
      x: 4.0
      y: 0.0
      z: 0.0
    use_sim_time: false`

I'm still relatively new to ros, but I'm very psyched I've made it this far with micro-ros-arduino and ROS2 proper. 

Any ideas of what to try?

pablogs9, 2022-01-19:
Could you try `ros2 topic hz [topic_name]` for both topics?

igillespie, 2022-01-19:
I think these are the relevant topics/info you need (let me know if you need something else!)

CONTROLLING WITH JOYSTICK
Here ros2 topic hz /joy (I went forward several times, then went backward several times. Interestingly, the first few command with the joystick have manageable latency, but it seemed to get progressively worse and worse.)
 
average rate: 3.269
	min: 0.007s max: 1.939s std dev: 0.66712s window: 7
average rate: 3.207
	min: 0.006s max: 3.078s std dev: 0.80071s window: 18
average rate: 3.089
	min: 0.006s max: 3.078s std dev: 0.81456s window: 28
average rate: 3.498
	min: 0.006s max: 3.078s std dev: 0.76340s window: 43
average rate: 4.061
	min: 0.004s max: 3.078s std dev: 0.67761s window: 59
average rate: 4.282
	min: 0.004s max: 3.078s std dev: 0.63734s window: 71
average rate: 4.470
	min: 0.003s max: 3.078s std dev: 0.60852s window: 81
average rate: 4.585
	min: 0.003s max: 3.078s std dev: 0.58986s window: 92
average rate: 4.918
	min: 0.003s max: 3.078s std dev: 0.56482s window: 112
average rate: 4.778
	min: 0.003s max: 3.078s std dev: 0.56731s window: 116
average rate: 5.218
	min: 0.003s max: 3.078s std dev: 0.53444s window: 132
average rate: 5.095
	min: 0.003s max: 3.078s std dev: 0.53742s window: 138
average rate: 5.333
	min: 0.003s max: 3.078s std dev: 0.51704s window: 150
average rate: 5.204
	min: 0.003s max: 3.078s std dev: 0.52649s window: 160
average rate: 6.014
	min: 0.003s max: 3.078s std dev: 0.48554s window: 191
average rate: 5.470
	min: 0.003s max: 6.345s std dev: 0.62766s window: 212

Here is ros2 topic hz /cmd_vel when using the joystick to control

average rate: 3.267
	min: 0.008s max: 1.941s std dev: 0.66762s window: 7
average rate: 3.207
	min: 0.005s max: 3.077s std dev: 0.80069s window: 18
average rate: 3.089
	min: 0.005s max: 3.077s std dev: 0.81456s window: 28
average rate: 3.498
	min: 0.005s max: 3.077s std dev: 0.76344s window: 43
average rate: 4.061
	min: 0.003s max: 3.077s std dev: 0.67770s window: 59
average rate: 4.281
	min: 0.003s max: 3.077s std dev: 0.63750s window: 71
average rate: 4.470
	min: 0.003s max: 3.077s std dev: 0.60874s window: 81
average rate: 4.585
	min: 0.003s max: 3.077s std dev: 0.59008s window: 92
average rate: 4.918
	min: 0.003s max: 3.077s std dev: 0.56504s window: 112
average rate: 4.778
	min: 0.003s max: 3.077s std dev: 0.56751s window: 116
average rate: 5.218
	min: 0.003s max: 3.077s std dev: 0.53463s window: 132
average rate: 5.095
	min: 0.003s max: 3.077s std dev: 0.53760s window: 138
average rate: 5.333
	min: 0.003s max: 3.077s std dev: 0.51723s window: 150
average rate: 5.204
	min: 0.003s max: 3.077s std dev: 0.52663s window: 160
average rate: 6.014
	min: 0.003s max: 3.077s std dev: 0.48567s window: 191
average rate: 5.470
	min: 0.003s max: 6.346s std dev: 0.62778s window: 212

CONTROLLING WITH KEYBOARD
ros2 topic hz /cmd_vel
average rate: 1.627
	min: 0.601s max: 0.628s std dev: 0.01093s window: 3
average rate: 0.976
	min: 0.601s max: 2.255s std dev: 0.71015s window: 4
average rate: 0.778
	min: 0.601s max: 2.326s std dev: 0.82129s window: 5
average rate: 0.714
	min: 0.601s max: 2.326s std dev: 0.79231s window: 6
average rate: 0.712
	min: 0.601s max: 2.326s std dev: 0.73364s window: 7
average rate: 0.737
	min: 0.601s max: 2.326s std dev: 0.69782s window: 8
average rate: 0.832
	min: 0.545s max: 2.326s std dev: 0.69716s window: 10
average rate: 1.136
	min: 0.020s max: 2.326s std dev: 0.73312s window: 15
average rate: 1.047
	min: 0.020s max: 3.630s std dev: 0.95827s window: 18
average rate: 1.152
	min: 0.020s max: 3.630s std dev: 0.91266s window: 21
average rate: 1.031
	min: 0.020s max: 3.683s std dev: 1.05070s window: 23
average rate: 1.019
	min: 0.020s max: 3.683s std dev: 1.02992s window: 24
average rate: 0.853
	min: 0.020s max: 5.740s std dev: 1.37397s window: 25
average rate: 0.858
	min: 0.020s max: 5.740s std dev: 1.34765s window: 26
average rate: 0.830
	min: 0.020s max: 5.740s std dev: 1.33755s window: 27

Acuadros95, 2022-01-20:
Hi, could you share your micro-ros code? 

Also, your agent output on debug move (`-v6` flag) could also be useful, so we can see the timing of the messages as they are sended and acknowledge by the board.

igillespie, 2022-01-20:
There is a LOT of output from the agent. I'm attaching as a text file as it is too big for a comment. I'll upload my Arduino code next.

[micro_ros_agent_v6_log.txt](https://github.com/micro-ROS/micro_ros_arduino/files/7907588/micro_ros_agent_v6_log.txt)

igillespie, 2022-01-20:
Arduino code:

```
#include <micro_ros_arduino.h>

#include <stdio.h>
#include <rcl/rcl.h>
#include <rcl/error_handling.h>
#include <rclc/rclc.h>
#include <rclc/executor.h>

#include <geometry_msgs/msg/twist.h>
#include <std_msgs/msg/int32.h>

#include "TB66MotorController.h"

/*TEST*/
#include <Wire.h>
#include "SSD1306Ascii.h"
#include "SSD1306AsciiWire.h"

#define I2C_ADDRESS 0x3C
SSD1306AsciiWire oled;
// Define proper RST_PIN if required.
#define RST_PIN -1
/*END TEST*/

rcl_subscription_t subscriber;
geometry_msgs__msg__Twist msg;
rclc_executor_t executor;
rcl_allocator_t allocator;
rclc_support_t support;
rcl_node_t node;

rcl_timer_t tick_timer;
rcl_timer_t left_tick_timer;

rcl_publisher_t right_publisher;
std_msgs__msg__Int32 right_tick_msg;

rcl_publisher_t left_publisher;
std_msgs__msg__Int32 left_tick_msg;

//Define the Pins for Motors
//Motor 1
int pinAIN1 = 11; //Direction
int pinAIN2 = 8; //Direction
int pinPWMA = 3; //Speed

//Motor 2
int pinBIN1 = 12; //Direction
int pinBIN2 = 13; //Direction
int pinPWMB = 5; //Speed

//Standby
int pinSTBY = 7;

#define WHEEL_DIST 0.34 //is this distance of wheel to center or distance between wheels?

// Encoder output to Arduino Interrupt pin. Tracks the tick count.
// Other encoder output to Arduino to keep track of wheel direction
// Tracks the direction of rotation.
#define ENC_IN_RIGHT_A 28 //Yellow wire to motor
#define ENC_IN_RIGHT_B 29 //White wire to motor

#define ENC_IN_LEFT_A 30 //yellow
#define ENC_IN_LEFT_B 31 //white 

// Minumum and maximum values for 16-bit integers
const int encoder_minimum = -32768;
[... 247 lines cut ...]
   
  if (Direction_left) {
    if (left_wheel_tick_count == encoder_maximum) {
      left_wheel_tick_count = encoder_minimum;
    }
    else {
      left_wheel_tick_count++;  
    }  
  }
  else {
    if (left_wheel_tick_count == encoder_minimum) {
      left_wheel_tick_count = encoder_maximum;
    }
    else {
      left_wheel_tick_count--;  
    }   
  }
}
void loop()
{
  delay(100);
  RCCHECK(rclc_executor_spin_some(&executor, RCL_MS_TO_NS(100)));

}
```

igillespie, 2022-01-20:
Well, when I pasted the code for the Arduino I saw the delay(100); I commented that out and it WORKS! 

I left it in there because it was in the micro_ros_publisher sample code, I figured it was important. 
I'd love to hear your thoughts on if the delay is important or not. I'm happy to close this too! THANK YOU again!

closed 2022-01-21 by pablogs9

Acuadros95, 2022-01-21:
That's great!
And don't worry about the delay, its not necessary at all.