micro-ROS/micro_ros_arduino #712: 'rosidl_typesupport_c__get_message_type_support_handle__std_msgs__msg_int__Int32' was not declared in this scope

bfan94, 2022-01-23:
### Problem
Trouble when trying to combine two micro_ros_Arduino examples.

### Setup
**Teensy 3.2** using **Arduino 1.8.19** and **Teensyduino 1.56**
PC side is a virtual box vm (this has been working fine for both examples unmodified), **Ubuntu 20.04**,  **ROS2 Galactic**  / **micro_ros_arduino agent** running in docker with --privileged --net host and --dev /dev:/dev for serial pass through.  
Following this tutorial for the pc side of things: [Teensy with Arduino](https://micro.ros.org/docs/tutorials/core/teensy_with_arduino/)

Example projects are [mirco-ros_publisher.ino](https://github.com/micro-ROS/micro_ros_arduino/blob/galactic/examples/micro-ros_publisher/micro-ros_publisher.ino) and [micro-ros_subscriber_twist.ino](https://github.com/micro-ROS/micro_ros_arduino/blob/galactic/examples/micro-ros_subscriber_twist/micro-ros_subscriber_twist.ino).

Error message is as below dealing with the ROSIDL_GET_MSG_TYPE_SUPPORT macro definition from here: [rosidl_runtime_c](https://docs.ros2.org/foxy/api/rosidl_runtime_c/message__type__support__struct_8h.html#a1073bea64e3814387ae5f2e36156155a)

### Short error message:
```cpp
  /home/billy/Arduino/libraries/micro_ros_arduino-2.0.2-galactic/src/rosidl_runtime_c/message_type_support_struct.h:78:59: error: 
  'rosidl_typesupport_c__get_message_type_support_handle__std_msgs__msg_int__Int32' was not declared in this scope
  rosidl_typesupport_c, PkgName, MsgSubfolder, MsgName)()
```

### Lines of code causing error: (specifically ROSIDL Macro)
```cpp
  // create publisher
  RCCHECK(rclc_publisher_init_default(
    &publisher,
    &node,
    ROSIDL_GET_MSG_TYPE_SUPPORT(std_msgs, msg_int, Int32),
    "micro_ros_arduino_node_publisher"));
```

<details><summary>Full Error message</summary>
<p>

```cpp
  /home/billy/Arduino/micro-ros_pub-sub/micro-ros_pub-sub.ino: In function 'void setup()':
  /home/billy/Arduino/libraries/micro_ros_arduino-2.0.2-galactic/src/rosidl_runtime_c/message_type_support_struct.h:78:59: error: 'rosidl_typesupport_c__get_message_type_support_handle__std_msgs__msg_int__Int32' was not declared in this scope
  rosidl_typesupport_c, PkgName, MsgSubfolder, MsgName)()
                                                        ^
  /home/billy/Arduino/micro-ros_pub-sub/micro-ros_pub-sub.ino:28:43: note: in definition of macro 'RCCHECK'
   #define RCCHECK(fn) { rcl_ret_t temp_rc = fn; if((temp_rc != RCL_RET_OK)){error_loop();}}
                                             ^
  /home/billy/Arduino/micro-ros_pub-sub/micro-ros_pub-sub.ino:83:5: note: in expansion of macro 'ROSIDL_GET_MSG_TYPE_SUPPORT'
       ROSIDL_GET_MSG_TYPE_SUPPORT(std_msgs, msg_int, Int32),
       ^
  Using library micro_ros_arduino-2.0.2-galactic at version 0.0.4 in folder: /home/billy/Arduino/libraries/micro_ros_arduino-2.0.2-galactic 
  Error compiling for board Teensy 3.2 / 3.1.

```
</p>
</details>

<details><summary>Full Modified example code here</summary>
<p>

```cpp
#include <micro_ros_arduino.h>

#include <stdio.h>
#include <rcl/rcl.h>
#include <rcl/error_handling.h>
#include <rclc/rclc.h>
#include <rclc/executor.h>

#include <geometry_msgs/msg/twist.h>
#include <std_msgs/msg/int32.h>

rcl_subscription_t subscriber;
geometry_msgs__msg__Twist msg;

rcl_publisher_t publisher;
std_msgs__msg__Int32 msg_int;

rclc_executor_t executor;
rcl_allocator_t allocator;
rclc_support_t support;
rcl_node_t node;

rcl_timer_t timer;

#define LED_PIN 13

#define RCCHECK(fn) { rcl_ret_t temp_rc = fn; if((temp_rc != RCL_RET_OK)){error_loop();}}
#define RCSOFTCHECK(fn) { rcl_ret_t temp_rc = fn; if((temp_rc != RCL_RET_OK)){error_loop();}}

void error_loop(){
  while(1){
    digitalWrite(LED_PIN, !digitalRead(LED_PIN));
    delay(100);
  }
}

void timer_callback(rcl_timer_t * timer, int64_t last_call_time)
{  
  RCLC_UNUSED(last_call_time);
  if (timer != NULL) {
    RCSOFTCHECK(rcl_publish(&publisher, &msg_int, NULL));
    msg_int.data++;
  }
}

//twist message cb
void subscription_callback(const void *msgin) {
  const geometry_msgs__msg__Twist * msg = (const geometry_msgs__msg__Twist *)msgin;
  // if velocity in x direction is 0 turn off LED, if 1 turn on LED
  digitalWrite(LED_PIN, (msg->linear.x == 0) ? LOW : HIGH);
}

void setup() {
  SerialUSB1.begin(9600);
  set_microros_transports();
  pinMode(LED_PIN, OUTPUT);
  digitalWrite(LED_PIN, HIGH);  
  
  delay(7000);
  SerialUSB1.println("testing");

  allocator = rcl_get_default_allocator();

   //create init_options
  RCCHECK(rclc_support_init(&support, 0, NULL, &allocator));

  // create node
  RCCHECK(rclc_node_init_default(&node, "micro_ros_arduino_node", "", &support));

  // create subscriber
  RCCHECK(rclc_subscription_init_default(
    &subscriber,
    &node,
    ROSIDL_GET_MSG_TYPE_SUPPORT(geometry_msgs, msg, Twist),
    "/turtle1/cmd_vel"));

  // create publisher
  RCCHECK(rclc_publisher_init_default(
    &publisher,
    &node,
    ROSIDL_GET_MSG_TYPE_SUPPORT(std_msgs, msg_int, Int32),
    "micro_ros_arduino_node_publisher"));

  // create timer,
  const unsigned int timer_timeout = 1000;
  RCCHECK(rclc_timer_init_default(
    &timer,
    &support,
    RCL_MS_TO_NS(timer_timeout),
    timer_callback));

  // create executor
  RCCHECK(rclc_executor_init(&executor, &support.context, 2, &allocator));
  RCCHECK(rclc_executor_add_subscription(&executor, &subscriber, &msg, &subscription_callback, ON_NEW_DATA));
  RCCHECK(rclc_executor_add_timer(&executor, &timer));

  msg_int.data = 0;

}

void loop() {
  delay(100);
  RCCHECK(rclc_executor_spin_some(&executor, RCL_MS_TO_NS(100)));
  SerialUSB1.print("Random int: ");
  SerialUSB1.print(msg_int.data);
  SerialUSB1.print("Lx: ");
  SerialUSB1.print(msg.linear.x);
  SerialUSB1.print("Ly: ");
  SerialUSB1.print(msg.linear.y);
  SerialUSB1.print("Lz: ");
  SerialUSB1.print(msg.linear.z);

  SerialUSB1.print("Ax: ");
  SerialUSB1.print(msg.angular.x);
  SerialUSB1.print("Ay: ");
  SerialUSB1.print(msg.angular.y);
  SerialUSB1.print("Az: ");
  SerialUSB1.println(msg.angular.z);
}
```
</p>
</details>

### Other notes
Aside from merging the two examples there are a few changes.  

Change third argument from 1 to 2 handles, on for the sub and one for the time.  
`RCCHECK(rclc_executor_init(&executor, &support.context, 2, &allocator));`

Add some serial prints in the main loop on the second USB serial port (this is confirmed to work with the agent and either example individually).  
`SerialUSB1.print(....);`

Change the twist topic (testing working in the original example):
```
  RCCHECK(rclc_subscription_init_default(
    &subscriber,
    &node,
    ROSIDL_GET_MSG_TYPE_SUPPORT(geometry_msgs, msg, Twist),
    "/turtle1/cmd_vel"));
```

pablogs9, 2022-01-24:
It is `ROSIDL_GET_MSG_TYPE_SUPPORT(std_msgs, msg, Int32)` not `ROSIDL_GET_MSG_TYPE_SUPPORT(std_msgs, msg_int, Int32)`. This macro is only retrieves the typesupport and has nothing to do with your type variables.

Create the publisher using `ROSIDL_GET_MSG_TYPE_SUPPORT(std_msgs, msg, Int32)` and then you can do `msg_int.data = ...`

closed 2022-01-24 by pablogs9

bfan94, 2022-01-24:
Thanks for the help.  That is working now!!  

Though I must say I am a little confused about what that value is.  
I can reason the the the first parameter is the message package (geometry_msgs / std_msgs)
And the last parameter is the message name in that package (Int32 / Twist)

I can see that it is labeled as MsgSubfolder but I'm not  sure where that subfolder is specifically.  
It doesn't seem to be shown here: [Int32 msg](http://docs.ros.org/en/melodic/api/std_msgs/html/msg/Int32.html)

EDIT: 

Need to look at the right docs above link is for melodic (ros1) not galactic(ros2)
[https://docs.ros2.org/galactic/api/std_msgs/msg/Int32.html](https://docs.ros2.org/galactic/api/std_msgs/msg/Int32.html)

pablogs9, 2022-01-25:
Those are ROS 2 internal macros, not micro-ROS related.