micro-ROS/micro_ros_arduino #717: Issus while compiling the micro-ros_addtwoints_service in the examples folder

Avinashmuthukumarasamy, 2022-01-25:
### Issue template

- Hardware description: Teensy 3.6
- Installation type: micro-Ros-Arduino
- Version or commit hash: foxy

#### Steps to reproduce the issue
Followed the instruction as in the https://micro.ros.org/docs/tutorials/core/teensy_with_arduino/ ,to upload the code in the teensy. The given micro_ros_publisher code worked as expected in the post.However , while trying to compile the the micro-ros_addtwoints_service in the examples folder, the below mentioned undefined reference error.
Thanks
Note: added the missing example_interfaces header from the [micro_ros library ](https://github.com/micro-ROS/micro_ros_arduino/releases)given in the post
```
WARNING: library micro_ros_arduino claims to run on stm32, OpenCR, Teensyduino, samd, sam, mbed, esp32 architecture(s) and may be incompatible with your current board which runs on avr architecture(s).
Library micro_ros_arduino has been declared precompiled:
Using precompiled library in /home/sl/Arduino/libraries/micro_ros_arduino/src/mk66fx1m0/fpv4-sp-d16-hard
/tmp/arduino_build_495923/sketch/micro-ros_addtwoints_service.ino.cpp.o: In function `setup':
/home/sl/Arduino/libraries/micro_ros_arduino/examples/micro-ros_addtwoints_service/micro-ros_addtwoints_service.ino:45: undefined reference to `rosidl_typesupport_c__get_service_type_support_handle__example_interfaces__srv__AddTwoInts'
collect2: error: ld returned 1 exit status
Error compiling for board Teensy 3.6.
```
#### Expected behavior
successful compilation

#### Actual behavior

throws undefined reference error

#### Additional information

pablogs9, 2022-01-26:
Hello @Avinashmuthukumarasamy,

1.  could you please try to use the latest version of the `foxy` branch in this repo instead of the package available in the release section?
2. Also, I have been able to build the example with any modification, what do you mean with `Note: added the missing example_interfaces header from the micro_ros library given in the post`?
3. Could you share the code you are trying to build?
4. Which versions of Arduino and Teensyduino are you using?

Avinashmuthukumarasamy, 2022-01-26:
Hello @pablogs9
thanks for the much needed reply.
Please find the below points:
1)Initially ,utilized the latest release given in this[ section](https://github.com/micro-ROS/micro_ros_arduino/archive/refs/tags/v2.0.2-foxy.zip),got the below error.Please find the screenshot -ss1
```

WARNING: library micro_ros_arduino-2.0.2-foxy claims to run on OpenCR, Teensyduino, samd, sam, mbed architecture(s) and may be incompatible with your current board which runs on avr architecture(s).
Library micro_ros_arduino-2.0.2-foxy has been declared precompiled:
Precompiled library in "/home/sl/Arduino/libraries/micro_ros_arduino-2.0.2-foxy/src/mk66fx1m0/fpv4-sp-d16-hard" not found
Precompiled library in "/home/sl/Arduino/libraries/micro_ros_arduino-2.0.2-foxy/src/mk66fx1m0" not found
/tmp/arduino_build_458219/sketch/micro-ros_addtwoints_service.ino.cpp.o: In function `set_microros_transports':
/home/sl/Arduino/libraries/micro_ros_arduino-2.0.2-foxy/src/micro_ros_arduino.h:32: undefined reference to `rmw_uros_set_custom_transport'
/tmp/arduino_build_458219/sketch/micro-ros_addtwoints_service.ino.cpp.o: In function `setup':
/home/sl/Arduino/libraries/micro_ros_arduino-2.0.2-foxy/examples/micro-ros_addtwoints_service/micro-ros_addtwoints_service.ino:36: undefined reference to `rcutils_get_default_allocator'
/home/sl/Arduino/libraries/micro_ros_arduino-2.0.2-foxy/examples/micro-ros_addtwoints_service/micro-ros_addtwoints_service.ino:39: undefined reference to `rclc_support_init'
/home/sl/Arduino/libraries/micro_ros_arduino-2.0.2-foxy/examples/micro-ros_addtwoints_service/micro-ros_addtwoints_service.ino:42: undefined reference to `rclc_node_init_default'
/home/sl/Arduino/libraries/micro_ros_arduino-2.0.2-foxy/examples/micro-ros_addtwoints_service/micro-ros_addtwoints_service.ino:45: undefined reference to `rosidl_typesupport_c__get_service_type_support_handle__example_interfaces__srv__AddTwoInts'
/home/sl/Arduino/libraries/micro_ros_arduino-2.0.2-foxy/examples/micro-ros_addtwoints_service/micro-ros_addtwoints_service.ino:45: undefined reference to `rclc_service_init_default'
/home/sl/Arduino/libraries/micro_ros_arduino-2.0.2-foxy/examples/micro-ros_addtwoints_service/micro-ros_addtwoints_service.ino:48: undefined reference to `rclc_executor_init'
/home/sl/Arduino/libraries/micro_ros_arduino-2.0.2-foxy/examples/micro-ros_addtwoints_service/micro-ros_addtwoints_service.ino:49: undefined reference to `rclc_executor_add_service'
/tmp/arduino_build_458219/sketch/micro-ros_addtwoints_service.ino.cpp.o: In function `loop':
/home/sl/Arduino/libraries/micro_ros_arduino-2.0.2-foxy/examples/micro-ros_addtwoints_service/micro-ros_addtwoints_service.ino:55: undefined reference to `rclc_executor_spin_some'
collect2: error: ld returned 1 exit status
Error compiling for board Teensy 3.6.
```
Then I included the recently added  patch-mk66fx1m0 from the [release ](https://github.com/micro-ROS/micro_ros_arduino.git) since my teensy version is 3.6.That time, unfortunately there was no example_interfaces while cloning the library using this [link](https://github.com/micro-ROS/micro_ros_arduino.git)  Please find the screenshot -ss2 

Hence added the example_interfaces folder in the latest release micro_ros_arduino-2.0.2-foxy and tried compiling the micro-ros_addtwo_ints_service  code given in the micro-ros  examples folder,got the above mentioned error in the issue.Please find the screenshot -ss3&ss4

```
WARNING: library micro_ros_arduino-2.0.2-foxy claims to run on OpenCR, Teensyduino, samd, sam, mbed architecture(s) and may be incompatible with your current board which runs on avr architecture(s).
Library micro_ros_arduino-2.0.2-foxy has been declared precompiled:
Using precompiled library in /home/sl/Arduino/libraries/micro_ros_arduino-2.0.2-foxy/src/mk66fx1m0/fpv4-sp-d16-hard
/tmp/arduino_build_458219/sketch/micro-ros_addtwoints_service.ino.cpp.o: In function `setup':
/home/sl/Arduino/libraries/micro_ros_arduino-2.0.2-foxy/examples/micro-ros_addtwoints_service/micro-ros_addtwoints_service.ino:45: undefined reference to `rosidl_typesupport_c__get_service_type_support_handle__example_interfaces__srv__AddTwoInts'
collect2: error: ld returned 1 exit status
Error compiling for board Teensy 3.6.
```

2)as mentioned in 1st point, got the "No file error" while cloning this[ link](https://github.com/micro-ROS/micro_ros_arduino.git). 

3) as same as the example code.No changes/modifications

4)arduino IDE : 1.8.19 
Teensyarduino : 1.56

**Please do the needful.**

[image]
[image]
![ss3](https://user-images.githubusercontent.com/9495
[image]
3760/151165603-d147b59d-7bb3-4394-9423-2fa4605050c8.png)

Acuadros95, 2022-01-26:
Working on this.

Anyway, this bug only happens on galactic. Can you try with the latest commit of `foxy` branch?

Avinashmuthukumarasamy, 2022-01-27:
Hello@Acuadros95

Please find this as an issue about the latest release of foxy.
I have tried with the same.

Acuadros95, 2022-01-27:
This should be fixed with this PRs: https://github.com/micro-ROS/micro_ros_arduino/pull/721

Please update your library to the latest commit and try it out.

pablogs9, 2022-01-31:
Is this solved @Avinashmuthukumarasamy ?

Avinashmuthukumarasamy, 2022-01-31:
Hi @pablogs9!
I used the latest commit of the repo.
The 'micro-ros_addtwo_ints_service sample' code got compiled and uploaded into the teensy 3.6 board successfully.

`root@pi:/microros_ws# ros2 run micro_ros_agent micro_ros_agent serial --dev /dev/ttyACM0
[1643611828.383772] info     | TermiosAgentLinux.cpp | init                     | running...             | fd: 3
[1643611828.384982] info     | Root.cpp           | set_verbose_level        | logger setup           | verbose_level: 4
[1643611856.341986] info     | TermiosAgentLinux.cpp | fini                     | server stopped         | fd: 3
[1643611856.352232] info     | TermiosAgentLinux.cpp | init                     | Serial port not found. | device: /dev/ttyACM0, error 2, waiting for connection...
[1643611857.354447] info     | TermiosAgentLinux.cpp | init                     | Serial port not found. | device: /dev/ttyACM0, error 2, waiting for connection...
[1643611858.357167] info     | TermiosAgentLinux.cpp | init                     | Serial port not found. | device: /dev/ttyACM0, error 2, waiting for connection...
[1643611859.360144] info     | TermiosAgentLinux.cpp | init                     | Serial port not found. | device: /dev/ttyACM0, error 2, waiting for connection...
[1643611860.363784] info     | TermiosAgentLinux.cpp | init                     | Serial port not found. | device: /dev/ttyACM0, error 2, waiting for connection...
[1643611860.934514] info     | TermiosAgentLinux.cpp | init                     | running...             | fd: 3
[1643611861.828202] info     | Root.cpp           | create_client            | create                 | client_key: 0x4FB2BB77, session_id: 0x81
[1643611861.831788] info     | SessionManager.hpp | establish_session        | session established    | client_key: 0x4FB2BB77, address: 0
[1643611861.918958] info     | ProxyClient.cpp    | create_participant       | participant created    | client_key: 0x4FB2BB77, participant_id: 0x000(1)
[1643611861.925885] info     | ProxyClient.cpp    | create_replier           | replier created        | client_key: 0x4FB2BB77, requester_id: 0x000(7), participant_id: 0x000(1)
`
But I am having hiccups while trying to avail the service from the command line. 

`root@pi:/microros_ws# ros2 service list
/addtwoints
root@pi:/microros_ws# ros2 node list
/add_twoints_client_rclc
root@pi:/microros_ws# ros2 service call /addtwoints example_interfaces/srv/AddTwoInts "{a: 3 b: 4}"
Traceback (most recent call last):
  File "/opt/ros/foxy/bin/ros2", line 11, in <module>
    load_entry_point('ros2cli==0.9.10', 'console_scripts', 'ros2')()
  File "/opt/ros/foxy/lib/python3.8/site-packages/ros2cli/cli.py", line 67, in main
    rc = extension.main(parser=parser, args=args)
  File "/opt/ros/foxy/lib/python3.8/site-packages/ros2service/command/service.py", line 41, in main
    return extension.main(args=args)
  File "/opt/ros/foxy/lib/python3.8/site-packages/ros2service/verb/call.py", line 58, in main
    return requester(
  File "/opt/ros/foxy/lib/python3.8/site-packages/ros2service/verb/call.py", line 69, in requester
    module = importlib.import_module('.'.join(parts[:-1]))
  File "/usr/lib/python3.8/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1014, in _gcd_import
  File "<frozen importlib._bootstrap>", line 991, in _find_and_load
  File "<frozen importlib._bootstrap>", line 961, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "<frozen importlib._bootstrap>", line 1014, in _gcd_import
  File "<frozen importlib._bootstrap>", line 991, in _find_and_load
  File "<frozen importlib._bootstrap>", line 973, in _find_and_load_unlocked
ModuleNotFoundError: No module named 'example_interfaces'
`
Please let me know if I am missing out something.

pablogs9, 2022-01-31:
Have you installed `ros-foxy-example-interfaces` in your ROS 2 installation?

`apt install ros-foxy-example-interfaces`

Avinashmuthukumarasamy, 2022-01-31:
Thanks for the reply. I got it working now.

In case of creating custom ROS2 interfaces similar to example interfaces,what is the procedure to add their header files in the arduino libraries .

pablogs9, 2022-01-31:
You have to compile your [own library](https://github.com/micro-ROS/micro_ros_arduino#how-to-build-the-precompiled-library) but include your message packages [here](https://github.com/micro-ROS/micro_ros_arduino/tree/galactic/extras/library_generation/extra_packages) in the .repos or directly in the folder.

[Here](https://micro.ros.org/docs/tutorials/advanced/create_new_type/) you have instructions for creating the package.

closed 2022-01-31 by pablogs9