micro-ROS/micro_ros_arduino #731: Dependency issues while building micro-ROS node with custom interfaces

Avinashmuthukumarasamy, 2022-01-31:
## Issue template

- Hardware description: 
- RTOS: 
- Version or commit hash: 

#### Steps to reproduce the issue
Followed the instructions as in the [post](https://micro.ros.org/docs/tutorials/advanced/create_new_type/) ,to create custom service and message in MyCustomMessage folder within the micro-ros workspace.After building the workspace[Please find SS-1],transferred the header files from /microros_ws/install/my_custom_message/include/my_custom_message/ to the [micro_ros_arduino/src/](https://github.com/micro-ROS/micro_ros_arduino/tree/galactic/src). However while compiling the sample code for testing purpose with above mentioned header file,got the following error. Please let me know if Am missing any steps.

#### Expected behavior

successful compilation

#### Actual behavior

throws some dependency issue

#### Additional information
```

MyCustomService.srv
int32 command
int32 code
---
int32 position
string direction
```
```

MyCustomMessage.msg
int32 command
int32 code

```
```
//sample code to replicate the issue
#include <micro_ros_arduino.h>
#include <example_interfaces/srv/add_two_ints.h>
#include <my_custom_message/srv/my_custom_service.h>
#include <stdio.h>
#include <rcl/error_handling.h>
#include <rclc/rclc.h>
#include <rclc/executor.h>
#include <std_msgs/msg/int64.h>

rcl_node_t node;
rclc_support_t support;
rcl_allocator_t allocator;
rclc_executor_t executor;

rcl_service_t service;
rcl_wait_set_t wait_set;

example_interfaces__srv__AddTwoInts_Response res;
example_interfaces__srv__AddTwoInts_Request req;

#define RCCHECK(fn) { rcl_ret_t temp_rc = fn; if((temp_rc != RCL_RET_OK)){while(1){};}}
#define RCSOFTCHECK(fn) { rcl_ret_t temp_rc = fn; if((temp_rc != RCL_RET_OK)){}}

void service_callback(const void * req, void * res){
  example_interfaces__srv__AddTwoInts_Request * req_in = (example_interfaces__srv__AddTwoInts_Request *) req;
  example_interfaces__srv__AddTwoInts_Response * res_in = (example_interfaces__srv__AddTwoInts_Response *) res;

  //printf("Service request value: %d + %d.\n", (int) req_in->a, (int) req_in->b);

  res_in->sum = req_in->a + req_in->b;
}

void setup() {
  set_microros_transports();
  delay(1000); 

  allocator = rcl_get_default_allocator();

  // create init_options
  RCCHECK(rclc_support_init(&support, 0, NULL, &allocator));

  // create node
  RCCHECK(rclc_node_init_default(&node, "add_twoints_client_rclc", "", &support));

  // create service
  RCCHECK(rclc_service_init_default(&service, &node, ROSIDL_GET_SRV_TYPE_SUPPORT(example_interfaces, srv, AddTwoInts), "/addtwoints"));

  // create executor
  RCCHECK(rclc_executor_init(&executor, &support.context, 1, &allocator));
  RCCHECK(rclc_executor_add_service(&executor, &service, &req, &res, service_callback));
}

void loop() {
  delay(100);
  RCSOFTCHECK(rclc_executor_spin_some(&executor, RCL_MS_TO_NS(100)));
}

```
`Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/teensy/teensy36.html
PLATFORM: Teensy (4.14.0) > Teensy 3.6
HARDWARE: MK66FX1M0 180MHz, 256KB RAM, 1MB Flash
DEBUG: Current (jlink) External (jlink)
PACKAGES: 
 - framework-arduinoteensy 1.155.0 (1.55) 
 - toolchain-gccarmnoneeabi 1.50401.210715 (5.4.1)
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 94 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <Ethernet> 2.0.0
|   |-- <SPI> 1.0
|-- <micro_ros_arduino> 0.0.4+sha.6659d98
|   |-- <Ethernet> 2.0.0
|   |   |-- <SPI> 1.0
|   |-- <SPI> 1.0
|-- <TeensyStep> 2.3.1+sha.1f68646
Building in release mode
Compiling .pio/build/teensy36/lib818/micro_ros_arduino/my_custom_message/msg/detail/my_custom_message__type_support.cpp.o
Compiling .pio/build/teensy36/lib818/micro_ros_arduino/my_custom_message/srv/detail/my_custom_service__type_support.cpp.o
.pio/libdeps/teensy36/micro_ros_arduino/src/my_custom_message/msg/detail/my_custom_message__type_support.cpp:10:59: fatal error: rosidl_typesupport_cpp/message_type_support.hpp: No such file or directory
compilation terminated.
.pio/libdeps/teensy36/micro_ros_arduino/src/my_custom_message/srv/detail/my_custom_service__type_support.cpp:10:59: fatal error: rosidl_typesupport_cpp/message_type_support.hpp: No such file or directory
compilation terminated.
*** [.pio/build/teensy36/lib818/micro_ros_arduino/my_custom_message/msg/detail/my_custom_message__type_support.cpp.o] Error 1
*** [.pio/build/teensy36/lib818/micro_ros_arduino/my_custom_message/srv/detail/my_custom_service__type_support.cpp.o] Error 1`

[image]

pablogs9, 2022-02-01:
In order to add a new message to the Arduino precompiled library, I would suggest to:

1. Create a new ROS 2 package with the message using:

> ```
> ros2 pkg create --build-type ament_cmake my_custom_message
> cd my_custom_message
> mkdir msg
> touch msg/MyCustomMessage.msg
> ```
> In the autogenerated CMakeLists.txt file you should add the following lines just before ament_package():
> 
> ```
> find_package(rosidl_default_generators REQUIRED)
> 
> rosidl_generate_interfaces(${PROJECT_NAME}
>   "msg/MyCustomMessage.msg"
>  )
> ```
> 
> In the autogenerated package.xml file you should add the following lines:
> 
> ```
> <build_depend>rosidl_default_generators</build_depend>
> <exec_depend>rosidl_default_runtime</exec_depend>
> <member_of_group>rosidl_interface_packages</member_of_group>
> ```
> 
> The content of the msg/MyCustomMessage.msg file contains your message definition. For example, let’s include these fields:
> 
> ```
> char char_test
> float32 float32_test
> float64 double_test
> int8 int8_test
> uint8 uint8_test
> ```

2. Once you have the package just copy the generated folder to [`extras/library_generation/extra_packages`](https://github.com/micro-ROS/micro_ros_arduino/tree/galactic/extras/library_generation/extra_packages)
3. Just follow the instructions for rebuilding the Arduino library with your added folders taken into account: [check here](https://github.com/micro-ROS/micro_ros_arduino#how-to-build-the-precompiled-library)

Avinashmuthukumarasamy, 2022-02-01:
Hi @pablogs9,
Thanks for the reply.

I uploaded the generated custom service folder in the [specified folder](https://github.com/micro-ROS/micro_ros_arduino/tree/galactic/extras/library_generation/extra_packages).

`ubuntu@pi:~/Documents/PlatformIO/Projects\Micro_ros_Sample1/.pio/libdeps/teensy36/micro_ros_arduino/extras/library_generation/extra_packages/custom_interfaces$ ls
CMakeLists.txt  package.xml  srv
ubuntu@pi:~/Documents/PlatformIO/Projects\Micro_ros_Sample1/.pio/libdeps/teensy36/micro_ros_arduino/extras/library_generation/extra_packages/custom_interfaces$ `

However while trying to rebuild the arduino folder, could not proceed because the given steps in applicable in the raspberrypi4 which is currently connected to teensy 3.6.

```
ubuntu@pi:~/Documents/PlatformIO/Projects\Micro_ros_Sample1/.pio/libdeps/teensy36/micro_ros_arduino/extras/library_generation/extra_packages/custom_interfaces$ docker run -it --rm -v $(pwd):/project --env MICROROS_LIBRARY_FOLDER=extras microros/micro_ros_static_library_builder:galactic
WARNING: The requested image's platform (linux/amd64) does not match the detected host platform (linux/arm64/v8) and no specific platform was requested
standard_init_linux.go:228: exec user process caused: exec format error
```

Please let me what else I can do to use the custom ROS2 service in the arduino code.

pablogs9, 2022-02-01:
You need to run the docker on a Linux/amd64 computer instead of in a Raspberry Pi. Is that not possible for your use case?

pablogs9, 2022-03-18:
Any update on this?

Avinashmuthukumarasamy, 2022-03-21:
used the same on linux/amd64 computer itself. Thanks.

closed 2022-03-21 by Avinashmuthukumarasamy