micro-ROS/micro_ros_arduino #732: Reliable UDP Transport Breaks Randomly

LucienMorey, 2022-02-01:
Hi,
I am trying to publish messages with the default reliable QOS using the udp ethernet transport. After a random amount of time all messages coming from the micro_ros client cease. The teensy never indicates that it is in an error state at any point.

- Hardware description: Teensy4.1
- RTOS:
- Installation type: micro_ros_arduino with custom messages
- Version or commit hash: Foxy

#### Steps to reproduce the issue
Run the *micro_ros_agent* with udp4 on the default port
```bash
ros2 run micro_ros_agent micro_ros_agent udp4 --port 8888
```
flash the teensy with a basic counting which increments with every message out
```arduino
#include <micro_ros_arduino.h>
#include <rcl/error_handling.h>
#include <rcl/rcl.h>
#include <rclc/executor.h>
#include <rclc/rclc.h>

#include <std_msgs/msg/int32.h>

const int TEENSY_ID = 1;

const String AGENT_IP = "192.168.20.3";
const int AGENT_PORT = 8888;

const String CLIENT_IP = String("192.168.20.").concat(String(TEENSY_ID));

String NODE_NAME = String("teensy").concat(String(TEENSY_ID));

IPAddress client_ip;
IPAddress agent_ip;
byte mac[] = {0x02, 0x47, 0x00, 0x00, 0x00, 0x01};

rclc_executor_t executor;
rclc_support_t support;
rcl_allocator_t allocator;
rcl_node_t node;

rcl_publisher_t publisher;

std_msgs__msg__Int32 counter_msg;

#define LED_PIN 13

#define RCCHECK(fn)                \
  {                                \
    rcl_ret_t temp_rc = fn;        \
    if ((temp_rc != RCL_RET_OK)) { \
      error_loop();                \
    }                              \
  }
#define RCSOFTCHECK(fn)            \
  {                                \
    rcl_ret_t temp_rc = fn;        \
    if ((temp_rc != RCL_RET_OK)) { \
    }                              \
  }

void error_loop()
{
  while (1) {
    digitalWrite(LED_PIN, !digitalRead(LED_PIN));
    delay(100);
  }
}

void setup() {
  // configure ethernet for network comms
  client_ip.fromString(CLIENT_IP);
  agent_ip.fromString(AGENT_IP);

  // set ethernet as the ros transport
  set_microros_native_ethernet_udp_transports(mac, client_ip, agent_ip, AGENT_PORT);

  // configure onboard LED for debugging
  pinMode(LED_PIN, OUTPUT);
  digitalWrite(LED_PIN, HIGH);

  delay(2000);

  allocator = rcl_get_default_allocator();

  //create init_options
  RCCHECK(rclc_support_init(&support, 0, NULL, &allocator));

  // create node
  RCCHECK(rclc_node_init_default(&node, NODE_NAME.c_str(), "", &support));

  // create publisher
  RCCHECK(rclc_publisher_init_default(
    &publisher, &node, ROSIDL_GET_MSG_TYPE_SUPPORT(std_msgs, msg, Int32),
    "test_topic"));

  counter_msg.data = 0;

  RCCHECK(rclc_executor_init(&executor, &support.context, 2, &allocator));

}

void loop() {
  RCSOFTCHECK(rcl_publish(&publisher, &counter_msg, NULL));
  counter_msg.data++;
  RCCHECK(rclc_executor_spin_some(&executor, RCL_MS_TO_NS(100)));
}
```

#### Expected behaviour

I expected this to continue publishing normally indefinitely at least until overflow.

#### Actual behaviour

Messages stop after a random amount of time. This code was based on the examples provided in this repo so I expected the onboard LED to flash if there was a breakdown in communication.

Here is a clip of the agent running with verbose output level 5. The stop at the beginning of the clip was just me reflashing the teensy.

https://user-images.githubusercontent.com/3282391/151907073-8907dc15-2334-4d0d-b6f4-d4e9b4202cb5.mp4

#### Additional information

I have recompiled the library with additional custom messages but since they are not included in this simplified test, I don't think they are relevant.

I saw in another issue that it was suggested to add and spin an executor to help the program deal with the load. That is why it has been included

pablogs9, 2022-02-01:
You are including an executor but your publisher is not using any executor timer. This executor is empty.

Unfortunately, UDP transport for Teensy was a community contribution and we do not have any way to test it. https://github.com/micro-ROS/micro_ros_arduino/pull/625

Could you confirm if using a USB-serial transport has the same behavior?

LucienMorey, 2022-02-01:
The USB-serial transport does not show the same behaviour. I guess that means that the issue is likely either in the UDP transport or in the teensy NativeEthernet lib. Do you have any tips on debugging this issue further?

pablogs9, 2022-02-02:
## Questions

- Which are your custom messages and how are you initializing them? Memory management for types in micro-ROS can lead to some problems.
- When your board stops working, it does not enter in `error_loop()`?

## Transports

I have been taking a look on the UDP transport and I have seen some refactors that are needed:

- [This function](https://github.com/micro-ROS/micro_ros_arduino/blob/6d8337bc3a78573beca230cd58ee710b0c4c02a1/src/native_ethernet_transport.cpp#L69) should be deleted in favour of [this one](https://github.com/micro-ROS/micro_ros_arduino/blob/6d8337bc3a78573beca230cd58ee710b0c4c02a1/src/default_transport.cpp#L17). Maybe we should put it in separate file.
- [Here](https://github.com/micro-ROS/micro_ros_arduino/blob/6d8337bc3a78573beca230cd58ee710b0c4c02a1/src/native_ethernet_transport.cpp#L44-L46) both `beginPacket()` and `endPacket()` can return a failure error code according to [documentation](https://www.arduino.cc/en/Reference/EthernetUDPBeginPacket). Those are not handled properly by the transport.
- [`read()`](https://github.com/micro-ROS/micro_ros_arduino/blob/6d8337bc3a78573beca230cd58ee710b0c4c02a1/src/native_ethernet_transport.cpp#L63) should be called after `parsePacket()` but I'm not sure what happens when you `read()` after a non-suscesfull `parsePacket()`

LucienMorey, 2022-02-02:
## Answers
- some of my custom messages use sequences. here are two of them:
ChannelConfig
```
bool output
bool digital
uint32 channel_number
uint32 resolution
```
TeensyConfig
```
uint32 channel_count
uint32 teensy_id
ChannelConfig[] channel_configs
```
My project involves creating a combination of data acquisition nodes and actuation nodes for different hardware. I send these messages to configure the teensies dynamically. The teensies then send back readings in a similar message type. When initialising the messages and their structures I declare them globally and then initialise capacity and leave memory for it during the setup function. something like this:
```c
smc_msgs__msg__TeensyConfig teensy_config_msg;

void setup(){
  teensy_config_msg.channel_configs.capacity = 24;
  teensy_config_msg.channel_configs.data = (smc_msgs__msg__ChannelConfig *)malloc(
    teensy_config_msg.channel_configs.capacity * sizeof(smc_msgs__msg__ChannelConfig));
  teensy_config_msg.channel_configs.size = 0
}
```
I increment the size attribute before I copy anything into the data array. There is other stuff in *setup* but I haven't included it here because it is not microros related. I don't get any segfaults or other errors from what I can tell so I assume I am initialising memory correctly. Even with these custom types built into the library, they shouldn't have any impact on the code I left in my initial post since they aren't used, right? That code also sees the same stop in data out from the teensy.

- It does not enter the error loop from what I can tell. The onboard LED remains static to indicate there were no errors in *RCCHECK* or *RCSOFTCHECK* during operation.

## Transports
- That seems like a good idea. They look identical to me.
- That does seem like an oversight but what kind of error handling would you expect here? Possibly an attempt to retry sending the packet with a timeout?
- Looking at the source code for the read function I found in the Arduino teensy library:
```c
int EthernetUDP::read(unsigned char *buffer, size_t len)
{
	if (_remaining > 0) {
		int got;
		if (_remaining <= len) {
			// data should fit in the buffer
			got = Ethernet.socketRecv(sockindex, buffer, _remaining);
		} else {
			// too much data for the buffer,
			// grab as much as will fit
			got = Ethernet.socketRecv(sockindex, buffer, len);
		}
		if (got > 0) {
			_remaining -= got;
			//Serial.printf("UDP read %d\n", got);
			return got;
		}
	}
	// If we get here, there's no data available or recv failed
	return -1;
}
```
It seems like it will just return -1 as an error. This means the transport will return 0 which seems like another error condition that may not be properly handled but I am not sure what the transport does with this after it returns.

pablogs9, 2022-02-02:
Said that I guess that the error is not related to the micro-ROS core but to micro-ROS UDP transports for Teensy or the UDP stack itself.

The idea behind handling correctly the transport API is to keep the transports safe. It seems that the read operation is ok with this inner implementation. Regarding the write operation, maybe we should avoid writing when `beginPacket` is faulty. In any case in a write operation, if anything goes wrong is safe to tell the micro-ROS core that 0 Bytes were sent. It will handle retries internally.

Is it possible to take a look on your application code?

LucienMorey, 2022-02-02:
Sure thing. here it is:

```c
#include <Arduino.h>
#include <ads8681.h>

// microros required include. Must be before other microros includes to avoid compilation errors
#include <micro_ros_arduino.h>

// required microros includes
#include <rcl/error_handling.h>
#include <rcl/rcl.h>
#include <rclc/executor.h>
#include <rclc/rclc.h>

// smc msg includes for publishing data
#include <smc_msgs/msg/channel_config.h>
#include <smc_msgs/msg/channel_reading.h>
#include <smc_msgs/msg/teensy_config.h>
#include <smc_msgs/msg/teensy_sample.h>

// teensy id. need to check into burning this into eeprom so it can be read and then removed
const int TEENSY_ID = 1;

// network setting variables
// Agent IP should be changed once used on real hardware to match network
const String AGENT_IP = "192.168.20.3";
const String CLIENT_IP = String("192.168.20.").concat(String(TEENSY_ID));
const int AGENT_PORT = 8888;
IPAddress client_ip;
IPAddress agent_ip;
//check if this can also be done based on the teensy ID
byte mac[] = {0x02, 0x47, 0x00, 0x00, 0x00, 0x01};

String NODE_NAME = String("teensy").concat(String(TEENSY_ID));

//required ros2 structures to create a node
rclc_executor_t executor;
rclc_support_t support;
rcl_allocator_t allocator;
rcl_node_t node;

// reading publisher variables
rcl_publisher_t reading_publisher;
String raw_data_topic = (String("teensy").concat(String(TEENSY_ID))).concat(String("/raw_data"));
smc_msgs__msg__TeensySample teensy_sample_msg;

// config subscriber variables
rcl_subscription_t config_subscriber;
String configuration_topic =
  (String("teensy").concat(String(TEENSY_ID))).concat(String("/hardware_configuration"));
smc_msgs__msg__TeensyConfig teensy_config_msg;

bool publishing = false;

ADS8681 * adc;

// ADC chip constants
// Pins
const uint8_t SCLK = 13;
const uint8_t DIN = 11;
const uint8_t DOUT = 12;
const uint8_t CS = 35;
const uint8_t RST = 34;
const uint8_t S1 = 8;
const uint8_t S2 = 9;
const uint8_t S3 = 24;

uint16_t * adc_data;  // Pointer to the adc data array

[... 124 lines cut ...]
}

void loop()
{
  // put your main code here, to run repeatedly:

  if (publishing) {
    // Get latest ADC readings
    adc_data = adc->readADCs();

    // copy adc reading to msg for each sample
    for (int i = 0; i < teensy_sample_msg.samples.size; i++) {
      teensy_sample_msg.samples.data[i].value = *(adc_data + i);
    }

    RCSOFTCHECK(rcl_publish(&reading_publisher, &teensy_sample_msg, NULL));
  }

  //  TODO Deal with time sync/timing stuff.
  delay(10);
  RCSOFTCHECK(rclc_executor_spin_some(&executor, RCL_MS_TO_NS(100)));
}
```

I can provide copies of the other messages if you need them as well. The *ADC8681* class helps to read from an external SPI ADC chip. I have been substituting it with calls to the onboard ADC during testing and the same behaviour is observed.

pablogs9, 2022-02-02:
Some other comments:
- I will worry about the `new` memory allocation in the subscription callback. Are you sure that it is not being called more times? Maybe you can skip the whole callback functionality if `publishing == true`.
- Is it guaranteed that the node will not receive `smc_msgs__msg__TeensyConfig` with more data than assigned in `msgin.capacity`? 
- In any case, if it works as expected in USB-Serial we can assume that your application is working properly, do you have any way of debugging it using a JTAG adaptor or something similar?
- Does it fails always when it reaches a certain number or when a certain amount of time is elapsed?
- Does it fails if you send with a lower rate? Maybe sending only one data each N `loop()` calls
- Does it fails if you configure a best effort publisher?

LucienMorey, 2022-02-02:
- The operation with other nodes means that the callback only needs to be used once so the callback won't be used again but you are correct about wrapping it in a check for ```publishing== true```.
- It is guaranteed that there will not be more data assigned. 25 sensors are what the board will be configured to handle at the absolute most and additional nodes will be created on other boards if more are required
- I have nothing of the sort to attempt any debugging. I will chat with my company about trying to get hold of one
- There is no consistency in the number of loops before failure. I have observed a difference of up to 2000 loops difference before it fails. I haven't tried running a stopwatch against failure time and it is totally possible that there is some consistency to the amount of time before failure since it sends data almost as fast as it can. I will double-check this.
- I haven't tried changing the number of loops between publishing messages but I have tried increasing the delay time which I think should do the same thing, albeit worse since it is blocking. I will try this.
- It does not fail with the best effort publisher and I haven't noticed any dropped data when it is initialised in this way. I was asked to investigate using reliable transports so that data is guaranteed to be published.

pablogs9, 2022-02-02:
I have created a test branch with the UDP transport refactor: https://github.com/micro-ROS/micro_ros_arduino/pull/739

Can you checkout it and test if it has the same behavior?

Also, I have remembered that we had [some issues with the Teensy timer](https://github.com/micro-ROS/micro_ros_arduino/issues/232#issuecomment-829351791), if this UDP refactor does not work I will prepare a test branch with a test with a timer refactor.

LucienMorey, 2022-02-03:
Your test branch works fine on my end.

It turns out that I am a complete idiot and assigned my teensy the same IP as something else on my network without even thinking about it. I was so surprised that the best effort initialisation worked and the default reliable transport didn't that it didn't even cross my mind. 

It is probably worth merging in your refactor since it handles failures better.

Thank you for all your help trying to debug the situation. I appreciate it!

closed 2022-02-03 by LucienMorey