micro-ROS/micro_ros_arduino #736: Compilation fails for any example project for ESP32 on Platformio

wesleygas, 2022-02-02:
Hi! 

I am trying to build any of the examples for the esp32dev board on PlaformIO, however, the build fails on the final linking step.
 
- Hardware description: ESP32 
- RTOS: None
- Installation type: micro_ros_arduino 
- Version or commit hash: galaxy (6d8337bc3a78573beca230cd58ee710b0c4c02a1)

#### Steps to reproduce the issue
1. Open a new project on platformio
2. Choose the ESP32 Dev Module as board and arduino as framework
3. Add the gitub repository under libdeps and setup the build flags as per readme to get a .ini file as the following:

```ini
[env:esp32dev]
platform = espressif32
board = esp32dev
framework = arduino
lib_deps = 
    https://github.com/micro-ROS/micro_ros_arduino.git
build_flags =
    -L ./.pio/libdeps/esp32dev/micro_ros_arduino/src/esp32/
    -l microros
    -D ESP32
```
4. Copy the content inside any of the examples contained in the `./.pio/libdeps/esp32dev/micro_ros_arduino/src/esp32/` folder eg.
`.pio\libdeps\esp32dev\micro_ros_arduino\examples\micro-ros_publisher\micro-ros_publisher.ino` to your main.cpp file:

```C++
#include <micro_ros_arduino.h>

#include <stdio.h>
#include <rcl/rcl.h>
#include <rcl/error_handling.h>
#include <rclc/rclc.h>
#include <rclc/executor.h>

#include <std_msgs/msg/int32.h>

rcl_publisher_t publisher;
std_msgs__msg__Int32 msg;
rclc_executor_t executor;
rclc_support_t support;
rcl_allocator_t allocator;
rcl_node_t node;
rcl_timer_t timer;

#define LED_PIN 13

#define RCCHECK(fn) { rcl_ret_t temp_rc = fn; if((temp_rc != RCL_RET_OK)){error_loop();}}
#define RCSOFTCHECK(fn) { rcl_ret_t temp_rc = fn; if((temp_rc != RCL_RET_OK)){}}

void error_loop(){
  while(1){
    digitalWrite(LED_PIN, !digitalRead(LED_PIN));
    delay(100);
  }
}

void timer_callback(rcl_timer_t * timer, int64_t last_call_time)
{  
  RCLC_UNUSED(last_call_time);
  if (timer != NULL) {
    RCSOFTCHECK(rcl_publish(&publisher, &msg, NULL));
    msg.data++;
  }
}

void setup() {
  set_microros_transports();
  
  pinMode(LED_PIN, OUTPUT);
  digitalWrite(LED_PIN, HIGH);  
  
  delay(2000);

  allocator = rcl_get_default_allocator();

  //create init_options
  RCCHECK(rclc_support_init(&support, 0, NULL, &allocator));

  // create node
  RCCHECK(rclc_node_init_default(&node, "micro_ros_arduino_node", "", &support));

  // create publisher
  RCCHECK(rclc_publisher_init_default(
    &publisher,
    &node,
    ROSIDL_GET_MSG_TYPE_SUPPORT(std_msgs, msg, Int32),
    "micro_ros_arduino_node_publisher"));

  // create timer,
  const unsigned int timer_timeout = 1000;
  RCCHECK(rclc_timer_init_default(
    &timer,
    &support,
    RCL_MS_TO_NS(timer_timeout),
    timer_callback));

  // create executor
  RCCHECK(rclc_executor_init(&executor, &support.context, 1, &allocator));
  RCCHECK(rclc_executor_add_timer(&executor, &timer));

  msg.data = 0;
}

void loop() {
  delay(100);
  RCSOFTCHECK(rclc_executor_spin_some(&executor, RCL_MS_TO_NS(100)));
}

```

5. Start the build process

#### Expected behavior
sucessful compilation 
#### Actual behavior
Compiling fails with the following error:
```
c:/users/[USER]/.platformio/packages/toolchain-xtensa32/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: final link failed: Bad value
collect2.exe: error: ld returned 1 exit status
*** [.pio\build\esp32dev\firmware.elf] Error 1
```

Acuadros95, 2022-02-02:
This looks like a problem of compatibility between the espressif32 platform and arduino framework, check here: https://github.com/platformio/platform-espressif32/issues/616

As a workaround, this versioning should work:
```
[env:esp32dev]
platform = https://github.com/platformio/platform-espressif32.git#feature/arduino-upstream
board = esp32dev
framework = arduino
lib_deps = 
    https://github.com/micro-ROS/micro_ros_arduino.git
build_flags =
    -L ./.pio/libdeps/esp32dev/micro_ros_arduino/src/esp32/
    -l microros
    -D ESP32

platform_packages =
  toolchain-xtensa32 @ ~2.80400.0
  framework-arduinoespressif32@https://github.com/espressif/arduino-esp32.git#2.0.2

```

wesleygas, 2022-02-02:
That seems to be the problem ineded. Using the versioning you provided I was able to successfully compile the examples.

Thank you for the quick answer!

closed 2022-02-02 by wesleygas