micro-ROS/micro_ros_arduino #740: RMW_RET_ERROR code 1

ajaykumar666, 2022-02-03:
I'm using the stm32f412 board for which I developed the firmware to publish multiple topics for imu, encoders, ultrasonic sensors, and motor control subscribers. everything works well when I echo topics in ROS2 but when I run tele-op micro-ros stop with the error code 1 (RMW_RET_ERROR=1). But when I remove the ultrasonic sensors publisher then everything works fine with tele-op. Any help would be appreciated.

pablogs9, 2022-02-03:
Can you share your application code?
Where is your micro-ROS application failing?

ajaykumar666, 2022-02-03:
``#include <micro_ros_arduino.h>
#include <stdio.h>
#include <rcl/rcl.h>
#include <rcl/error_handling.h>
#include <rclc/rclc.h>
#include <rclc/executor.h>
#include <rmw_microros/rmw_microros.h>

#include <sensor_msgs/msg/range.h>
#include <std_msgs/msg/int16.h>
#include <std_msgs/msg/int32.h>
#include <std_msgs/msg/u_int8.h>
#include <std_msgs/msg/float32_multi_array.h>
#include <std_msgs/msg/int32_multi_array.h>
#include <std_msgs/msg/multi_array_dimension.h>
#include <std_msgs/msg/multi_array_layout.h>

rcl_publisher_t imu;
std_msgs__msg__Float32MultiArray msgImu;

rcl_publisher_t us;
std_msgs__msg__Int32MultiArray uscan;

rcl_publisher_t lwheel_encoder;
std_msgs__msg__Int32 lwheel_int;

rcl_publisher_t rwheel_encoder;
std_msgs__msg__Int32 rwheel_int;

rcl_publisher_t clutch_state_fb;
std_msgs__msg__UInt8 clutch_state_int;

rcl_subscription_t lmotor_cmd;
std_msgs__msg__Int16 ltspeed;

rcl_subscription_t rmotor_cmd;
std_msgs__msg__Int16 rtspeed;

rcl_subscription_t clutch_state;
std_msgs__msg__UInt8 cstate;

rclc_executor_t executor;
rclc_support_t support;
rcl_allocator_t allocator;
rcl_node_t node;
rcl_timer_t timer;
rcl_timer_t us_timer;
rcl_ret_t temp_rc;
#define RCCHECK(fn) {  temp_rc = fn; if((temp_rc != RCL_RET_OK)){error_loop();}}

int break_count = 0;

bool micro_ros_init_successful;

void error_loop() {
  while (1) {
    delay(100);
    Serial.println(temp_rc);
  }
}

void timer_callback(rcl_timer_t * timer, int64_t last_call_time)
{
  (void)last_call_time;
  if (timer != NULL) {
    rcl_publish(&clutch_state_fb, &clutch_state_int, NULL);
    rcl_publish(&lwheel_encoder, &lwheel_int, NULL);
    rcl_publish(&rwheel_encoder, &rwheel_int, NULL);
        rcl_publish(&us, &uscan, NULL);
    rcl_publish(&imu, &msgImu, NULL);
[... 160 lines cut ...]
  uscan.layout.dim.data = (std_msgs__msg__MultiArrayDimension*) malloc(uscan.layout.dim.capacity * sizeof(std_msgs__msg__MultiArrayDimension));

  msgImu.data.capacity = 9;
  msgImu.data.size = sizeof(imuSensor::imu_out) / sizeof(float);
  msgImu.data.data = imuSensor::imu_out;//(float*) malloc(msgImu.data.capacity * sizeof(float));

  msgImu.layout.dim.capacity = 0;
  msgImu.layout.dim.size = 0;
  msgImu.layout.dim.data = (std_msgs__msg__MultiArrayDimension*) malloc(msgImu.layout.dim.capacity * sizeof(std_msgs__msg__MultiArrayDimension));

  if (RMW_RET_OK == rmw_uros_ping_agent(50, 10))
  {
    if (!micro_ros_init_successful) {
      create_entities();
    } else {
      rclc_executor_spin_some(&executor, RCL_MS_TO_NS(100));
    }
  }
  else if (micro_ros_init_successful)
  {
    destroy_entities();

  }

}

pablogs9, 2022-02-03:
Some comments:
- You are `malloc`ing memory in a loop without `free`ing.
- Try to initialize `uscan` and `msgImu` in `setup()`. In the `loop()` you should only set values in `.data` and set `.size`. Also I recommend assigning some memory to these types. In any case you are not initializing properly the `std_msgs/**MultiArray` messages. [Here](https://github.com/micro-ROS/freertos_apps/issues/89#issuecomment-1026521732) you have an example of initing them. Take care of the `.label` members.

For example:
```c
  std_msgs__msg__Float32MultiArray msg;

// In setup() configure and allocate type memory
  float buffer[BUFFER_SIZE] = {};
  msg.data.data = buffer;
  msg.data.size = 0;
  msg.data.capacity = BUFFER_SIZE;

  std_msgs__msg__MultiArrayDimension dim[DIM_SIZE] = {};
  msg.layout.dim.data = dim;
  msg.layout.dim.size = 0;
  msg.layout.dim.capacity = DIM_SIZE;

  char labels[DIM_SIZE][STRING_SIZE] = {};
  for (size_t i = 0; i < DIM_SIZE; i++)
  {
    msg.layout.dim.data[i].label.data = labels[i];
    msg.layout.dim.data[i].label.size = 0;
    msg.layout.dim.data[i].label.capacity = STRING_SIZE;
  }

// In loop() assign values and set sizes
msg.data.data[0] = 3.24;
msg.data.data[1] = 1.23;
msg.data.size = 2;
```

- In order to test the application first avoid the `rmw_uros_ping_agent` block and just use `rclc_executor_spin_some`. When you have the application working then you should handle connections and reconnections.

ajaykumar666, 2022-02-03:
I did change the code as you mentioned but the error is still there.
`#include "config.h"
#include "BotState.h"
#include "Motors.h"
#include "usSensor.h"
#include "Clutch.h"
#include "imuSensor.h"

//kusdu
#include <micro_ros_arduino.h>
#include <stdio.h>
#include <rcl/rcl.h>
#include <rcl/error_handling.h>
#include <rclc/rclc.h>
#include <rclc/executor.h>
#include <rmw_microros/rmw_microros.h>

#include <sensor_msgs/msg/range.h>
#include <std_msgs/msg/int16.h>
#include <std_msgs/msg/int32.h>
#include <std_msgs/msg/u_int8.h>
#include <std_msgs/msg/float32_multi_array.h>
#include <std_msgs/msg/int32_multi_array.h>
#include <std_msgs/msg/multi_array_dimension.h>
#include <std_msgs/msg/multi_array_layout.h>

rcl_publisher_t imu;
std_msgs__msg__Float32MultiArray msgImu;

//rcl_publisher_t us;
//std_msgs__msg__Int32MultiArray uscan;

rcl_publisher_t lwheel_encoder;
std_msgs__msg__Int32 lwheel_int;

rcl_publisher_t rwheel_encoder;
std_msgs__msg__Int32 rwheel_int;

rcl_publisher_t clutch_state_fb;
std_msgs__msg__UInt8 clutch_state_int;

rcl_subscription_t lmotor_cmd;
std_msgs__msg__Int16 ltspeed;

rcl_subscription_t rmotor_cmd;
std_msgs__msg__Int16 rtspeed;

rcl_subscription_t clutch_state;
std_msgs__msg__UInt8 cstate;

rclc_executor_t executor;
rclc_support_t support;
rcl_allocator_t allocator;
rcl_node_t node;
rcl_timer_t timer;
rcl_timer_t us_timer;
rcl_ret_t temp_rc;
#define RCCHECK(fn) {  temp_rc = fn; if((temp_rc != RCL_RET_OK)){error_loop();}}

int break_count = 0;

bool micro_ros_init_successful;

void error_loop() {
  while (1) {
    delay(100);
    Serial.println(temp_rc);
  }
}

[... 195 lines cut ...]
msgImu.data.data[2] = imuSensor::imu_out[2];
msgImu.data.data[3] = imuSensor::imu_out[3];
msgImu.data.data[4] = imuSensor::imu_out[4];
msgImu.data.data[5] = imuSensor::imu_out[5];
msgImu.data.data[6] = imuSensor::imu_out[6];
msgImu.data.data[7] = imuSensor::imu_out[7];
msgImu.data.data[8] = imuSensor::imu_out[8];

msgImu.data.size = 9;

  if (RMW_RET_OK == rmw_uros_ping_agent(50, 10))
  {
    if (!micro_ros_init_successful) {
      create_entities();
    } else {
      rclc_executor_spin_some(&executor, RCL_MS_TO_NS(100));
    }
  }
  else if (micro_ros_init_successful)
  {
    destroy_entities();

  }

}`

pablogs9, 2022-02-03:
`stm32f412` is not a supported board by this package. Can you explain how are you using `micro_ros_arduino` with this board? Which of the precompiled libraries are you using? How are you building your application?

ajaykumar666, 2022-02-03:
I've added the variant file of stm32f412 into Arduino ide and the only library that I'm using are for ultrasonic by Erick Simoes and Adafruit_ICM20948

pablogs9, 2022-02-03:
Which one of the precompiled libraries are you using?

ajaykumar666, 2022-02-03:
Arduino/libraries/micro_ros_arduino-foxy/src/cortex-m4

pablogs9, 2022-02-03:
I have tested this code and it works as expected. Can you try it in your board?

<details>
  <summary>Code</summary>
  
```c
#include <rcl/rcl.h>
#include <rcl/error_handling.h>
#include <rclc/rclc.h>
#include <rclc/executor.h>
#include <rmw_microros/rmw_microros.h>

#include <std_msgs/msg/int16.h>
#include <std_msgs/msg/int32.h>
#include <std_msgs/msg/u_int8.h>
#include <std_msgs/msg/float32_multi_array.h>
#include <std_msgs/msg/int32_multi_array.h>

#define RCCHECK(fn) { rcl_ret_t temp_rc = fn; if((temp_rc != RCL_RET_OK)){printf("Failed status on line %d: %d. Aborting.\n",__LINE__,(int)temp_rc); return;}}
#define RCSOFTCHECK(fn) { rcl_ret_t temp_rc = fn; if((temp_rc != RCL_RET_OK)){printf("Failed status on line %d: %d. Continuing.\n",__LINE__,(int)temp_rc);}}

rcl_publisher_t imu;
std_msgs__msg__Float32MultiArray msgImu;

rcl_publisher_t us;
std_msgs__msg__Int32MultiArray uscan;

rcl_publisher_t lwheel_encoder;
std_msgs__msg__Int32 lwheel_int;

rcl_publisher_t rwheel_encoder;
std_msgs__msg__Int32 rwheel_int;

rcl_publisher_t clutch_state_fb;
std_msgs__msg__UInt8 clutch_state_int;

rcl_subscription_t lmotor_cmd;
std_msgs__msg__Int16 ltspeed;

rcl_subscription_t rmotor_cmd;
std_msgs__msg__Int16 rtspeed;

rcl_subscription_t clutch_state;
std_msgs__msg__UInt8 cstate;

rclc_executor_t executor;
rclc_support_t support;
rcl_allocator_t allocator;
rcl_node_t node;
rcl_timer_t timer;
rcl_timer_t us_timer;
rcl_ret_t temp_rc;

// Callbacks

void timer_callback(rcl_timer_t * timer, int64_t last_call_time)
{
  (void)last_call_time;
  if (timer != NULL) {
    (void)! rcl_publish(&clutch_state_fb, &clutch_state_int, NULL);
    (void)! rcl_publish(&lwheel_encoder, &lwheel_int, NULL);
    (void)! rcl_publish(&rwheel_encoder, &rwheel_int, NULL);
    (void)! rcl_publish(&us, &uscan, NULL);
    (void)! rcl_publish(&imu, &msgImu, NULL);
  }
}

void lcmdVelCB(const void * msgin)
{
  const std_msgs__msg__Int16 * ltspeed = (const std_msgs__msg__Int16 *)msgin;
[... 129 lines cut ...]
  msgImu.data.data[2] = 1.2;
  msgImu.data.data[3] = 1.2;
  msgImu.data.data[4] = 1.2;
  msgImu.data.data[5] = 1.2;
  msgImu.data.data[6] = 1.2;
  msgImu.data.data[7] = 1.2;
  msgImu.data.data[8] = 1.2;
  msgImu.data.size = 9;

  uscan.data.data[0] = 12;
  uscan.data.data[1] = 12;
  uscan.data.data[2] = 12;
  uscan.data.data[3] = 12;
  uscan.data.data[4] = 12;
  uscan.data.data[5] = 12;
  uscan.data.data[6] = 12;
  uscan.data.data[7] = 12;
  uscan.data.data[8] = 12;
  uscan.data.size = 9;

  rclc_executor_spin_some(&executor, RCL_MS_TO_NS(100));
}

```
</details>

pablogs9, 2022-02-07:
Any update on this?

ajaykumar666, 2022-02-07:
Sorry for the late response. the issue is fixed. Thanks for the support.

closed 2022-02-07 by pablogs9