micro-ROS/micro_ros_arduino #744: I can't see my nodes

OtaviodaCruz, 2022-02-03:
## General Description: 

- Arduino Due with  [this code](https://github.com/SciCoBot/scicobot_arduino/blob/main/examples/moveTwist/moveTwist.ino) that uses [ros_init](https://github.com/SciCoBot/ros_init), [motor control](https://github.com/SciCoBot/motor_control) and [ros_motor_control](https://github.com/SciCoBot/ros_motor_control);
- no RTOS;
- Raspberry with 20.04 ubuntu server and ROS 2 foxy;
- Computer with 20.04 ubuntu desktop and ROS 2 foxy;

## Steps to reproduce the issue

I initialized micro micro xrce dds communication between arduino and raspberry:

```
source /opt/ros/foxy/setup.bash
sudo snap install micro-xrce-dds-agent --edge
snap set micro-xrce-dds-agent daemon=true
sudo snap refresh core --edge
sudo snap set core experimental.hotplug=true
sudo systemctl restart snapd
snap interface serial-port
snap connect micro-xrce-dds-agent:serial-port snapd:<slots snapd>
sudo micro-xrce-dds-agent serial --dev /dev/ttyACM0 -v6
```
On the computer I used:

```
source /opt/ros/foxy/setup.bash
ros2 run teleop_twist_keyboard teleop_twist_keyboard
```
So I try to check the topics and nodes in Raspberry with:

```
ros2 topic list
ros2 node list
```

## Expected behavior

I was hoping to see the node created in Due, called "scicobot_arduino", defined in [ros_init](https://github.com/SciCoBot/ros_init/blob/main/src/scicobot_ros_init.cpp)

## Actual behavior

When trying to list, one has

[image]

Note that the scicobot_arduino node does not appear, however the topics appear and the robot receives the commands properly.

### Additional information

Micro xrce dds seems to be working properly

[image]

pablogs9, 2022-02-04:
Hello, Micro XRCE-DDS Agent and micro-ROS Agent are not exactly the same. In fact, Micro XRCE-DDS Agent is not aware of ROS 2 nodes because there is no concept of "node" in DDS.

You need to use micro-ROS Agent. You can build this package in your ROS 2 installation: https://github.com/micro-ROS/micro-ROS-Agent

I do not recommend using snap packages because they are outdated. Use Docker or compile the packages locally.
Reopen if needed.

closed 2022-02-04 by pablogs9

pablogs9, 2022-02-07:
Micro XRCE-DDS is the default middleware of micro-ROS. It bridges the client part with the ROS 2 / DDS world. It works because ROS 2 topics are plain DDS topics, but for example, ROS 2 node discovery has nothing to do with DDS participant discovery.

micro-ROS Agent **uses** Micro XRCE-DDS Agent to implement more functionality on top of the XRCE part. One of the main things that it implements is the [graph manager](https://github.com/micro-ROS/micro-ROS-Agent/tree/galactic/micro_ros_agent/src/agent/graph_manager) that is the module in charge of announcing micro-ROS nodes in the ROS 2 namespace.

Regarding the compilation: just clone [micro-ROS Agent repo](https://github.com/micro-ROS/micro-ROS-Agent) inside your raspberry and build it with colcon:

```bash
source /opt/ros/foxy/setup.bash
mkdir ws && cd ws
git clone -b foxy https://github.com/micro-ROS/micro-ROS-Agent src/micro-ros-agent
git clone -b foxy https://github.com/micro-ROS/micro_ros_msgs src/micro_ros_msgs
colcon build
```

OtaviodaCruz, 2022-02-09:
I was able to use the commands you passed.

Just an adeno for future reference, when I went to build with colcon build the build stuck at one point every time I tried.
From my research it may be a problem related to the raspberry RAM, anyway, [this post](https://answers.ros.org/question/304300/compilling-ros2-on-rasperry-pi/) helped me to solve the problem. In my case I run:
`MAKEFLAGS="-j1 -l1" colcon build --executor sequential`