micro-ROS/micro_ros_arduino #750: Invoking a service yields 2 issues -- (a) incorrect data is received by the service handler on the MCU, and (b) no response is received by the caller

safdark, 2022-02-07:
I've looked through the previous issues regarding service invocations, and incorporated the suggestions into my code, but I still face issues with service invocation. I am trying to use a precompiled micro_ros_arduino library in platformio (obtained from the below repository):
```
https://github.com/DominikN/micro_ros_arduino.git
(branch: galactic_custom_build)
```

I'm hoping for one of the following:
(a) help with what I might be doing wrong in the node.cpp code below, or
(b) which other micro_ros_arduino library/repository I should include in platformio.ini that would work on an ESP32 MCU. (The original one at github.com/micro-ROS/micro_ros_arduino.git does not work for the platformio + esp32 + arduino combination).

Thanks in advance for your help!

## Issue template

- Hardware description: Ubuntu 20.04 (Agent), ESP32 (Client)
- RTOS: Arduino
- Installation type: Platformio
- Version or commit hash: https://github.com/DominikN/micro_ros_arduino.git#galactic_custom_build

#### Steps to reproduce the issue
I've implemented a node with just one service ("/control") handler:
```
node
  /platformio.ini
  /src
    /node.cpp
```

Platformio.ini file:
```
[env]
platform = espressif32
framework = arduino

[env:esp32_node]
board = esp32dev
monitor_speed = 115200
upload_speed = 921600
src_filter = +<node.cpp>
lib_deps =
	https://github.com/DominikN/micro_ros_arduino#galactic-custom-build
build_flags = 
    -L $PROJECT_DIR/.pio/libdeps/$PIOENV/micro_ros_arduino/src/esp32_5_2_0/
    -l microros
```

Node.cpp file:
```
#include <micro_ros_arduino.h>
#include <micro_ros_utilities/string_utilities.h>
#include <rcl/error_handling.h>
#include <rcl/rcl.h>
#include <rclc/executor.h>
#include <rclc/rclc.h>
#include <std_msgs/msg/string.h>
#include <stdio.h>

#include <test_msgs/srv/basic_types.h>

// ------------------
// Setup wifi
// ------------------
const char *ssid = "........";          // <<<<<< FILL >>>>>>>
const char *password = ".........";     // <<<<<< FILL >>>>>>>
const char* agent_ip = ".........";     // <<<<<< FILL >>>>>>>
const uint agent_port = ".........";    // <<<<<< FILL >>>>>>>

// ----------------------
// Error handling macros
// ----------------------
[... 253 lines cut ...]
```

The loop() continues on the MCU, but the caller never gets a response from the service.
```
.
.
.
.
<...>
```

#### Additional information

I can repeatedly issue the service calls. They are received on the MCU, but the issues above are consistent (missing data in the request message, and no response received by the caller).

```
. <Service Invocation callback>Control/State: 0 / -1~
.
.
. <Service Invocation callback>Control/State: 0 / -1~
.
.
.
. <Service Invocation callback>Control/State: 0 / -1~
```

pablogs9, 2022-02-07:
On the ROS 2 side are you using Fast DDS as middleware? In ROS 2 galactic there is another default middleware and micro-ROS is only guaranteed to work with Fast DDS: https://docs.ros.org/en/galactic/Installation/DDS-Implementations/Working-with-eProsima-Fast-DDS.html

safdark, 2022-02-07:
Thanks @pablogs9! That worked perfectly! Unless I've overlooked that detail somewhere, it would be great if this could be documented in the installation section for micro-ROS, since (as you said) the default RMW in galactic is no longer the one that micro-ROS is compatible with.
Thanks again for the prompt reply!

safdark, 2022-02-07:
@pablogs9 ... I did some more testing (after enabling Fast DDS), and am seeing now that though the service calls do generally return, sometimes the invocations are not even received on the MCU (and the call just hangs). So, the issue from before was resolved (in that the MCU now responds to any service call it receives), but this new issue has surfaced. Any thoughts?

safdark, 2022-02-07:
There's no option for me to reopen the issue from this end.

pablogs9, 2022-02-08:
You are not supposed to use `std_msgs__msg__String__init`-like functions in micro-ROS. For example in this case, reallocations are expected to happen when writing the string.

You have to provide memory to your data types in micro-ROS. Please read [this tutorial](https://micro.ros.org/docs/tutorials/advanced/handling_type_memory/). And check [this package](https://github.com/micro-ROS/micro_ros_utilities).

Let us know if this solves the issue.

safdark, 2022-02-08:
Thanks @pablogs9 ! I got this working for initializing the std_msgs/msg/String type (but now the service handler doesn't respond again). So, I presume similar memory allocation is also needed for the service messages? In my case, I use the test_msgs.srv.BasicTypes service interface, for which I define two message objects:
```
test_msgs__srv__BasicTypes_Request request;
test_msgs__srv__BasicTypes_Response response;
```

But I don't see any APIs that seem intuitively matched to the *service* message types, for memory allocation (other than *test_msgs__srv__BasicTypes_Request__init* and *test_msgs__srv__BasicTypes_Response__init*, which you said should not be used in Galactic). Perhaps you could provide some sample code for allocating memory for the above *service* messages?

pablogs9, 2022-02-09:
Is it possible for you to use this repo instead of `galactic_custom_build` of `https://github.com/DominikN/micro_ros_arduino.git`?

safdark, 2022-02-09:
I did try that earlier, but get a linker error:
```
> Executing task: platformio run --environment esp32_node <

Processing esp32_node (board: esp32dev; platform: espressif32; framework: arduino)
----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32dev.html
PLATFORM: Espressif 32 (3.5.0) > Espressif ESP32 Dev Module
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-arduinoespressif32 3.10006.210326 (1.0.6) 
 - tool-esptoolpy 1.30100.210531 (3.1.0) 
 - toolchain-xtensa32 2.50200.97 (5.2.0)
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 29 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <micro_ros_arduino> 0.0.4+sha.d63e576
|   |-- <SPI> 1.0
|   |-- <WiFi> 1.0
|   |-- <ESP32 BLE Arduino> 1.0.1
Building in release mode
Linking .pio/build/esp32_node/firmware.elf
/home/safdar/.platformio/packages/toolchain-xtensa32/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/bin/ld: final link failed: Bad value
collect2: error: ld returned 1 exit status
*** [.pio/build/esp32_node/firmware.elf] Error 1
```

Here's my updated platformio.ini:
```
[env]
platform = espressif32
framework = arduino

[env:esp32_node]
board = esp32dev
monitor_speed = 115200
upload_speed = 921600
src_filter = +<node.cpp>
lib_deps =
    https://github.com/micro-ROS/micro_ros_arduino
build_flags = 
    -L $PROJECT_DIR/.pio/libdeps/$PIOENV/micro_ros_arduino/src/esp32/
    -l microros
```

pablogs9, 2022-02-09:
@Acuadros95 any idea on this linking issue? Do you know how to handle the ESP32 compilers in order to use the library with micro-ROS repo?

@safdark are you using Arduino/Platform.IO for any reason? When using ESP32 we recommend using its development environment (ESP-IDF) and integrating micro-ROS with [micro_ros_espidf_component](https://github.com/micro-ROS/micro_ros_espidf_component) that is well-tested and works smotthly.

safdark, 2022-02-09:
I will check out the micro_ros_espidf_component. I'm new to embedded systems development, so unless tutorials are detailed, I can feel a bit lost ;). In the morning I will give that a try. If you know a detailed article that walks one through setting up the esp-idf component and integration with micro-ROS (assuming no prior familiarity with embedded systems), I'd much appreciate a link to it.

Meanwhile, I do have some Teensy boards as well, and had punted on that for the time being since Teensy's support for certain motor drivers seemed limited (the last time I checked). My end goal is to run various consumer-grade stepper and servo motors from the MCU, using the most commonly available/standard motor driver libraries. I'd be open to trying out a Teensy/Arduino combo, if that is *better* integrated with micro-ros/arduino/platformio. I'm not familiar with FreeRTOS yet ... hence the decision to stick with Arduino. Platformio's accessibility and simplicity for a layman, and its flexibility to switch between different boards, libraries and platforms was why I chose it.

Acuadros95, 2022-02-09:
That linking bug is not related to micro-ROS, but as a work arround:
```
[env:esp32dev]
platform = https://github.com/platformio/platform-espressif32.git#feature/arduino-upstream
board = esp32dev
framework = arduino
lib_deps = 
    https://github.com/micro-ROS/micro_ros_arduino.git
build_flags =
    -L ./.pio/libdeps/esp32dev/micro_ros_arduino/src/esp32/
    -l microros
    -D ESP32

platform_packages =
  toolchain-xtensa32 @ ~2.80400.0
  framework-arduinoespressif32@https://github.com/espressif/arduino-esp32.git#2.0.2
```

This issue was resolved here: https://github.com/micro-ROS/micro_ros_arduino/issues/736

pablogs9, 2022-02-09:
@safdark You have a bunch of documentation on ESP-IDF in their [webpage](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/). Including micro-ROS in an ESP-IDF project is simple, you have instructions in our [repo](https://github.com/micro-ROS/micro_ros_espidf_component#micro-ros-component-for-esp-idf). Open issues there if you have any questions.

safdark, 2022-02-10:
Thanks @Acuadros95 and @pablogs9 .

@Acuadros95 , I tried out the platformio.ini file you listed, and the build compiles and links now. **However**, the service handler on the server is still *intermittently* unresponsive. It is a slight improvement from the previous case, in that at least the service responds a couple of times, but overall, it is still unusable for the above reason. For example, if I issue 6 service calls, ~50% of the calls consistently hang without a response (and the service handler on the MCU doesn't get invoked in those cases). So, as it stands, I am blocked on moving forward with this particular framework/platform/board combination :(.

@pablogs9 Yes, I will switch to using freertos and esp-idf, and see if I have better luck there.

Thanks again for both your help.

pablogs9, 2022-02-10:
Can you share the agent output (with the flag `-v6`) of a normal operation of your setup. Preferably indicating when you are launching a service that is ignored.

pablogs9, 2022-02-10:
Here you have a basic test of your app in our ESP32. No hangs. Have you checked your network?

https://drive.google.com/file/d/1y7e5yruNDLzbIcAv6m7hBrt9TiCoreAD/view?usp=sharing

safdark, 2022-02-10:
That is interesting, and reassuring!!! Thanks so much for the video!

I will check my wifi connection soon, and report back with what I find.

Much appreciated, @pablogs9 :)

safdark, 2022-02-10:
@pablogs9 and @Acuadros95 ... the issue was indeed with the network. I switched to another modality and am not facing this issue now. So, the access point I was previously using between the agent and client was likely the culprit. It is possible the intermittency with sending the service calls (despite all the code fixes) could have been due to using *udp* to communicate over that crappy access point. That might explain why the service invocations never reached the client, though I'd have hoped that a failed invocation would have shown an error on the caller side (ros2 service run ...). Given the inner workings of DDS, can a service caller programmatically detect if an invocation did indeed fail over an unreliable medium between the agent and client? If not, could this be something that a diagnostic tool like *ros wtf* (or other) could have detected earlier on as part of the investigation?

My apologies for the false alarm, and the trouble! And many thanks again for all the help!

closed 2022-02-11 by pablogs9