micro-ROS/micro_ros_arduino #759: Arduino nano rp2040 connect compilation issue

tobratland, 2022-02-10:
## Issue template

- Hardware description: arduino nano rp2040 connect
- RTOS: linux -> arduino ide 1.8.19
- Installation type:  install via .zip import
- Version or commit hash: galactic latest

#### Steps to reproduce the issue
try to compile example micro-ros_decibel

#### Expected behavior
successful compilation
#### Actual behavior
Getting error:

> WARNING: library micro_ros_arduino-2.0.2-galactic claims to run on OpenCR, Teensyduino, samd, sam, mbed architecture(s) and may be incompatible with your current board which runs on mbed_nano architecture(s).
In file included from /home/tobratland/Arduino/sketch_feb10b/sketch_feb10b.ino:1:0:
/home/tobratland/Arduino/libraries/micro_ros_arduino-2.0.2-galactic/src/micro_ros_arduino.h:40:7: warning: extra tokens at end of #else directive [-Wendif-labels]
 #else defined(ARDUINO_NANO_RP2040_CONNECT)
       ^~~~~~~
In file included from /home/tobratland/Arduino/libraries/micro_ros_arduino-2.0.2-galactic/src/rclc/executor.h:30:0,
                 from /home/tobratland/Arduino/sketch_feb10b/sketch_feb10b.ino:9:
/home/tobratland/Arduino/libraries/micro_ros_arduino-2.0.2-galactic/src/rclc/executor_handle.h:35:3: error: redeclaration of 'TIMER'
   TIMER,
   ^~~~~
In file included from /home/tobratland/.arduino15/packages/arduino/hardware/mbed_nano/2.7.2/cores/arduino/Arduino.h:103:0,
                 from sketch/sketch_feb10b.ino.cpp:1:
/home/tobratland/.arduino15/packages/arduino/hardware/mbed_nano/2.7.2/cores/arduino/timer.h:7:3: note: previous declaration 'TimerType TIMER'
   TIMER = 0x1,
   ^~~~~
/home/tobratland/Arduino/sketch_feb10b/sketch_feb10b.ino: In function 'void onPDMdata()':
/home/tobratland/Arduino/sketch_feb10b/sketch_feb10b.ino:107:24: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
   for (size_t i = 0; i < localsamplesRead; i++)
                      ~~^~~~~~~~~~~~~~~~~~
exit status 1
Error compiling for board Arduino Nano RP2040 Connect.

> 

#### Additional information

pablogs9, 2022-02-11:
This has been solved here: https://github.com/ros2/rclc/pull/242. Please use the latest commit in branch `galactic`.

@Acuadros95 We should prepare a new release of micro-ROS for Arduino packages.

Acuadros95, 2022-02-11:
New release `v2.0.3` is out!

tobratland, 2022-02-11:
Thank you, works now.

closed 2022-02-11 by tobratland