micro-ROS/micro_ros_arduino #780: PlatformIO Issue

Salsus2191, 2022-02-24:
## Issue template

- Hardware description: Arduino Nano RP 2040
- RTOS: 
- Installation type: 
- Version or commit hash: galactic v2.0.3

#### Steps to reproduce the issue
Create PlatformIO project
Add 
lib_deps = 
	https://github.com/micro-ROS/micro_ros_arduino
	arduino-libraries/WiFiNINA@^1.8.13

build_flags = 
	-L ./.pio/libdeps/nanorp2040connect/micro_ros_arduino/src/cortex-m0plus/
	-l microros
	-D TARGET=/dev/ttyACM0

lib_ldf_mode = chain+

to the Platformio.ini file
then include the micro_ros_arduino header in the main cpp file.
use pio lib install
then pio run

#### Expected behavior

successful compilation

#### Actual behavior

The program runs into a variety of issues:
It can't link properly and outputs an error message about a lot of different undefined references
It tries to include the SPI.h header but it can't find it
Sometimes it works as expected even though nothing about it was changed
If the line lib_ldf_mode =  chain+ is missing, it tries to include a variety of files that it cant find 
All of these errors seem random since it sometimes works and sometimes it doesn't

#### Additional information

I've tried excluding the micro_ros_arduino header and it sometimes works then

Acuadros95, 2022-02-24:
Could you share your complete `Platformio.ini` file?

Acuadros95, 2022-02-24:
Ok, here is the fix:
- First, we need to modify the linker scripts (Related: https://github.com/micro-ROS/micro_ros_arduino/issues/774):
  Create a python script `fix_linker.py` on your directory with the following code:
  ```
  Import("env")
  
  env["_LIBFLAGS"] =  ('-Wl,--start-group -Wl,--whole-archive '
                      '${_stripixes(LIBLINKPREFIX, LIBS, LIBLINKSUFFIX, LIBPREFIXES, LIBSUFFIXES, __env__)} '
                      '-Wl,--no-whole-archive -lstdc++ -lsupc++ -lm -lc -lgcc -lnosys -lmicroros -Wl,--end-group')
  ```
  
 Add this script to your `platformio.ini` like this: `extra_scripts = fix_linker.py` and delete the `-l libmicroros` line on build_flags.

- The final `platformio.ini` with some versions fix looks like this: 

  ```
  [env:nanorp2040connect]
  platform = raspberrypi
  board = nanorp2040connect
  framework = arduino
  extra_scripts = fix_linker.py
  
  build_flags =
      -L ./.pio/libdeps/nanorp2040connect/micro_ros_arduino/src/cortex-m0plus/
      -D ARDUINO_NANO_RP2040_CONNECT
  
  lib_deps =
      arduino-libraries/WiFiNINA@^1.8.13
      https://github.com/micro-ROS/micro_ros_arduino
  
  platform_packages =
    toolchain-gccarmnoneeabi @ ~1.70201.0
    framework-arduino-mbed @ ~2.4.1
  
  lib_ldf_mode = chain+
  ```

closed 2022-02-24 by Acuadros95

Redstone-RM, 2022-03-23:
I had ran into the above issue of SPI.h being reported as missing and this solution did not work on it's own. 
I additionally had to include SPI otherwise it had the same error. 

It compiles now but I do get a couple of warnings...
Should I worry or did I miss something else with the fix?
`
[env:nanorp2040connect]
platform = raspberrypi
board = nanorp2040connect
framework = arduino
extra_scripts = fix_linker.py

build_flags =
    -L ./.pio/libdeps/nanorp2040connect/micro_ros_arduino/src/cortex-m0plus/
    -D ARDUINO_NANO_RP2040_CONNECT

lib_deps =
    arduino-libraries/WiFiNINA@^1.8.13
    https://github.com/micro-ROS/micro_ros_arduino
    SPI

platform_packages =
  toolchain-gccarmnoneeabi @ ~1.70201.0
  framework-arduino-mbed @ ~2.4.1

lib_ldf_mode = chain+

`

`
Archiving .pio/build/nanorp2040connect/libb5c/libWiFiNINA.a
Indexing .pio/build/nanorp2040connect/libb5c/libWiFiNINA.a
Compiling .pio/build/nanorp2040connect/FrameworkArduinoVariant/nina_pins.cpp.o
Compiling .pio/build/nanorp2040connect/FrameworkArduinoVariant/variant.cpp.o
.pio/libdeps/nanorp2040connect/micro_ros_arduino/src/wifi_transport.cpp: In function 'size_t arduino_wifi_transport_read(uxrCustomTransport*, uint8_t*, size_t, int, uint8_t*)':
.pio/libdeps/nanorp2040connect/micro_ros_arduino/src/wifi_transport.cpp:53:33: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
     while(millis() - start_time < timeout && udp_client.parsePacket() == 0){
           ~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~
.pio/libdeps/nanorp2040connect/micro_ros_arduino/src/wifi_transport.cpp:59:20: warning: comparison of unsigned expression < 0 is always false [-Wtype-limits]
     return (readed < 0) ? 0 : readed;
             ~~~~~~~^~~
Compiling .pio/build/nanorp2040connect/FrameworkArduino/Interrupts.cpp.o
`

Acuadros95, 2022-03-24:
@Redstone-RM Don't worry about those warnings, they are minor signed warnings on the wifi transport.