micro-ROS/micro_ros_arduino #781: ROS2 not detecting custom micro_ros_arduino service

kajMork, 2022-02-24:
## Issue template

- Hardware description: esp32 nodemcu
- RTOS: PlatformIO
- Version or commit hash: Galactic

### Steps to reproduce the issue
I have pretty much stumbled upon issues at every step of the process, so I documented them all on this wiki:
https://github.com/kajMork/Brick_Feeder/wiki  

#### micro_ros_arduino service code:
```C++
#include <micro_ros_arduino.h>

#include <brickfeeder_cmd/srv/brickfeederservice.h>
#include <example_interfaces/srv/add_two_ints.h>
#include <stdio.h>
#include <rcl/rcl.h>
#include <rcl/error_handling.h>
#include <rclc/rclc.h>
#include <rclc/executor.h>
#include <std_msgs/msg/int64.h>

#if CONFIG_FREERTOS_UNICORE
static const BaseType_t app_cpu = 0;
#else
static const BaseType_t app_cpu = 1;
#endif 

static const int led_pin = LED_BUILTIN;

  int state = 0;
  int steptimer = 5;
  int read_ir_timer = 3;
  int motor_timer = 6;
  int obj_count = 0;
  int obj_dispensed = 0;
  unsigned long solenoid_off_timer = 1000;
  unsigned long solenoid_on_timer = 100;
  unsigned long begin_time;
  unsigned long end_time;
  unsigned long currentMillis;
  unsigned long previous_step_Millis = 0;
  unsigned long previous_ir_Millis = 0;
  unsigned long previous_motor_Millis = 0;
  boolean new_object = true;
  boolean new_job = false;
  boolean job_ran = false;

  int step_pin = 26;
  int dir_pin = 27;
  int enable_pin = 14;
  int solenoid_pin =0;
  int MS1_pin = 12;
  int MS2_pin = 32;
  int MS3_pin = 33;
  int ir_pin = 25;

  int out_ir = 23;
  int out_read = 22;
  int out_solenoid = 21; 

  char agent_ip[15] = "YOUR_IP";
  char SSID_name[15] = "YOUR_PSW";
  char SSID_psw[13] = "YOUR_PSW";

rcl_node_t node;
rclc_support_t support;
rcl_allocator_t allocator;
rclc_executor_t executor;
[... 301 lines cut ...]
0000: 81 80 01 00 01 07 D2 00 00 0B 00 08 08 03 00 00 C4 00 00 00 0D 00 00 00 2F 62 72 69 63 6B 66 65
0020: 65 64 65 72 00 00 00 00 38 00 00 00 62 72 69 63 6B 66 65 65 64 65 72 5F 63 6D 64 3A 3A 73 72 76
0040: 3A 3A 64 64 73 5F 3A 3A 42 72 69 63 6B 66 65 65 64 65 72 73 65 72 76 69 63 65 5F 52 65 71 75 65
0060: 73 74 5F 00 39 00 00 00 62 72 69 63 6B 66 65 65 64 65 72 5F 63 6D 64 3A 3A 73 72 76 3A 3A 64 64
0080: 73 5F 3A 3A 42 72 69 63 6B 66 65 65 64 65 72 73 65 72 76 69 63 65 5F 52 65 73 70 6F 6E 73 65 5F
00A0: 00 01 3F B3 16 00 00 00 72 71 2F 62 72 69 63 6B 66 65 65 64 65 72 52 65 71 75 65 73 74 00 01 00
00C0: 14 00 00 00 72 72 2F 62 72 69 63 6B 66 65 65 64 65 72 52 65 70 6C 79 00 00 01 00 00
[1645727565.571386] info     | ProxyClient.cpp    | create_replier           | replier created        | client_key: 0x4B419E83, requester_id: 0x000(7), participant_id: 0x000(1)
[1645727565.571557] debug    | UDPv4AgentLinux.cpp | send_message             | [** <<UDP>> **]        | client_key: 0x4B419E83, len: 14, data: 
0000: 81 80 01 00 05 01 06 00 00 0B 00 08 00 00
[1645727565.571589] debug    | UDPv4AgentLinux.cpp | send_message             | [** <<UDP>> **]        | client_key: 0x4B419E83, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 02 00 00 00 80
[1645727565.575548] debug    | UDPv4AgentLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0x4B419E83, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 02 00 00 00 80
[1645727565.684303] debug    | UDPv4AgentLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0x4B419E83, len: 24, data: 
0000: 81 80 02 00 08 01 10 00 00 0C 00 08 80 00 00 01 FF FF 00 00 00 00 00 00
[1645727565.684739] debug    | UDPv4AgentLinux.cpp | send_message             | [** <<UDP>> **]        | client_key: 0x4B419E83, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 03 00 00 00 80
[1645727565.687940] debug    | UDPv4AgentLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0x4B419E83, len: 13, data: 
0000: 81 00 00 00 0B 01 05 00 01 00 02 00 80
[1645727565.688236] debug    | UDPv4AgentLinux.cpp | send_message             | [** <<UDP>> **]        | client_key: 0x4B419E83, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 03 00 00 00 80
```
### Additional information
I have tested the same code, but with the prebuild service AddTwoInts from the package examle_interfaces which works fine.

pablogs9, 2022-02-25:
Does this work you use the `ros2 service call ...` CLI command instead of using a custom Python application?

I assume that you are using Fast DDS in all your ROS 2 Galactic installations, Cyclone is not compatible with micro-ROS.

Acuadros95, 2022-02-25:
Whats inside of your custom `brickfeederservice` message?

kajMork, 2022-02-25:
> Whats inside of your custom `brickfeederservice` message?

It contains the following:
```srv
int64 brickgoal
---
bool goaldone
```

Acuadros95, 2022-02-25:
Check the return value of `rclc_executor_spin_some`, it may give some info.

Also, try to replicate this with a minimum example, leaving out the complementary task.
If the problem persist on a minimal example I will build and test it out.

kajMork, 2022-02-25:
> Does this work you use the `ros2 service call ...` CLI command instead of using a custom Python application?
> 
> I assume that you are using Fast DDS in all your ROS 2 Galactic installations, Cyclone is not compatible with micro-ROS.

Okay, so i tested with the following CLI command:
```console
ros2 service call /brickfeeder brickfeeder_cmd/srv/Brickfeederservice "{brickgoal: 3}"
```
Which results in the following: 
```console
requester: making request: brickfeeder_cmd.srv.Brickfeederservice_Request(brickgoal=3)

response:
brickfeeder_cmd.srv.Brickfeederservice_Response(goaldone=False)
```
Which is the expected result!

This helped me find the problem, which was a stupid mistake from my end.
Turns out that i had the following line:
```python
self.future = self.cli.call_async(self.req)
``` 
In an if statement that would only be accessed when i tested with service AddTwoInts from the package examle_interfaces
```python
def send_request(self):
    if int(sys.argv[3]) == 0: # if sys.argv[3] == 0 then is use my custom brickfeeder service
        self.req.brickgoal = int(sys.argv[1])
    if int(sys.argv[3]) == 1: # if sys.argv[3]  == 1 then is use AddTwoInts service
        self.req.a = int(sys.argv[1])
        self.req.b = int(sys.argv[2])
        self.future = self.cli.call_async(self.req) # Here is the mistake.
```

Thanks for the help and sorry for the inconvenience.

closed 2022-02-25 by kajMork