micro-ROS/micro_ros_arduino #788: Compile errors Teensy 4.0 "Undefined reference"

bodely97, 2022-02-28:
Hi, 

I installed the micro-ros library with the latest release for foxy.
When I try to compile any example I get multiple **undefined reference** errors:

`
C:\Users\felix\Documents\Arduino\sketch_feb28a/sketch_feb28a.ino:48: undefined reference to `arduino_transport_read'

C:\Users\felix\Documents\Arduino\sketch_feb28a/sketch_feb28a.ino:48: undefined reference to `arduino_transport_write'

C:\Users\felix\Documents\Arduino\sketch_feb28a/sketch_feb28a.ino:48: undefined reference to `arduino_transport_open'

C:\Users\felix\Documents\Arduino\sketch_feb28a/sketch_feb28a.ino:48: undefined reference to `arduino_transport_close'

C:\Users\felix\Documents\Arduino\libraries\micro_ros_arduino-2.0.3-foxy\src\imxrt1062\fpv5-d16-hard\libmicroros.a(librcutils-time_unix.c.obj): In function `rcutils_system_time_now':

time_unix.c:(.text.rcutils_system_time_now+0xc): undefined reference to `clock_gettime'

C:\Users\felix\Documents\Arduino\libraries\micro_ros_arduino-2.0.3-foxy\src\imxrt1062\fpv5-d16-hard\libmicroros.a(librcutils-time_unix.c.obj): In function `rcutils_steady_time_now':

time_unix.c:(.text.rcutils_steady_time_now+0xc): undefined reference to `clock_gettime'

C:\Users\felix\Documents\Arduino\libraries\micro_ros_arduino-2.0.3-foxy\src\imxrt1062\fpv5-d16-hard\libmicroros.a(libmicroxrcedds_client-time.c.obj): In function `uxr_millis':

time.c:(.text.uxr_millis+0xa): undefined reference to `clock_gettime'

C:\Users\felix\Documents\Arduino\libraries\micro_ros_arduino-2.0.3-foxy\src\imxrt1062\fpv5-d16-hard\libmicroros.a(libmicroxrcedds_client-time.c.obj): In function `uxr_nanos':

time.c:(.text.uxr_nanos+0x8): undefined reference to `clock_gettime'

collect2.exe: error: ld returned 1 exit status

Bibliothek micro_ros_arduino-2.0.3-foxy in Version 0.0.4 im Ordner: C:\Users\felix\Documents\Arduino\libraries\micro_ros_arduino-2.0.3-foxy  wird verwendet
Fehler beim Kompilieren für das Board Teensy 4.0.
`

What can I do? Can you please help me with this issue?
Thx 
Felix :)

pablogs9, 2022-03-01:
Is [this file](https://github.com/micro-ROS/micro_ros_arduino/blob/galactic/src/default_transport.cpp) being built?

bodely97, 2022-03-01:
I thought that it is already built in the ZIP-library. 
Do I have to add this class file manually to my Arduino project??

pablogs9, 2022-03-01:
Well if you are including the library properly, it should be detected and built, which is the log of your compiler?

bodely97, 2022-03-01:
Do you mean the logs when I add the library?
When I try to compile my project it detects the correct library I added via Sketch -> Add library .zip file

bodely97, 2022-03-01:
This is already activated. 
All the errors I get are the ones in my issue.
Here is the complete output:

Arduino: 1.8.12 (Windows 10), TD: 1.56, Board: "Teensy 4.0, Serial, 600 MHz, Faster, US English"

C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\felix\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\felix\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\felix\Documents\Arduino\libraries -fqbn=teensy:avr:teensy40:usb=serial,speed=600,opt=o2std,keys=en-us -ide-version=10812 -build-path C:\Users\felix\AppData\Local\Temp\arduino_build_489404 -warnings=all -build-cache C:\Users\felix\AppData\Local\Temp\arduino_cache_808423 -verbose C:\Users\felix\Documents\Arduino\sketch_feb28a\sketch_feb28a.ino
C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\felix\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\felix\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\felix\Documents\Arduino\libraries -fqbn=teensy:avr:teensy40:usb=serial,speed=600,opt=o2std,keys=en-us -ide-version=10812 -build-path C:\Users\felix\AppData\Local\Temp\arduino_build_489404 -warnings=all -build-cache C:\Users\felix\AppData\Local\Temp\arduino_cache_808423 -verbose C:\Users\felix\Documents\Arduino\sketch_feb28a\sketch_feb28a.ino
Using board 'teensy40' from platform in folder: C:\Program Files (x86)\Arduino\hardware\teensy\avr
Using core 'teensy4' from platform in folder: C:\Program Files (x86)\Arduino\hardware\teensy\avr
Detecting libraries used...
"C:\\Program Files (x86)\\Arduino\\hardware\\teensy/../tools/arm/bin/arm-none-eabi-g++" -E -CC -x c++ -w -g -Wall -ffunction-sections -fdata-sections -nostdlib -std=gnu++14 -fno-exceptions -fpermissive -fno-rtti -fno-threadsafe-statics -felide-constructors -Wno-error=narrowing -mthumb -mcpu=cortex-m7 -mfloat-abi=hard -mfpu=fpv5-d16 -D__IMXRT1062__ -DTEENSYDUINO=156 -DARDUINO=10812 -DARDUINO_TEENSY40 -DF_CPU=600000000 -DUSB_SERIAL -DLAYOUT_US_ENGLISH "-IC:\\Program Files (x86)\\Arduino\\hardware\\teensy\\avr\\cores\\teensy4" "C:\\Users\\felix\\AppData\\Local\\Temp\\arduino_build_489404\\sketch\\sketch_feb28a.ino.cpp" -o nul
Alternatives for micro_ros_arduino.h: [micro_ros_arduino-2.0.3-foxy@0.0.4]
ResolveLibrary(micro_ros_arduino.h)
  -> candidates: [micro_ros_arduino-2.0.3-foxy@0.0.4]
"C:\\Program Files (x86)\\Arduino\\hardware\\teensy/../tools/arm/bin/arm-none-eabi-g++" -E -CC -x c++ -w -g -Wall -ffunction-sections -fdata-sections -nostdlib -std=gnu++14 -fno-exceptions -fpermissive -fno-rtti -fno-threadsafe-statics -felide-constructors -Wno-error=narrowing -mthumb -mcpu=cortex-m7 -mfloat-abi=hard -mfpu=fpv5-d16 -D__IMXRT1062__ -DTEENSYDUINO=156 -DARDUINO=10812 -DARDUINO_TEENSY40 -DF_CPU=600000000 -DUSB_SERIAL -DLAYOUT_US_ENGLISH "-IC:\\Program Files (x86)\\Arduino\\hardware\\teensy\\avr\\cores\\teensy4" "-IC:\\Users\\felix\\Documents\\Arduino\\libraries\\micro_ros_arduino-2.0.3-foxy\\src" "C:\\Users\\felix\\AppData\\Local\\Temp\\arduino_build_489404\\sketch\\sketch_feb28a.ino.cpp" -o nul
"C:\\Program Files (x86)\\Arduino\\hardware\\teensy/../tools/arm/bin/arm-none-eabi-g++" -E -CC -x c++ -w -g -Wall -ffunction-sections -fdata-sections -nostdlib -std=gnu++14 -fno-exceptions -fpermissive -fno-rtti -fno-threadsafe-statics -felide-constructors -Wno-error=narrowing -mthumb -mcpu=cortex-m7 -mfloat-abi=hard -mfpu=fpv5-d16 -D__IMXRT1062__ -DTEENSYDUINO=156 -DARDUINO=10812 -DARDUINO_TEENSY40 -DF_CPU=600000000 -DUSB_SERIAL -DLAYOUT_US_ENGLISH "-IC:\\Program Files (x86)\\Arduino\\hardware\\teensy\\avr\\cores\\teensy4" "-IC:\\Users\\felix\\Documents\\Arduino\\libraries\\micro_ros_arduino-2.0.3-foxy\\src" "C:\\Users\\felix\\Documents\\Arduino\\libraries\\micro_ros_arduino-2.0.3-foxy\\src\\default_transport.cpp" -o nul
"C:\\Program Files (x86)\\Arduino\\hardware\\teensy/../tools/arm/bin/arm-none-eabi-g++" -E -CC -x c++ -w -g -Wall -ffunction-sections -fdata-sections -nostdlib -std=gnu++14 -fno-exceptions -fpermissive -fno-rtti -fno-threadsafe-statics -felide-constructors -Wno-error=narrowing -mthumb -mcpu=cortex-m7 -mfloat-abi=hard -mfpu=fpv5-d16 -D__IMXRT1062__ -DTEENSYDUINO=156 -DARDUINO=10812 -DARDUINO_TEENSY40 -DF_CPU=600000000 -DUSB_SERIAL -DLAYOUT_US_ENGLISH "-IC:\\Program Files (x86)\\Arduino\\hardware\\teensy\\avr\\cores\\teensy4" "-IC:\\Users\\felix\\Documents\\Arduino\\libraries\\micro_ros_arduino-2.0.3-foxy\\src" "C:\\Users\\felix\\Documents\\Arduino\\libraries\\micro_ros_arduino-2.0.3-foxy\\src\\native_ethernet_transport.cpp" -o nul
"C:\\Program Files (x86)\\Arduino\\hardware\\teensy/../tools/arm/bin/arm-none-eabi-g++" -E -CC -x c++ -w -g -Wall -ffunction-sections -fdata-sections -nostdlib -std=gnu++14 -fno-exceptions -fpermissive -fno-rtti -fno-threadsafe-statics -felide-constructors -Wno-error=narrowing -mthumb -mcpu=cortex-m7 -mfloat-abi=hard -mfpu=fpv5-d16 -D__IMXRT1062__ -DTEENSYDUINO=156 -DARDUINO=10812 -DARDUINO_TEENSY40 -DF_CPU=600000000 -DUSB_SERIAL -DLAYOUT_US_ENGLISH "-IC:\\Program Files (x86)\\Arduino\\hardware\\teensy\\avr\\cores\\teensy4" "-IC:\\Users\\fe [... cut ...]
C:\Users\felix\Documents\Arduino\libraries\micro_ros_arduino-2.0.3-foxy\src\imxrt1062\fpv5-d16-hard\libmicroros.a(libmicroxrcedds_client-time.c.obj): In function `uxr_nanos':

time.c:(.text.uxr_nanos+0x8): undefined reference to `clock_gettime'

collect2.exe: error: ld returned 1 exit status

Bibliothek micro_ros_arduino-2.0.3-foxy in Version 0.0.4 im Ordner: C:\Users\felix\Documents\Arduino\libraries\micro_ros_arduino-2.0.3-foxy  wird verwendet
Fehler beim Kompilieren für das Board Teensy 4.0.

pablogs9, 2022-03-02:
Which version of teensyduino are you using?

bodely97, 2022-03-02:
1.5.6

pablogs9, 2022-03-03:
Did you [patch the teensyduino installation](https://github.com/micro-ROS/micro_ros_arduino#patch-teensyduino)?

bodely97, 2022-03-03:
Yes, I replaced the platform.txt file with the one in the **extras** folder of this project.
Do I have to place it in multiple directories or only into the folder **Arduino/harware/teensy/avr**?

pablogs9, 2022-03-03:
In my installation, it is under `~/.arduino15/packages/teensy/avr/platform.txt`.

In any case, it is weird because the file is being built here:

```
...
"C:\\Program Files (x86)\\Arduino\\hardware\\teensy/../tools/arm/bin/arm-none-eabi-g++" -E -CC -x c++ -w -g -Wall -ffunction-sections -fdata-sections -nostdlib -std=gnu++14 -fno-exceptions -fpermissive -fno-rtti -fno-threadsafe-statics -felide-constructors -Wno-error=narrowing -mthumb -mcpu=cortex-m7 -mfloat-abi=hard -mfpu=fpv5-d16 -D__IMXRT1062__ -DTEENSYDUINO=156 -DARDUINO=10812 -DARDUINO_TEENSY40 -DF_CPU=600000000 -DUSB_SERIAL -DLAYOUT_US_ENGLISH "-IC:\\Program Files (x86)\\Arduino\\hardware\\teensy\\avr\\cores\\teensy4" "-IC:\\Users\\felix\\Documents\\Arduino\\libraries\\micro_ros_arduino-2.0.3-foxy\\src" "C:\\Users\\felix\\Documents\\Arduino\\libraries\\micro_ros_arduino-2.0.3-foxy\\src\\default_transport.cpp" -o nul
...
```

But then it is not being linked against the final binary:

```
...
"C:\\Program Files (x86)\\Arduino\\hardware\\teensy/../tools/arm/bin/arm-none-eabi-gcc" -O2 -Wl,--gc-sections,--relax "-TC:\\Program Files (x86)\\Arduino\\hardware\\teensy\\avr\\cores\\teensy4/imxrt1062.ld" -mthumb -mcpu=cortex-m7 -mfloat-abi=hard -mfpu=fpv5-d16 -o "C:\\Users\\felix\\AppData\\Local\\Temp\\arduino_build_489404/sketch_feb28a.ino.elf" "C:\\Users\\felix\\AppData\\Local\\Temp\\arduino_build_489404\\sketch\\sketch_feb28a.ino.cpp.o" "C:\\Users\\felix\\AppData\\Local\\Temp\\arduino_build_489404/core\\core.a" "-LC:\\Users\\felix\\Documents\\Arduino\\libraries\\micro_ros_arduino-2.0.3-foxy\\src\\imxrt1062\\fpv5-d16-hard" -lmicroros "-LC:\\Users\\felix\\AppData\\Local\\Temp\\arduino_build_489404" -larm_cortexM7lfsp_math -lm -lstdc++
...

bodely97, 2022-03-03:
So after reinstalling everything most of the errors disappeared.
I get only one right now:

'D:\Dokumente Dateien Verknüpfungen Sonstiges\Dropbox\Dropbox\Dropbox\Freizeit\Projekte\MecanumBot\Arduino\Stepper_Teensy_Micro_ROS\main\main.ino:6:37: fatal error: geometry_msgs/msg/twist.h: No such file or directory'

Any ideas?

pablogs9, 2022-03-04:
Please reopen if you have any other issues!

closed 2022-03-04 by pablogs9