micro-ROS/micro_ros_arduino #791: uROS-arduino failingto build after update to 2.0.3

carmiac, 2022-03-02:
I've just tried to update to the 2.0.3 package and it's not building.
Full build log:
`
milner@canterbury:~/Arduino/libraries/micro_ros_arduino-2.0.3-foxy$ docker run -it --rm -v $(pwd):/project --env MICROROS_LIBRARY_FOLDER=extras microros/micro_ros_static_library_builder:foxy -p teensy4
Get:1 http://packages.ros.org/ros2/ubuntu focal InRelease [4673 B]
Get:2 http://archive.ubuntu.com/ubuntu focal InRelease [265 kB]                
Get:3 http://security.ubuntu.com/ubuntu focal-security InRelease [114 kB]
Get:4 http://packages.ros.org/ros2/ubuntu focal/main amd64 Packages [997 kB]             
Get:5 http://security.ubuntu.com/ubuntu focal-security/restricted amd64 Packages [994 kB]         
Get:6 http://archive.ubuntu.com/ubuntu focal-updates InRelease [114 kB]                        
Get:7 http://archive.ubuntu.com/ubuntu focal-backports InRelease [108 kB]                      
Get:8 http://security.ubuntu.com/ubuntu focal-security/universe amd64 Packages [844 kB]             
Get:9 http://archive.ubuntu.com/ubuntu focal/restricted amd64 Packages [33.4 kB]      
Get:10 http://archive.ubuntu.com/ubuntu focal/universe amd64 Packages [11.3 MB]                  
Get:11 http://security.ubuntu.com/ubuntu focal-security/multiverse amd64 Packages [25.8 kB]                              
Get:12 http://security.ubuntu.com/ubuntu focal-security/main amd64 Packages [1592 kB]                                     
Get:13 http://archive.ubuntu.com/ubuntu focal/multiverse amd64 Packages [177 kB]                                                                          
Get:14 http://archive.ubuntu.com/ubuntu focal/main amd64 Packages [1275 kB]                                                                               
Get:15 http://archive.ubuntu.com/ubuntu focal-updates/main amd64 Packages [2017 kB]                                                                       
Get:16 http://archive.ubuntu.com/ubuntu focal-updates/universe amd64 Packages [1136 kB]                                                                   
Get:17 http://archive.ubuntu.com/ubuntu focal-updates/multiverse amd64 Packages [29.4 kB]                                                                 
Get:18 http://archive.ubuntu.com/ubuntu focal-updates/restricted amd64 Packages [1063 kB]                                                                 
Get:19 http://archive.ubuntu.com/ubuntu focal-backports/main amd64 Packages [51.2 kB]                                                                     
Get:20 http://archive.ubuntu.com/ubuntu focal-backports/universe amd64 Packages [26.0 kB]                                                                 
Fetched 22.2 MB in 40s (557 kB/s)                                                                                                                         
Reading package lists... Done
Building dependency tree       
Reading state information... Done
163 packages can be upgraded. Run 'apt list --upgradable' to see them.
Creating firmware for generate_lib platform generic
.......
=== ./ament/ament_cmake (git) ===
Cloning into '.'...
=== ./ament/ament_index (git) ===
Cloning into '.'...
=== ./ament/ament_lint (git) ===
Cloning into '.'...
=== ./ament/ament_package (git) ===
Cloning into '.'...
=== ./ament/googletest (git) ===
Cloning into '.'...
=== ./ament/uncrustify_vendor (git) ===
Cloning into '.'...
=== ./ros2/ament_cmake_ros (git) ===
Cloning into '.'...
List of repositories is empty
#All required rosdeps installed successfully
..............
=== ./ros2/common_interfaces (git) ===
Cloning into '.'...
=== ./ros2/example_interfaces (git) ===
Cloning into '.'...
=== ./ros2/libyaml_vendor (git) ===
Cloning into '.'...
=== ./ros2/rcl (git) ===
Cloning into '.'...
=== ./ros2/rcl_interfaces (git) ===
Cloning into '.'...
=== ./ros2/rcl_logging (git) ===
Cloning into '.'...
=== ./ros2/rcpputils (git) ===
Cloning into '.'...
=== ./ros2/rmw (git) ===
Cloning into '.'...
=== ./ros2/rmw_implementation (git) ===
Cloning into '.'...
=== ./ros2/rosidl (git) ===
Cloning into '.'...
=== ./ros2/rosidl_dds (git) ===
Cloning into '.'...
[... 145 lines cut ...]
Starting >>> ament_lint_common
Finished <<< ament_lint_common [0.32s]                  
Starting >>> ament_cmake_ros
Starting >>> ament_index_cpp
Finished <<< ament_cmake_ros [0.37s]                                                               
Finished <<< ament_index_cpp [4.55s]                        

Summary: 60 packages finished [10.8s]
#All required rosdeps installed successfully
#All required rosdeps installed successfully
Cloning into 'geometry2'...
remote: Enumerating objects: 14412, done.
remote: Counting objects: 100% (1730/1730), done.
remote: Compressing objects: 100% (648/648), done.
remote: Total 14412 (delta 1257), reused 1369 (delta 997), pack-reused 12682
Receiving objects: 100% (14412/14412), 3.40 MiB | 3.68 MiB/s, done.
Resolving deltas: 100% (9836/9836), done.
=== ./control_msgs (git) ===
Cloning into '.'...
fatal: not a git repository (or any parent up to mount point /)
Stopping at filesystem boundary (GIT_DISCOVERY_ACROSS_FILESYSTEM not set).
No changes detected
milner@canterbury:~/Arduino/libraries/micro_ros_arduino-2.0.3-foxy$ 

`

Acuadros95, 2022-03-04:
This should be fixed with this PR: https://github.com/micro-ROS/micro_ros_arduino/pull/794
Please update the library with the lastest commit, will release the fix if the feedback is positive.

sejego, 2022-03-04:
@Acuadros95 Hey, not sure if I am doing it correctly, but I still have an issue, albeit a different one.

I have pulled the latest foxy version from github and added it as a library to Arduino IDE. Next, I tried running the same docker command but now it gives me a different error. Full log below:

```
sejego@pop-os:~/Arduino/libraries/micro_ros_arduino$ sudo docker run -it --rm -v $(pwd):/project --env MICROROS_LIBRARY_FOLDER=extras microros/micro_ros_static_library_builder:foxy
/entrypoint.sh: line 9: /project/extras/library_generation/library_generation.sh: Permission denied

```

Acuadros95, 2022-03-04:
@sejego you need to modify the permissions of the `library_generation.sh` file, use:
`sudo chmod a+x <Your path>/library_generation.sh`

sejego, 2022-03-04:
@carmiac @Acuadros95 The issue is fixed, was able to compile everything! Thanks a ton <3

closed 2022-03-04 by Acuadros95