micro-ROS/micro_ros_arduino #804: Rclc_support_init fails to initalize support

Salsus2191, 2022-03-04:
## Issue template

- Hardware description: Arduino Nano RP2040
- RTOS: 
- Installation type: 
- Version or commit hash: galactic v2.0.3

#### Steps to reproduce the issue
To reproduce the issue upload the code below to the Arduino. The platform used is PlatformIO.
```
#include <Arduino.h>

#include <micro_ros_arduino.h>

#include <rcl/error_handling.h>
#include <rcl/rcl.h>
#include <rclc/rclc.h>
#include <rclc/executor.h>

void setup()
{
    Serial.begin(115200);
    delay(10000);
    set_microros_transports();
    rcl_allocator_t allocator = rcl_get_default_allocator();
    rclc_support_t support;
    rcl_ret_t rc = rclc_support_init(&support, 0, NULL, &allocator);
    if (rc != RCL_RET_OK)
    {
        Serial.print("Could not initialize the support. Error: ");
        Serial.print(rc);
        Serial.print("\n");
    }
}

void loop()
{
    // put your main code here, to run repeatedly:
}
```

#### Expected behavior

The code executes correctly.

#### Actual behavior

The support can not be initialized and the init function returns error code 1.

#### Additional information

There are some weird things going on here. First, if i exclude the `set_microros_transports()` function, then the init function causes the entire system to crash. The Arduino starts drawing too much power which disables all usb ports. Second, I already successfully ran some microros code on my Arduino, but for some reason it won't work now. I also tried switching out the Arduino in case it had some hardware damage but other Arduino showed the same behaviour.

pablogs9, 2022-03-04:
Which is the output of the micro-ROS agent with `-v6` parameter?

Acuadros95, 2022-03-04:
`set_microros_transports()` uses the `Serial` instance to communicate with the agent, so take care using it to debug the code.

Also, you may interfere with the agent communications checking the serial port data.

Salsus2191, 2022-03-04:
The platformIO.ini file looks like this:

```
[env:nanorp2040connect]
platform = raspberrypi
board = nanorp2040connect
framework = arduino
extra_scripts = fix_linker.py
build_flags = 
	-L ./.pio/libdeps/nanorp2040connect/micro_ros_arduino/src/cortex-m0plus/
	-D ARDUINO_NANO_RP2040_CONNECT
lib_deps = 
	arduino-libraries/WiFiNINA@^1.8.13
	https://github.com/micro-ROS/micro_ros_arduino
platform_packages = 
	toolchain-gccarmnoneeabi @ ~1.70201.0
	framework-arduino-mbed @ ~2.4.1
lib_ldf_mode = chain+
```

The added python script looks like this:

```
Import("env")

env["_LIBFLAGS"] =  ('-Wl,--start-group -Wl,--whole-archive '
                    '${_stripixes(LIBLINKPREFIX, LIBS, LIBLINKSUFFIX, LIBPREFIXES, LIBSUFFIXES, __env__)} '
                    '-Wl,--no-whole-archive -lstdc++ -lsupc++ -lm -lc -lgcc -lnosys -lmicroros -Wl,--end-group')
```

And the agent output is:
[image]

Acuadros95, 2022-03-04:
The agent output looks fine in that image, `rclc_support_init` should be succeeding.
Try with a full example

Salsus2191, 2022-03-04:
I tried the included Soundmeter node example yesterday and ran into the same problem. The program would go into the erro_loop(). I ran it now and for some reason it worked. The only difference is that the agent was running while I uploaded the program. Does the agent always have to be running?

pablogs9, 2022-03-04:
I'm closing, please reopen if you have further questions.

closed 2022-03-04 by pablogs9

Salsus2191, 2022-03-04:
Thank you very much. I never had the client open as I didn't think it was necessary.