micro-ROS/micro_ros_arduino #815: Multiple services not executed properly

Avinashmuthukumarasamy, 2022-03-07:
## Issue template

- Hardware description: Teesny 3.2
- Installation type: micro_ros_arduino
- Version or commit hash: foxy

#### Steps to reproduce the issue
the rclc node needs to have multiple services and respective callback to perform specific tasks for each call. Hence I created multiple services and added to the rclc node.the code executed successfully but only he first service is visible while listing and also it can be called ,in turn executing successfully.The rest of the services are not available.

```
#include <Arduino.h>
#include <micro_ros_arduino.h>
#include <custom_interfaces/srv/task.h>
#include <stdio.h>
#include <rcl/error_handling.h>
#include <rclc/rclc.h>
#include <rclc/executor.h>
#include <std_msgs/msg/int64.h>
#include "TeensyStep.h"
#include <std_msgs/msg/string.h>
rcl_node_t node;
rclc_support_t support;
rcl_allocator_t allocator;
rclc_executor_t executor;

rcl_service_t Service1,Service2,Service3;
rcl_wait_set_t wait_set;

custom_interfaces__srv__Task_Response res;
custom_interfaces__srv__Task_Request req;

#define RCCHECK(fn) { rcl_ret_t temp_rc = fn; if((temp_rc != RCL_RET_OK)){while(1){};}}
#define RCSOFTCHECK(fn) { rcl_ret_t temp_rc = fn; if((temp_rc != RCL_RET_OK)){}}

//static variables declartion

void callback1(const void * req, void * res){
  custom_interfaces__srv__Task_Request * req_in = (custom_interfaces__srv__Task_Request *) req;
  custom_interfaces__srv__Task_Response * res_in = (custom_interfaces__srv__Task_Response *) res;
//logic 

}

void callback2(const void * req, void * res){
  custom_interfaces__srv__Task_Request * req_in = (custom_interfaces__srv__Task_Request *) req;
  custom_interfaces__srv__Task_Response * res_in = (custom_interfaces__srv__Task_Response *) res;
//logic
}

void callback3(const void * req, void * res){ 
//logic 

}

void setup() {
  set_microros_transports();
  delay(1000);

  allocator = rcl_get_default_allocator();

  // create init_options
  RCCHECK(rclc_support_init(&support, 0, NULL, &allocator));

  // create node
  RCCHECK(rclc_node_init_default(&node, "sampleNode", "", &support));

  // create service
  RCCHECK(rclc_service_init_default(&Service1, &node, ROSIDL_GET_SRV_TYPE_SUPPORT(custom_interfaces, srv,Task), "/Services1"));
  RCCHECK(rclc_service_init_default(&Service2, &node, ROSIDL_GET_SRV_TYPE_SUPPORT(custom_interfaces, srv, Task), "/Services2"));
  RCCHECK(rclc_service_init_default(&Service3, &node, ROSIDL_GET_SRV_TYPE_SUPPORT(custom_interfaces, srv, Task), "/Services3"));

  // create executor
  RCCHECK(rclc_executor_init(&executor, &support.context, 1, &allocator));
  RCCHECK(rclc_executor_add_service(&executor, &Service1, &req, &res, callback1));
  RCCHECK(rclc_executor_add_service(&executor, &Service2, &req, &res, callback2));
  RCCHECK(rclc_executor_add_service(&executor, &Service3, &req, &res, callback3));
}

void loop() {

  RCSOFTCHECK(rclc_executor_spin_some(&executor, RCL_MS_TO_NS(100)));
}

```

Please let me know if I am missing any steps.

#### Expected behavior
successful execution of all the services 

#### Actual behavior

while listing:
```
root@pi:/# ros2 service list 
/Services1

```
other services are not visible !

#### Additional information

pablogs9, 2022-03-07:
You are creating an executor with 1 [handle](https://github.com/ros2/rclc/blob/3e5e219ebb5f2c7b3da74f5838349a7969676c81/rclc/include/rclc/executor.h#L127) `RCCHECK(rclc_executor_init(&executor, &support.context, 1, &allocator));` and adding 3 services.

closed 2022-03-07 by pablogs9

Avinashmuthukumarasamy, 2022-03-23:
Hi @pablogs9 ,

Can you let me know how to recompile the library to make things happen for multiple services?

pablogs9, 2022-03-23:
[Here there are some instructions](https://github.com/micro-ROS/micro_ros_arduino#how-to-build-the-precompiled-library)

Avinashmuthukumarasamy, 2022-03-24:
It works well.Thanks @pablogs9 

For reference:
changed the contents of the colcon_lowmem.meta file[File location:/home/pi/Documents/PlatformIO/Projects/Teensy/.pio/libdeps/teensy36/micro_ros_arduino/extras/library_generation/colcon_lowmem.meta]  for multiple services and then recompiled the library as given in the instructions[refer attached image]. And the services are  visible.
```
{
    "names": {
        "tracetools": {
            "cmake-args": [
                "-DTRACETOOLS_DISABLED=ON",
                "-DTRACETOOLS_STATUS_CHECKING_TOOL=OFF"
            ]
        },
        "rosidl_typesupport": {
            "cmake-args": [
                "-DROSIDL_TYPESUPPORT_SINGLE_TYPESUPPORT=ON"
            ]
        },
        "rcl": {
            "cmake-args": [
                "-DBUILD_TESTING=OFF",
                "-DRCL_COMMAND_LINE_ENABLED=OFF",
                "-DRCL_LOGGING_ENABLED=OFF"
            ]
        }, 
        "rcutils": {
            "cmake-args": [
                "-DENABLE_TESTING=OFF",
                "-DRCUTILS_NO_FILESYSTEM=ON",
                "-DRCUTILS_NO_THREAD_SUPPORT=ON",
                "-DRCUTILS_NO_64_ATOMIC=ON",
                "-DRCUTILS_AVOID_DYNAMIC_ALLOCATION=ON"
            ]
        },
        "microxrcedds_client": {
            "cmake-args": [
                "-DUCLIENT_PIC=OFF",
                "-DUCLIENT_PROFILE_UDP=OFF",
                "-DUCLIENT_PROFILE_TCP=OFF",
                "-DUCLIENT_PROFILE_DISCOVERY=OFF",
                "-DUCLIENT_PROFILE_SERIAL=OFF",
                "-UCLIENT_PROFILE_STREAM_FRAMING=ON",
                "-DUCLIENT_PROFILE_CUSTOM_TRANSPORT=ON"
            ]
        },
        "rmw_microxrcedds": {
            "cmake-args": [
                "-DRMW_UXRCE_MAX_NODES=1",
                "-DRMW_UXRCE_MAX_PUBLISHERS=10",
                "-DRMW_UXRCE_MAX_SUBSCRIPTIONS=5",
                "-DRMW_UXRCE_MAX_SERVICES=4",
                "-DRMW_UXRCE_MAX_CLIENTS=3",
                "-DRMW_UXRCE_MAX_HISTORY=1",
                "-DRMW_UXRCE_TRANSPORT=custom"
            ]
        }
    }
}
```

complied the library using this command
`docker run -it --rm -v /home/pi/Documents/PlatformIO/Projects/Teensy --env MICROROS_LIBRARY_FOLDER=extras microros/micro_ros_static_library_builder:foxy -p teensy36`

[image]

Output:
[image]

scch1092, 2022-04-22:
> docker run -it --rm -v /home/pi/Documents/PlatformIO/Projects/Teensy --env MICROROS_LIBRARY_FOLDER=extras microros/micro_ros_static_library_builder:foxy -p teensy36

Hi, i have a question about it. I tried to follow your steps and found the file (in my case "colcon.meta" for teensy 4.1) and made my changes:
`docker run -it --rm -v /home/scch1092/Documents/PlatformIO/Projects/services --env MICROROS_LIBRARY_FOLDER=extras microros/micro_ros_static_library_builder:galactic -p teensy4`

But if i try to run docker, i get an error: 
`/entrypoint.sh: line 9: /project/extras/library_generation/library_generation.sh: No such file or directory`

Is it something to do with the build flags or that I have a dependency on a GitHub repo?

by build flags:
`build_flags =
    -L ./.pio/libdeps/teensy41/micro_ros_arduino/src/imxrt1062/fpv5-d16-hard/
    -l microros
    -D ARDUINO_TEENSY41`