micro-ROS/micro_ros_arduino #819: Error compiling example for Arduino DUE

BorgesJVT, 2022-03-09:
## Issue template

- Hardware description: Arduino DUE
- RTOS: NO
- Installation type: using static library
- Version or commit hash: micro_ros_arduino-2.0.3-foxy into Arduino 1.8.19 IDE's libraries directory.

#### Steps to reproduce the issue
Install micro_ros_arduino-2.0.3-foxy into Arduino 1.8.19 IDE's libraries directory.
Go to File > Examples > micro_ros_arduino > micro-ros_publisher
Try to compile the project.

#### Expected behavior
To finish compilation successfully, and get uploaded. Run the micro-ros_publisher as it is.

#### Actual behavior
Arduino: 1.8.19 (Linux), Board: "Arduino Due (Programming Port)"

Library micro_ros_arduino-2.0.3-foxy has been declared precompiled:
Using precompiled library in /home/borgesjvt/Arduino/libraries/micro_ros_arduino-2.0.3-foxy/src/cortex-m3
The platform does not support 'compiler.libraries.ldflags' for precompiled libraries.
sketch/micro-ros_publisher.ino.cpp.o: In function `timer_callback(rcl_timer_t*, long long)':
/home/borgesjvt/Arduino/libraries/micro_ros_arduino-2.0.3-foxy/examples/micro-ros_publisher/micro-ros_publisher.ino:36: undefined reference to `rcl_publish'
sketch/micro-ros_publisher.ino.cpp.o: In function `set_microros_transports':
/home/borgesjvt/Arduino/libraries/micro_ros_arduino-2.0.3-foxy/src/micro_ros_arduino.h:32: undefined reference to `rmw_uros_set_custom_transport'
sketch/micro-ros_publisher.ino.cpp.o: In function `setup':
/home/borgesjvt/Arduino/libraries/micro_ros_arduino-2.0.3-foxy/examples/micro-ros_publisher/micro-ros_publisher.ino:49: undefined reference to `rcutils_get_default_allocator'
/home/borgesjvt/Arduino/libraries/micro_ros_arduino-2.0.3-foxy/examples/micro-ros_publisher/micro-ros_publisher.ino:52: undefined reference to `rclc_support_init'
/home/borgesjvt/Arduino/libraries/micro_ros_arduino-2.0.3-foxy/examples/micro-ros_publisher/micro-ros_publisher.ino:55: undefined reference to `rclc_node_init_default'
/home/borgesjvt/Arduino/libraries/micro_ros_arduino-2.0.3-foxy/examples/micro-ros_publisher/micro-ros_publisher.ino:58: undefined reference to `rosidl_typesupport_c__get_message_type_support_handle__std_msgs__msg__Int32'
/home/borgesjvt/Arduino/libraries/micro_ros_arduino-2.0.3-foxy/examples/micro-ros_publisher/micro-ros_publisher.ino:58: undefined reference to `rclc_publisher_init_default'
/home/borgesjvt/Arduino/libraries/micro_ros_arduino-2.0.3-foxy/examples/micro-ros_publisher/micro-ros_publisher.ino:66: undefined reference to `rclc_timer_init_default'
/home/borgesjvt/Arduino/libraries/micro_ros_arduino-2.0.3-foxy/examples/micro-ros_publisher/micro-ros_publisher.ino:73: undefined reference to `rclc_executor_init'
/home/borgesjvt/Arduino/libraries/micro_ros_arduino-2.0.3-foxy/examples/micro-ros_publisher/micro-ros_publisher.ino:74: undefined reference to `rclc_executor_add_timer'
sketch/micro-ros_publisher.ino.cpp.o: In function `loop':
/home/borgesjvt/Arduino/libraries/micro_ros_arduino-2.0.3-foxy/examples/micro-ros_publisher/micro-ros_publisher.ino:81: undefined reference to `rclc_executor_spin_some'
collect2: error: ld returned 1 exit status
exit status 1
Error compiling for board Arduino Due (Programming Port).

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

#### Additional information
I have put the  SAM patch like mentioned:
```
export ARDUINO_PATH=[Your Arduino path]
cd $ARDUINO_PATH/hardware/sam/1.6.12/
curl https://raw.githubusercontent.com/micro-ROS/micro_ros_arduino/foxy/extras/patching_boards/platform_arduinocore_sam.txt > platform.txt
```
I've also seen other issues like #427 and #429 looking for insights of the problem but seems to be all like instructed.

pablogs9, 2022-03-10:
Hello @BorgesJVT, this is a problem with the SAM patch because the build process is complaining of `The platform does not support 'compiler.libraries.ldflags' for precompiled libraries.`. 

In fact, those patches includes [this ldflags](https://github.com/micro-ROS/micro_ros_arduino/blob/7807fc493348c5a430709a3366521ecd57d1b59e/extras/patching_boards/platform_arduinocore_sam.txt#L50) and add them [here also](https://github.com/micro-ROS/micro_ros_arduino/blob/7807fc493348c5a430709a3366521ecd57d1b59e/extras/patching_boards/platform_arduinocore_sam.txt#L82).

So make sure that you are replacing the correct file, should modify [this one](https://github.com/arduino/ArduinoCore-sam/blob/master/platform.txt)

BorgesJVT, 2022-03-10:
Hello @pablogs9, I checked again and the platform.txt file is correct, but still the same result.
That's my path _/home/borgesjvt/Downloads/arduino-1.8.19-linux64/arduino-1.8.19/hardware/arduino/avr/platform.txt._
At first I had installed Arduino with `snap`, but I was not able to edit the platform.txt file since there was some kind of mounted file system with snap.
I've also tested with foxy-2.0.0 version.

I don't know if it gives some insights of the problem. Any suggestions?

pablogs9, 2022-03-11:
Let us replicate the environment.

SpMakris, 2022-03-17:
I am trying to do the same thing, and run into this issue as well. My code now compiles correctly, although I have not uploaded or tested the actually application yet.
The path you mention contains the AVR core files, but not the SAMD ones (for whatever reason). You can find the correct path through the arduino IDE:
1. Open the Arduino IDE
2. File>Prefernces
3. At the bottom of this window you will see a note "More preferences can be edited directly in the file", followed by a link to a folder on your computer. 
4. Click the link. A folder will open. It should contain a folder named "Packages", among other things.
5. Navigate to Packages>Arduino. This is the actual path you need to give to the above commands to properly apply the patch. You can verify it by navigating to Packages>Arduino>Hardware. You should see a folder named SAMD inside.

There might be a better way to figure out the correct path, but this should do the trick.

pablogs9, 2022-03-17:
@BorgesJVT, does the @SpMakris solutions work for you?

BorgesJVT, 2022-03-23:
Thank you @SpMakris and @pablogs9. Yes, it worked. When changing the platform.txt file inside this path (/home/borgesjvt/.arduino15/packages/arduino/hardware/sam/1.6.12) it works fine.

closed 2022-03-23 by pablogs9

karimrahimi76, 2024-11-02:
> Thank you @SpMakris and @pablogs9. Yes, it worked. When changing the platform.txt file inside this path (/home/borgesjvt/.arduino15/packages/arduino/hardware/sam/1.6.12) it works fine.

hi , could you please share the exact file that you used and got no error 
i have the same problem

hippo5329, 2024-11-02:
Better switch to vscode and platformio.
Follow the wiki, https://github.com/hippo5329/micro_ros_arduino_examples_platformio/wiki

    pio run -e due

karimrahimi76, 2024-11-04:
thank you thomas
it worked fine