micro-ROS/micro_ros_arduino #821: Possible bug with Header message type

pnn16004, 2022-03-10:
## Issue template

- Hardware description: Arduino Due, ESP32 Feather
- Installation type: micro-ROS library for Arduino, PlatformIO, VS Code
- Version or commit hash: Galactic

#### Steps to reproduce the issue
Implement the [micro-ros_subscriber](https://github.com/micro-ROS/micro_ros_arduino/tree/galactic/examples/micro-ros_subscriber) example and change message type to `std_msgs/msg/header`.

Implement the [ROS2 C++ publisher](https://raw.githubusercontent.com/ros2/examples/foxy/rclcpp/topics/minimal_publisher/member_function.cpp) example and change message type to `std_msgs/msg/header`.

Have them communicate through the micro-ROS Agent on a topic.

#### Expected behavior
Reach the `subscription_callback()` function when debugging or see with serial monitor that it has been reached.

#### Actual behavior
No sign that it reaches the `subscription_callback()` function but the Agent reacts to the publisher.
No messages from `printf()` inside the function and only one response from the `printf` inside `loop()`.
The debugging goes into some memory files after `rclc_executor_spin_some()` and never returns to the main code.
These results are for both Arduino Due and ESP32 Feather.
Agent output:
[image]

#### Additional information
I implemented ping pong peng applications in micro-ROS with Arduino, ESP-IDF, and STM32CubeIDE.
Using the Header message type, the ping side works with all of them, but the pong side didn't work in Arduino, even though the code functionalities were identical.
Thus I tried with these basic examples and it doesn't work either, so I suspect it might be the message type, as I tried the same thing above but with Int32 and it worked.
Possibly related to [#614](https://github.com/micro-ROS/micro_ros_arduino/issues/614).

pablogs9, 2022-03-11:
- Could you share your code?
- Are you using `printf()` in the same serial port that micro-ROS is using as transport?
- If you are using ROS 2 Galactic in your ROS 2 side, have you switched to Fast DDS as the default middleware implementation?

pnn16004, 2022-03-11:
Yes, here: [main.txt](https://github.com/micro-ROS/micro_ros_arduino/files/8229806/main.txt), [min_sub.txt], (https://github.com/micro-ROS/micro_ros_arduino/files/8229809/min_sub.txt).

Yes, I know it won't work as normal, but at least with other applications I get some sort of output from the serial monitor.

No, I use the default.

pablogs9, 2022-03-11:
1. In main.txt you are not initing the memory inside `msg`. Check [here](https://micro.ros.org/docs/tutorials/advanced/handling_type_memory/). You need some memory inside `msg.frame_id` for example:

```c
std_msgs__msg__Header msg;
msg.frame_id.data = (char *) malloc(sizeof(char)*100);
msg.frame_id.size = 0;
msg.frame_id.capacity = 100;
```

2. You should not use the serial port micro-ROS to print stuff, it will mess with the agent. **But more important, you should not open the same serial port in two applications (i.e. the agent and any kind of serial software) because the agent won't receive all data.**

3. Is highly recommended to use Fast DDS with micro-ROS. Other ROS 2 DDS implementations are not guaranteed to work.

pnn16004, 2022-03-12:
Oh I forgot that, I have it in my original project. Tested the example and it works now, so the error is not with Header. 

Then I have no idea why it doesn't work in the other application, because I use the exact same setup in micro-ROS for ESP-IDF and STM32CubeIDE and that works.

Here is the code for my implementation if you could look at `setup()` and see if there is anything wrong: [main.txt](https://github.com/micro-ROS/micro_ros_arduino/files/8236700/main.txt), [test_ping.txt](https://github.com/micro-ROS/micro_ros_arduino/files/8236704/test_ping.txt)

closed 2022-03-14 by pablogs9