micro-ROS/micro_ros_arduino #829: Service example doesnt show up in service list, and isnt callable.

tobratland, 2022-03-15:
## Issue template

- Hardware description: Arduino nano rp2040 connect

- Installation type:  tried both through compiling with docker image, and using zip file
- Version or commit hash: Galactic

#### Steps to reproduce the issue
When trying to get service example to work on my arduino it compiles and transfers fine, but when I try to list services nothing shows up. 

#### Expected behavior
The "addtwoints" service should show up on ros2 service list, and the service should be callable.

#### Actual behavior
No service shows up in list.

#### Additional information
I have built a custom msg/srv package, and have had success publishing a topic with messages from this custom service. When i add a service to this node however, the topic isnt published on anymore. As mentioned earlier I have also tested with a zip directly from releases, with the add two ints service example, with no luck.

pablogs9, 2022-03-16:
Have you switched to Fast DDS as middleware in ROS 2 Galactic? micro-ROS is not compatible with Cyclone DDS (the default middleware for Galactic).

https://docs.ros.org/en/galactic/Installation/DDS-Implementations/Working-with-eProsima-Fast-DDS.html

tobratland, 2022-03-16:
Yes, i have changed to Fast DDS, still nothing shows up in service list.

tobratland, 2022-03-16:
#include <micro_ros_arduino.h>

#include <stdio.h>
#include <rcl/rcl.h>
#include <rcl/error_handling.h>
#include <rclc/rclc.h>
#include <rclc/executor.h>

#include <ductbot_messages/msg/drive_motor.h>
#include <ductbot_messages/msg/endpoint_status.h>
#include <ductbot_messages/srv/drive_motor_control.h>

enum Direction
{
  clockWize,
  counterClockWize
};

struct DriveMotorStruct
{
  int enable_pin;
  int pin1;
  int pin2;
  int currentSpeed;
  int requestedSpeed;
  Direction dir;
};

DriveMotorStruct driveMotors[4];

rcl_publisher_t publisher;
ductbot_messages__msg__DriveMotor driveMotorMsg;
ductbot_messages__msg__EndpointStatus statusMsg;

ductbot_messages__srv__DriveMotorControl_Request driveMotorReq;
ductbot_messages__srv__DriveMotorControl_Response driveMotorRes;

rclc_executor_t executor;
rclc_support_t support;
rcl_allocator_t allocator;
rcl_node_t node;
rcl_timer_t timer;
rcl_service_t service;

static char res_message = '\0';

#define LED_PIN 13

#define RCCHECK(fn)                                                                                                    \
  {                                                                                                                    \
    rcl_ret_t temp_rc = fn;                                                                                            \
    if ((temp_rc != RCL_RET_OK))                                                                                       \
    {                                                                                                                  \
      error_loop();                                                                                                    \
    }                                                                                                                  \
  }
#define RCSOFTCHECK(fn)                                                                                                \
  {                                                                                                                    \
    rcl_ret_t temp_rc = fn;                                                                                            \
    if ((temp_rc != RCL_RET_OK))                                                                                       \
    {                                                                                                                  \
    }                                                                                                                  \
  }

void error_loop()
{
  while (1)
  {
    digitalWrite(LED_PIN, !digitalRead(LED_PIN));
    delay(100);
[... 162 lines cut ...]
  /* RCCHECK(rclc_service_init_default(&service, &node,
                                    ROSIDL_GET_SRV_TYPE_SUPPORT(ductbot_messages, srv, DriveMotorControl),
                                    "/front_drive_motor_control")); */

  RCCHECK(rclc_timer_init_default(&timer, &support, RCL_MS_TO_NS(timer_timeout), timer_callback));
  delay(2000);
  // create executor
  RCCHECK(rclc_executor_init(&executor, &support.context, 2, &allocator));
  delay(2000);
  RCCHECK(rclc_executor_add_timer(&executor, &timer));
  driveMotorRes.message.data = &res_message;
  driveMotorRes.message.size = 1;
  driveMotorRes.message.capacity = 1; 
  //RCCHECK(rclc_executor_add_service(&executor, &service, &driveMotorReq, &driveMotorRes, drive_motor_callback));
}

void loop()
{
  delay(100);
  controlMotorSpeed();
  updateStatusMsg();
  RCSOFTCHECK(rclc_executor_spin_some(&executor, RCL_MS_TO_NS(100)));
}

Attaching code for disection. When i have the two lines for the service commented out, it publishes on the topic(ductbot_status), but when i add the service there is no activity on the topic either.

tobratland, 2022-03-16:
Using the built in pin i can see that the example code goes directly into an error loop. I'm not sure how i can "log out" whatever the error is, as i understand micro-ROS and the serial package doesnt play nice?

pablogs9, 2022-03-17:
Can you share your types? Are you initializing the memory correctly?

tobratland, 2022-03-17:
The service is defined as 
```
ductbot_messages/DriveMotor motor
---
bool success
string message
```

DriveMotor message:
```
int8 motor_id
int16 motor_speed
```
I am not 100% sure im initializing memory correctly, but as i mentioned before, the addtwoints example from the examples folder also fails in the same maner.

pablogs9, 2022-03-17:
I would suggest handling better the string inside the response message. For example:

```c
void drive_motor_callback(const void* req, void* res)
{
...
    snprintf(res_in.message.data, res_in.message.capacity, "blablabla\n");
    res_in.message.size = strlen(res_in.message.data);
...
}

void setup() {
....
  static char buffer[100];
  driveMotorRes.message.data = buffer;
  driveMotorRes.message.size = 0;
  driveMotorRes.message.capacity = 100; 
  RCCHECK(rclc_executor_add_service(&executor, &service, &driveMotorReq, &driveMotorRes, drive_motor_callback));
....
```

Does it behave better with this approach?

tobratland, 2022-03-17:
Thanks for your help this far, this sadly hasnt resolved the issue. The error loop is still reached. Not sure how to proceed in debugging, would appreciate any further help.

pablogs9, 2022-03-18:
I found it. Sorry I didn't realize that you were using Nano RP2040. This board is configured in this library with `colcon_verylowmem.meta` (check [here](https://github.com/micro-ROS/micro_ros_arduino#supported-boards)), and as you can see [here](https://github.com/micro-ROS/micro_ros_arduino/blob/1979daad2b92cd6620846a23106716f2dbca0224/extras/library_generation/colcon_verylowmem.meta#L46) it does not have configured services in the middleware layer.

Just rebuild the library using [these instructions](https://github.com/micro-ROS/micro_ros_arduino#how-to-build-the-precompiled-library) but modifying the `colcon_verylowmem.meta` to allocate static memory for the services you need. Let me know if you have any problem.

tobratland, 2022-03-18:
That did the trick, awesome!

Again, thanks a bunch for your great work, and for your quick and helpful responses.

closed 2022-03-18 by tobratland