micro-ROS/micro_ros_arduino #830: Eternal build loop when building with docker

tobratland, 2022-03-15:
## Issue template

- Hardware description: Arduino nano rp2040 connect
- Installation type:  build with docker
- Version or commit hash: Galactic, latest

#### Steps to reproduce the issue
When building the precompiled library the build completes with errors(Package 'rcutils' exports library 'dl' which couldn't be found). Then the build keeps looping withou an exit. It seems all the packages complete before loop restarts. 

#### Expected behavior
Build completes without error and exits.
#### Actual behavior
Build has errors, and never completes.
#### Additional information

pablogs9, 2022-03-16:
Please provide steps (commands) for replicating the issue.

Acuadros95, 2022-03-16:
The library is built multiple times for all the supported boards. So its not looping, its just building the library with different compilers/configurations and should eventually end.

You should be able to pass this arguments to docker, so the build only happens for certain boards: 
https://github.com/micro-ROS/micro_ros_arduino/blob/8eff9c2c435c72f10131d77fef8a98c49d5f676b/extras/library_generation/library_generation.sh#L12-L25

tobratland, 2022-03-16:
Thanks for the help, your solution worked!

closed 2022-03-16 by tobratland