micro-ROS/micro_ros_arduino #847: Portenta H7 on PlatformIO

amilcarlucas, 2022-03-17:
## Issue template

- Hardware description: Arduino Portenta H7 with machine control carrier board
- RTOS: mbed
- Installation type: copied a pre-compiled library into .pio/libdeps/portenta_h7_m7/micro_ros_arduino
I compiled it using:
```
export ROS_DISTRO=galactic
cd micro_ros_arduino
docker pull microros/micro_ros_static_library_builder:${ROS_DISTRO}
docker run -it --rm -v $(pwd):/project --env MICROROS_LIBRARY_FOLDER=extras microros/micro_ros_static_library_builder:${ROS_DISTRO}
cd ..
zip -r micro_ros_arduino-2.0.3-${ROS_DISTRO}.zip micro_ros_arduino
```
- Version or commit hash: latest galactic from today, with some custom messages added by me

#### Steps to reproduce the issue
I used the following `platform.ini`file to compile my code. The code compiles fine with arduino 1.8.5 BTW
```
[env:portenta_h7_m7]
platform = ststm32
board = portenta_h7_m7
framework = arduino
platform_packages =
  ; use GCC AVR < 5.0
  toolchain-gccarmnoneeabi@<1.50000.0
lib_deps = 
    arduino-libraries/Arduino_MachineControl@^1.1.0
build_flags =
    -L ./.pio/libdeps/portenta_h7_m7/micro_ros_arduino/src/cortex-m7/fpv5-d16-softfp/
    -l microros
    -D ARDUINO_PORTENTA_H7_M7
```

#### Expected behavior
Compiles without errors.

#### Actual behavior
Compiles fine but I get a ton of linking errors:
```
.pio/libdeps/portenta_h7_m7/micro_ros_arduino/src/cortex-m7/fpv5-d16-softfp/libmicroros.a(libexample_interfaces__rosidl_typesupport_introspection_c-char__type_support.c.obj):char__type_support.c:(.text.Char__rosidl_typesupport_introspection_c__Char_fini_function+0x0): first defined here
.pio/libdeps/portenta_h7_m7/micro_ros_arduino/src/cortex-m7/fpv5-d16-softfp/libmicroros.a(libstd_msgs__rosidl_typesupport_introspection_c-empty__type_support.c.obj): In function `Empty__rosidl_typesupport_introspection_c__Empty_init_function':
empty__type_support.c:(.text.Empty__rosidl_typesupport_introspection_c__Empty_init_function+0x0): multiple definition of `Empty__rosidl_typesupport_introspection_c__Empty_init_function'
.pio/libdeps/portenta_h7_m7/micro_ros_arduino/src/cortex-m7/fpv5-d16-softfp/libmicroros.a(libexample_interfaces__rosidl_typesupport_introspection_c-empty__type_support.c.obj):empty__type_support.c:(.text.Empty__rosidl_typesupport_introspection_c__Empty_init_function+0x0): first defined here
.pio/libdeps/portenta_h7_m7/micro_ros_arduino/src/cortex-m7/fpv5-d16-softfp/libmicroros.a(libstd_msgs__rosidl_typesupport_introspection_c-empty__type_support.c.obj): In function `Empty__rosidl_typesupport_introspection_c__Empty_fini_function':
empty__type_support.c:(.text.Empty__rosidl_typesupport_introspection_c__Empty_fini_function+0x0): multiple definition of `Empty__rosidl_typesupport_introspection_c__Empty_fini_function'
.pio/libdeps/portenta_h7_m7/micro_ros_arduino/src/cortex-m7/fpv5-d16-softfp/libmicroros.a(libexample_interfaces__rosidl_typesupport_introspection_c-empty__type_support.c.obj):empty__type_support.c:(.text.Empty__rosidl_typesupport_introspection_c__Empty_fini_function+0x0): first defined here
.pio/libdeps/portenta_h7_m7/micro_ros_arduino/src/cortex-m7/fpv5-d16-softfp/libmicroros.a(libstd_msgs__rosidl_typesupport_introspection_c-float32__type_support.c.obj): In function `Float32__rosidl_typesupport_introspection_c__Float32_init_function':
float32__type_support.c:(.text.Float32__rosidl_typesupport_introspection_c__Float32_init_function+0x0): multiple definition of `Float32__rosidl_typesupport_introspection_c__Float32_init_function'
.pio/libdeps/portenta_h7_m7/micro_ros_arduino/src/cortex-m7/fpv5-d16-softfp/libmicroros.a(libexample_interfaces__rosidl_typesupport_introspection_c-float32__type_support.c.obj):float32__type_support.c:(.text.Float32__rosidl_typesupport_introspection_c__Float32_init_function+0x0): first defined here
```

#### Additional information
I used the trick @Acuadros95 told to force an older GCC ABI. But it also fails without it.

Acuadros95, 2022-03-17:
You need to modify the linker script, check my answer on this issue: https://github.com/micro-ROS/micro_ros_arduino/issues/774.

amilcarlucas, 2022-03-17:
Your solution worked great, it should be added to the readme.md file.