micro-ROS/micro_ros_arduino #861: Service call failed

amilcarlucas, 2022-03-21:
## Issue template

- Hardware description: Arduino Portenta H7 on machine control carrier
- RTOS: mbed
- Installation type: micro_ros_arduino + custom messages and services
- Version or commit hash: latest master

#### Steps to reproduce the issue
Added custom messages to `extras/library_generation/extra_packages/` and compiled the static library with:
```bash
export ROS_DISTRO=galactic
cd micro_ros_arduino
cp -r ../../ros2_ws/src/emily_uc_messages extras/library_generation/extra_packages/
docker pull microros/micro_ros_static_library_builder:${ROS_DISTRO}
docker run -it --rm -v $(pwd):/project --env MICROROS_LIBRARY_FOLDER=extras microros/micro_ros_static_library_builder:${ROS_DISTRO} -p portenta-m7
```

Added the static library to the Platform IO IDE project. Compiled and downloaded to target.
Then run the agent on a docker:
```bash
docker run -it --rm -v /dev:/dev --privileged --net=host microros/micro-ros-agent:galactic serial --dev /dev/ttyACM0 -v6
```

And after that did:

```bash
sudo apt update
sudo apt install ros-galactic-rmw-fastrtps-cpp
export RMW_IMPLEMENTATION=rmw_fastrtps_cpp
cd ../ros2_ws/
. /opt/ros/galactic/setup.bash
colcon build --packages-select emily_uc_messages --allow-overriding emily_uc_messages
. install/setup.bash
```
 and got:
 ```
# ros2 interface show emily_uc_messages/srv/Gripper 
# coordinator -> µcontroller - gripper service request
bool close_gripper

---

# coordinator -> µcontroller - gripper service response
bool gripper_closed
# ros2 service list
/gripper
# ros2 service call /gripper emily_uc_messages/srv/Gripper close_gripper:\ true\
failed to create client: Type support not from this implementation. Got:
    error not set
    error not set
while fetching it, at /tmp/binarydeb/ros-galactic-rmw-fastrtps-cpp-5.0.1/src/rmw_client.cpp:120, at /tmp/binarydeb/ros-galactic-rcl-3.1.2/src/rcl/client.c:111
```

#### Expected behavior
I expected to see three services on the list:
```
/gripper
/load_cell
/semaphore_lights
```
But see only one.

And I expected the service call not to fail

#### Actual behavior

Only one service listed.
Service call failed.

Acuadros95, 2022-03-21:
As listed on the `Suported boards` table, Arduino Portenta H7 uses the `colcon.meta` file to configure the expected memory use.
As you can see [here](https://github.com/micro-ROS/micro_ros_arduino/blob/galactic/extras/library_generation/colcon.meta#L46), only 1 service is allowed.

Modify this file with your expected entities and recompile the library. You can checkout the documentation for this options [here](https://github.com/micro-ROS/rmw_microxrcedds#rmw-micro-xrce-dds-implementation).

pablogs9, 2022-03-21:
Are you initializing the memory in your custom msgs? Could you share `messages/Gripper`, `emily_uc_messages/LoadCell` and `emily_uc_messages/SemaphoreLights` ?

Also this error:
```
failed to create client: Type support not from this implementation. Got:
   error not set
   error not set
while fetching it, at /tmp/binarydeb/ros-galactic-rmw-fastrtps-cpp-5.0.1/src/rmw_client.cpp:120, at /tmp/binarydeb/ros-galactic-rcl-3.1.2/src/rcl/client.c:111
```

seems to be a lack of type support for your custom service types in the ROS 2 side. Are you building and sourcing correctly those type packages?

pablogs9, 2022-03-21:
Also, please share the agent output using `-v6` to check what is happening with the micro-ROS service initialization.

amilcarlucas, 2022-03-21:
```ros
[1647869912.916073] info     | TermiosAgentLinux.cpp | init                     | running...             | fd: 3
[1647869914.486839] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x477CB297, len: 24, data: 
0000: 80 00 00 00 00 01 10 00 58 52 43 45 01 00 01 0F 58 43 5B 38 81 00 FC 01
[1647869914.489792] info     | Root.cpp           | delete_client            | delete                 | client_key: 0x477CB297
[1647869914.489880] info     | SessionManager.hpp | destroy_session          | session closed         | client_key: 0x477CB297, address: 0
[1647869914.489932] info     | Root.cpp           | create_client            | create                 | client_key: 0x58435B38, session_id: 0x81
[1647869914.489960] info     | SessionManager.hpp | establish_session        | session established    | client_key: 0x58435B38, address: 0
[1647869914.490077] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x58435B38, len: 19, data: 
0000: 81 00 00 00 04 01 0B 00 00 00 58 52 43 45 01 00 01 0F 00
[1647869914.490356] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x58435B38, len: 56, data: 
0000: 81 80 00 00 01 07 2E 00 00 0A 00 01 01 03 00 00 1F 00 00 00 00 01 CC CC 17 00 00 00 6D 69 63 72
0020: 6F 5F 72 6F 73 5F 61 72 64 75 69 6E 6F 5F 6E 6F 64 65 00 00 00 00 00 00
[1647869914.495674] info     | ProxyClient.cpp    | create_participant       | participant created    | client_key: 0x58435B38, participant_id: 0x000(1)
[1647869914.495831] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x58435B38, len: 14, data: 
0000: 81 80 00 00 05 01 06 00 00 0A 00 01 00 00
[1647869914.495893] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x58435B38, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 01 00 00 00 80
[1647869914.496032] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x58435B38, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 01 00 00 00 80
[1647869914.496516] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x58435B38, len: 192, data: 
0000: 81 80 01 00 01 07 B6 00 00 0B 00 08 08 03 00 00 A8 00 00 00 09 00 00 00 2F 67 72 69 70 70 65 72
0020: 00 00 00 00 2F 00 00 00 65 6D 69 6C 79 5F 75 63 5F 6D 65 73 73 61 67 65 73 3A 3A 73 72 76 3A 3A
0040: 64 64 73 5F 3A 3A 47 72 69 70 70 65 72 5F 52 65 71 75 65 73 74 5F 00 08 30 00 00 00 65 6D 69 6C
0060: 79 5F 75 63 5F 6D 65 73 73 61 67 65 73 3A 3A 73 72 76 3A 3A 64 64 73 5F 3A 3A 47 72 69 70 70 65
0080: 72 5F 52 65 73 70 6F 6E 73 65 5F 00 01 C5 00 24 12 00 00 00 72 71 2F 67 72 69 70 70 65 72 52 65
00A0: 71 75 65 73 74 00 01 08 10 00 00 00 72 72 2F 67 72 69 70 70 65 72 52 65 70 6C 79 00 00 01 00 00
[1647869914.497922] info     | ProxyClient.cpp    | create_replier           | replier created        | client_key: 0x58435B38, requester_id: 0x000(7), participant_id: 0x000(1)
[1647869914.498044] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x58435B38, len: 14, data: 
0000: 81 80 01 00 05 01 06 00 00 0B 00 08 00 00
[1647869914.498090] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x58435B38, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 02 00 00 00 80
[1647869914.498237] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x58435B38, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 02 00 00 00 80
```

amilcarlucas, 2022-03-21:
On the ROS2 side the services seam to work fine:

```bash
# ros2 interface show emily_uc_messages/srv/Gripper 
# coordinator -> µcontroller - gripper service request
bool close_gripper

---

# coordinator -> µcontroller - gripper service response
bool gripper_closed

# ros2 interface show emily_uc_messages/srv/LoadCell
# coordinator -> µcontroller - load_cell service request

# operation mode/request
uint8 OPERATION_MODE_GET_WEIGHT=0
uint8 OPERATION_MODE_CALIBRATION_EMPTY=1
uint8 OPERATION_MODE_CALIBRATION_LOADED=2
uint8 operation_mode

---

# µcontroller -> coordinator - load_cell service response

# weight in [g]
float32 weight

# error/status flags
uint8 ERROR_FLAGS_NONE=0
uint8 ERROR_FLAGS_CALIBRATION=1
uint8 ERROR_FLAGS_OVERTEMPERATURE=2
uint8 error_flags

# ros2 interface show emily_uc_messages/srv/SemaphoreLights 
# coordinator -> µcontroller - semaphore lights service request
uint8 BLINK_OFF=0
uint8 BLINK_NO=1
uint8 BLINK_SLOW=2
uint8 BLINK_FAST=3

# each light can have a different blink status
uint8 green_ctrl
uint8 yellow_ctrl
uint8 red_ctrl

---

# µcontroller -> coordinator - semaphore lights service response
uint8 green_status
uint8 yellow_status
uint8 red_status
```

pablogs9, 2022-03-21:
It seems that the client is creating only the `Gripper` service. Have you regenerated correctly the library with slots for 3 services?

Also, where does `printf` prints in your code? Make sure that it is not using the same port as micro-ROS client uses.

amilcarlucas, 2022-03-21:
looks like I did not clean before building the code. I now get all three services listed. But the error message persists:
```
failed to create client: Type support not from this implementation. Got:
    error not set
    error not set
while fetching it, at /tmp/binarydeb/ros-galactic-rmw-fastrtps-cpp-5.0.1/src/rmw_client.cpp:120, at /tmp/binarydeb/ros-galactic-rcl-3.1.2/src/rcl/client.c:111
```

amilcarlucas, 2022-03-21:
I get no printf output, so I guess none of the services gets triggered

amilcarlucas, 2022-03-21:
OK, thanks

pablogs9, 2022-03-21:
Also make sure that you run `ros2 service call /gripper emily_uc_messages/srv/Gripper "{close_gripper: true}"` in the same terminal.

amilcarlucas, 2022-03-21:
Again the same issue, I had to `rm -Rf build/ log/ install/` and rebuild messages. Now I get:
```
# ros2 service call /gripper emily_uc_messages/srv/Gripper "{close_gripper: true}"
requester: making request: emily_uc_messages.srv.Gripper_Request(close_gripper=True)

response:
emily_uc_messages.srv.Gripper_Response(gripper_closed=True)
```
So the code is working, and I can confirm that the hardware also works!
You rock. Thanks

closed 2022-03-21 by amilcarlucas