micro-ROS/micro_ros_arduino #872: Creating a custom message on arduino nano RP2040 connect

SimonGodon, 2022-03-23:
Hi, 

I want to ask here what is the workflow to create and use custom messages on Arduino nano RP2040 connect. I have read many of the posts in this repo and the tutorials but I think I am mixing different concepts and would like to have clear ideas.
My setup:
- A computer with Ubuntu 20.04 and Ros2 Foxy
- An arduino nano rp2040 connect

I started with [this tutorial ](https://micro.ros.org/docs/tutorials/advanced/create_new_type/) saying that I first have to set up a RTOS on the microcontroller, but after also reading  [this post](https://github.com/micro-ROS/micro_ros_arduino/issues/14#issuecomment-722242175) for the specific case of Arduino, and after reading [this github readme](https://github.com/micro-ROS/micro_ros_setup#supported-platforms), I think I understand that no RTOS has to be installed on Arduino nano rp2040 connect  because it uses micro-ROS arduino, which is "just" a library (at least, [this diagram](https://micro.ros.org/docs/overview/features/) leads me in this direction).
Hence, I followed the previously mentionned post from @pablogs9 ([here](https://github.com/micro-ROS/micro_ros_arduino/issues/14#issuecomment-722242175) which asks to follow the steps [here](https://micro.ros.org/docs/tutorials/advanced/create_new_type/) , but by going into home/Arduino/libraries/micro_ros_arduino-0.0.1/extras/library_generation/extra_packages (home/Arduino/libraries/micro_ros_arduino-2.0.3-foxy /extras/library_generation/extra_packages in my case) instead of home/firmware/mcu_ws, which if I'm on the right track, I don't have to create if using Arduino Nano RP2040 connect.

Then I end up with the correct file structure in the extra_packages folder:
```

extras/library_generation/
├── ...
├── extra_packages
│   ├── extra_packages.repos
│   └── my_custom_message
│       ├── CMakeLists.txt
│       ├── include
│       │   └── my_custom_message
│       ├── msg
│       │   └── MyCustomMessage.msg
│       ├── package.xml
│       └── src
└── ...
```

Here, if I am still on the correct track, all I need to do is to build the library:
```
cd ~/Arduino/libraries/micro_ros_arduino-2.0.3-foxy
docker pull microros/micro_ros_static_library_builder:foxy
docker run -it --rm -v $(pwd):/project --env MICROROS_LIBRARY_FOLDER=extras microros/micro_ros_static_library_builder:foxy
```
and then add the following lines in my Arduino IDE script:
`#include <custom_message/msg/MyCustomMessage.h>`
and create a MyCustomMessage object:
`my_custom_message__msg__MyCustomMessage msg;`

But my issue is that I don't have the **MyCustomMessage.h** file, even though the build ended successfully. 
What am I doing wrong?

Thank you

pablogs9, 2022-03-23:
You are doing the process correctly. Just create a ROS 2 msg package and copy the folder in the `extras/library_generation/extra_packages` folder. This should be building your package.

Can you paste here the output of the library build docker command? We will check if the new packet is being included in the build process. BTW you can do `docker run -it --rm -v $(pwd):/project --env MICROROS_LIBRARY_FOLDER=extras microros/micro_ros_static_library_builder:foxy -p cortex_m0` in order to just rebuild the library for your platform.

SimonGodon, 2022-03-23:
Thank you for the quick answer,

So when you say that I have to create a ROS2 msg package, you mean that I should create it in microros_ws, follow the same steps as for creating the folder in home/Arduino/libraries/micro_ros_arduino-2.0.3-foxy /extras/library_generation/extra_packages (so that it includes the msg file and the package.xml and CmakeList.txt are edited), and then add it next to the already existing custom_message folder in `extras/library_generation/extra_packages`?
Or is it in replacement of the custom_message folder?

pablogs9, 2022-03-23:
After doing `ros2 pkg create --build-type ament_cmake my_custom_message` you will have a new folder with your new package. Just copy it inside `extras/library_generation/extra_packages`. You should not replace the folder, just copy your packages inside. In fact you can copy inside as many packages as you want, or even add your package repos to `extra_packages.repos`.

The library building procedure is just copying all those folders inside `extra_packages` to the micro-ROS packages and importing the ones that are referenced in `extra_packages.repos`. Take a look [here](https://github.com/micro-ROS/micro_ros_arduino/blob/ae2db85996dec98250b1fe71f38dcd701dc0277d/extras/library_generation/library_generation.sh#L51-L52).

pablogs9, 2022-03-23:
As you might notice, the package created with `ros2 pkg create --build-type ament_cmake my_custom_message` can be used in micro-ROS and ROS 2. Once you have it in your micro-ROS application, you will need to build it in your ROS 2 side in order to be able to do a `ros2 topic echo ...`.

SimonGodon, 2022-03-23:
Ok so I did `ros2 pkg create --build-type ament_cmake my_custom_message`, made a msg folder and created a message file in it, edited the package.xml, CmakeList.txt, and then I built the package. I copy pasted the `my_custom_message` in `~/Arduino/libraries/micro_ros_arduino-2.0.3-foxy /extras/library_generation/extra_packages`, then I did
`cd ~/Arduino/libraries/micro_ros_arduino-2.0.3-foxy 
docker run -it --rm -v $(pwd):/project --env MICROROS_LIBRARY_FOLDER=extras microros/micro_ros_static_library_builder:foxy -p cortex_m0`
And the output is this: 
```
Get:1 http://security.ubuntu.com/ubuntu focal-security InRelease [114 kB]                                 
Get:2 http://archive.ubuntu.com/ubuntu focal InRelease [265 kB]                                                     
Get:3 http://security.ubuntu.com/ubuntu focal-security/multiverse amd64 Packages [25.8 kB]
Get:4 http://security.ubuntu.com/ubuntu focal-security/universe amd64 Packages [860 kB]                             
Get:5 http://security.ubuntu.com/ubuntu focal-security/main amd64 Packages [1683 kB]                              
Get:6 http://packages.ros.org/ros2/ubuntu focal InRelease [4673 B]                                             
Get:7 http://security.ubuntu.com/ubuntu focal-security/restricted amd64 Packages [1068 kB]                     
Get:8 http://packages.ros.org/ros2/ubuntu focal/main amd64 Packages [999 kB]                                  
Get:9 http://archive.ubuntu.com/ubuntu focal-updates InRelease [114 kB]       
Get:10 http://archive.ubuntu.com/ubuntu focal-backports InRelease [108 kB]               
Get:11 http://archive.ubuntu.com/ubuntu focal/universe amd64 Packages [11.3 MB]
Get:12 http://archive.ubuntu.com/ubuntu focal/multiverse amd64 Packages [177 kB]
Get:13 http://archive.ubuntu.com/ubuntu focal/restricted amd64 Packages [33.4 kB]
Get:14 http://archive.ubuntu.com/ubuntu focal/main amd64 Packages [1275 kB]
Get:15 http://archive.ubuntu.com/ubuntu focal-updates/main amd64 Packages [2099 kB]
Get:16 http://archive.ubuntu.com/ubuntu focal-updates/restricted amd64 Packages [1139 kB]
Get:17 http://archive.ubuntu.com/ubuntu focal-updates/universe amd64 Packages [1146 kB]
Get:18 http://archive.ubuntu.com/ubuntu focal-updates/multiverse amd64 Packages [30.3 kB]
Get:19 http://archive.ubuntu.com/ubuntu focal-backports/universe amd64 Packages [26.0 kB]
Get:20 http://archive.ubuntu.com/ubuntu focal-backports/main amd64 Packages [51.2 kB]
Fetched 22.6 MB in 12s (1837 kB/s)                                                                                                
Reading package lists... Done
Building dependency tree       
Reading state information... Done
174 packages can be upgraded. Run 'apt list --upgradable' to see them.
Creating firmware for generate_lib platform generic
.......
=== ./ament/ament_cmake (git) ===
Cloning into '.'...
=== ./ament/ament_index (git) ===
Cloning into '.'...
=== ./ament/ament_lint (git) ===
Cloning into '.'...
=== ./ament/ament_package (git) ===
Cloning into '.'...
=== ./ament/googletest (git) ===
Cloning into '.'...
=== ./ament/uncrustify_vendor (git) ===
Cloning into '.'...
=== ./ros2/ament_cmake_ros (git) ===
Cloning into '.'...
List of repositories is empty
#All required rosdeps installed successfully
..............
=== ./ros2/common_interfaces (git) ===
Cloning into '.'...
=== ./ros2/example_interfaces (git) ===
Cloning into '.'...
=== ./ros2/libyaml_vendor (git) ===
Cloning into '.'...
=== ./ros2/rcl (git) ===
Cloning into '.'...
=== ./ros2/rcl_interfaces (git) ===
Cloning into '.'...
=== ./ros2/rcl_logging (git) ===
Cloning into '.'...
=== ./ros2/rcpputils (git) ===
Cloning into '.'...
=== ./ros2/rmw (git) ===
Cloning into '.'...
=== ./ros2/rmw_implementation (git) ===
Cloning into '.'...
=== ./ros2/rosidl (git) ===
Cloning into '.'...
=== ./ros2/rosidl_dds (git) ===
[... 146 lines cut ...]
Finished <<< ament_cmake_pycodestyle [0.92s]                                                                         
Starting >>> ament_index_cpp
Finished <<< ament_cmake_pclint [1.02s]                                                                         
Finished <<< ament_cmake_pyflakes [0.88s]                                                                    
Finished <<< ament_cmake_ros [1.00s]                                                                
Finished <<< ament_index_cpp [9.71s]                        

Summary: 60 packages finished [26.7s]
#All required rosdeps installed successfully
#All required rosdeps installed successfully
Cloning into 'geometry2'...
remote: Enumerating objects: 14440, done.
remote: Counting objects: 100% (1758/1758), done.
remote: Compressing objects: 100% (671/671), done.
remote: Total 14440 (delta 1278), reused 1360 (delta 998), pack-reused 12682
Receiving objects: 100% (14440/14440), 3.40 MiB | 5.80 MiB/s, done.
Resolving deltas: 100% (9857/9857), done.
=== ./control_msgs (git) ===
Cloning into '.'...
fatal: not a git repository (or any parent up to mount point /)
Stopping at filesystem boundary (GIT_DISCOVERY_ACROSS_FILESYSTEM not set).
No changes detected
```

For some reason it doesn't seem to include the new packet, and hence I still don't have the **MyCustomMessage.h**

pablogs9, 2022-03-23:
Hello @Acuadros95, can you take a look at this `No changes detected`. Is this related to the modifications you made?

pablogs9, 2022-03-23:
@SimonGodon try one thing, go inside your `~/Arduino/libraries/micro_ros_arduino-2.0.3-foxy` and make `git fetch && git checkout foxy && git pull`

@Acuadros95 can you make a new release of micro-ROS for Arduino?

SimonGodon, 2022-03-23:
Doing 
```
cd ~/Arduino/libraries/micro_ros_arduino-2.0.3-foxy
git fetch && git checkout foxy && git pull
```
results in 
`fatal: not a git repository (or any parent directory): .git`

pablogs9, 2022-03-23:
Well, it seems that downloading the repo as .zip is not a git repo.

Just do:

```bash
cd ~/Arduino/libraries
rm -rf micro_ros_arduino-2.0.3-foxy
git clone -b https://github.com/micro-ROS/micro_ros_arduino
```

in order to get the latest version of `foxy` branch. The version 2.0.3 might be outdated.

Acuadros95, 2022-03-23:
Exactly, this is why this error appears `fatal: not a git repository (or any parent up to mount point /)`
> Well, it seems that downloading the repo as .zip is not a git repo.

This was fixed here https://github.com/micro-ROS/micro_ros_arduino/pull/794, but its not included on release `2.0.3`.
Let me handle a new `2.0.4` release

Acuadros95, 2022-03-23:
Just tested it on latest foxy commit and its working. It should appear on the generated `available_ros2_types` file under the name: `my_custom_message/MyCustomMessage.msg`

Could you share your `extra_packages` folder?

SimonGodon, 2022-03-24:
Ok now I got it working, I think the latest issue was coming from the fact that I renamed the package and changed all files content accordingly, but somehow the old name must have stayed somewhere. The fix was to delete and redo the package from scratch, then the steps described earlier worked. 

Just in case it can help anyone later, I'll share the content of the extra_packages folder ([here](https://github.com/micro-ROS/micro_ros_arduino/files/8339967/extra_packages.zip)) and discuss how I made 2 messages and what is the general procedure, @pablogs9 @Acuadros95 please correct me if I'm making mistakes.

I created a package with 2 new messages named  MyCustomMessage.msg and MyOtherCustomMessage.msg, changed the CMakeList.txt and package.xml accordingly, following [this](https://micro.ros.org/docs/tutorials/advanced/create_new_type/) tutorial (the tutorial only creates 1 message, I just made 2 to see what differed in the arduino code). However you have to create the package in `~microros_ws/src` (not in `~firmware/mcu_ws/src`, which you shouldn't create since you don't have to install a RTOS on Arduino because Microros for Arduino is a library usable by Arduino IDE). Once the files are edited/created, build your microros_ws, and then copy paste your package folder in `~/Arduino/libraries/micro_ros_arduino/extras/library_generation/extra_packages` and build the library:
```
cd ~/Arduino/libraries/micro_ros_arduino
docker run -it --rm -v $(pwd):/project --env MICROROS_LIBRARY_FOLDER=extras microros/micro_ros_static_library_builder:foxy -p cortex_m0
```
The optional flag  `-p cortex_m0 `might be different if you are using another arduino board.
After build you can verify in `micro_ros_arduino/available_ros2_types` that your messages are listed, under a syntax like `my_custom_message/MyCustomMessage.msg`. 
To finish, use these messages in your Arduino sketch using
```
#include <micro_ros_arduino.h>
#include <my_custom_message/msg/my_custom_message.h>
#include <my_custom_message/msg/my_other_custom_message.h>
```
and

```
my_custom_message__msg__MyCustomMessage msg1;
my_custom_message__msg__MyOtherCustomMessage msg2;
```
Hope it will help someone, and thank you @pablogs9  and @Acuadros95  for helping on this!

closed 2022-03-24 by pablogs9