micro-ROS/micro_ros_arduino #881: Publishing issues on ESP32

MrOCW, 2022-03-24:
## Issue template

- Hardware description: ESP32, Ubuntu 20.04
- RTOS: N.A.
- Installation type: micro_ros_arduino on PlatformIO
- Version or commit hash: galactic

#### Steps to reproduce the issue
Publish IMU msg

#### Expected behavior
ros2 topic echo imu/data_raw to continuously update & display
#### Actual behavior
1. Had to replug the USB every time i rerun ros2 run micro_ros_agent micro_ros_agent serial --dev /dev/ttyUSB0, otherwise it would be stuck at 

> [1648115159.642399] info     | TermiosAgentLinux.cpp | init                     | running...             | fd: 3
> [1648115159.642567] info     | Root.cpp           | set_verbose_level        | logger setup           | verbose_level: 4

2. Had to wait for few seconds before displaying only once with the stamps = 0??

> header:
>   stamp:
>     sec: 0
>     nanosec: 0
>   frame_id: MPU9250_fr
> orientation:
>   x: 0.0
>   y: 0.0
>   z: 0.0
>   w: 0.0
> orientation_covariance:
> - -1.0
> - 0.0
> - 0.0
> - 0.0
> - 0.0
> - 0.0
> - 0.0
> - 0.0
> - 0.0
> angular_velocity:
>   x: 0.0
>   y: 0.0
>   z: -0.0
> angular_velocity_covariance:
> - 0.0
> - 0.0
> - 0.0
> - 0.0
> - 0.0
> - 0.0
> - 0.0
> - 0.0
> - 0.0
> linear_acceleration:
>   x: -5.0256853103637695
>   y: 6.016935348510742
>   z: 9.8100004196167
> linear_acceleration_covariance:
> - 0.0
> - 0.0
> - 0.0
> - 0.0
> - 0.0
> - 0.0
> - 0.0
> - 0.0
> - 0.0
> ---

#### Additional information
This is the code related to the IMU

extern "C" int clock_gettime(clockid_t unused, struct timespec *tp);
rcl_publisher_t imu_publisher;
sensor_msgs__msg__Imu imu_raw;
rcl_allocator_t allocator;
rclc_executor_t executor;
rclc_support_t support;
rcl_node_t node;

void setup()
{
  set_microros_transports();
  allocator = rcl_get_default_allocator();
  rclc_support_init(&support, 0, NULL, &allocator);
  rclc_node_init_default(&node, "micro_ros_node", "", &support);
  rclc_publisher_init_best_effort(
    &imu_publisher,
    &node,
    ROSIDL_GET_MSG_TYPE_SUPPORT(sensor_msgs, msg, Imu),
    "/imu/data_raw");
  
  imu_raw.header.frame_id.data = "MPU9250_fr";
  imu_raw.header.frame_id.size = 10;

}
void loop()
{
  struct timespec tv = {0};
  clock_gettime(0, &tv);
  // read the sensor
  IMU.readSensor();
  imu_raw.linear_acceleration.x = IMU.getAccelX_mss();
  imu_raw.linear_acceleration.y = -IMU.getAccelY_mss();
  imu_raw.linear_acceleration.z = 9.81f;
  imu_raw.angular_velocity.x = IMU.getGyroX_rads();
  imu_raw.angular_velocity.y = IMU.getGyroY_rads();
  imu_raw.angular_velocity.z = -IMU.getGyroZ_rads();
  imu_raw.header.stamp.sec = tv.tv_sec;
  imu_raw.header.stamp.nanosec = tv.tv_nsec;;
  imu_raw.orientation_covariance[0] = -1;
  rcl_publish(&imu_publisher, &imu_raw, NULL);
}

Acuadros95, 2022-03-24:
You need to initialize the message memory, in your case the `Header` strings and each data sequence. 
Check this tutorial: [Handling messages memory in micro-ROS](https://micro.ros.org/docs/tutorials/advanced/handling_type_memory/)

pablogs9, 2022-03-24:
You need to instantiate the micro-ROS agent before connecting the board because the client will try to connect to the agent in rclc_support_init() for a [certain amount of times](https://github.com/eProsima/Micro-XRCE-DDS-Client/blob/243b16b659f1b46e6678db21c30f246910a3b697/CMakeLists.txt#L57) with a [certain timeout](https://github.com/eProsima/Micro-XRCE-DDS-Client/blob/243b16b659f1b46e6678db21c30f246910a3b697/CMakeLists.txt#L58). You can check the return value of this function and retry until the agent is reached. You can [ping the agent](https://github.com/micro-ROS/micro_ros_arduino/blob/e85b9b47142601a72c742e1af71a0ff399b03c07/examples/micro-ros_reconnection_example/micro-ros_reconnection_example.ino#L93) in order to wait until it is available. But in general, you should handle the client-agent connection.

Regarding your code, try to use Arduino timing functions to fill the header's timestamp.

MrOCW, 2022-03-24:
i think i found the issue, it seems that removing all subscribers and rclc_executor_spin(&executor); allows the IMU messages to be published without issues but when i added the subscribers and executor related code, the IMU issue comes back again. is there anything i should note regarding using both subscibers and publishers together?

pablogs9, 2022-03-24:
There is no executor or subscriber in the code you shared.

MrOCW, 2022-03-24:
ah apologies, the solution was to use rclc_executor_spin_some instead of rclc_executor_spin.

closed 2022-03-24 by MrOCW