micro-ROS/micro_ros_arduino #884: nav_msgs/msg/Odometry data not being published

grassjelly, 2022-03-24:
- Hardware description:
Teensy 4.0, x86 Linux Machine
- RTOS: NA
- Installation type: micro_ros_arduino on PlatformIO
- Version or commit hash: galactic

#### Steps to reproduce the issue
Upload the code below:

```
#include <Arduino.h>
#include <micro_ros_arduino.h>
#include <stdio.h>

#include <rcl/rcl.h>
#include <rcl/error_handling.h>
#include <rclc/rclc.h>
#include <rclc/executor.h>

#include <nav_msgs/msg/odometry.h>

#define LED_PIN 13 //used for debugging status
#define RCCHECK(fn) { rcl_ret_t temp_rc = fn; if((temp_rc != RCL_RET_OK)){rclErrorLoop();}}
#define RCSOFTCHECK(fn) { rcl_ret_t temp_rc = fn; if((temp_rc != RCL_RET_OK)){}}

rcl_publisher_t odom_publisher;
nav_msgs__msg__Odometry odom_msg;

rclc_executor_t executor;
rclc_support_t support;
rcl_allocator_t allocator;
rcl_node_t node;
rcl_timer_t control_timer;

unsigned long long time_offset = 0;
bool micro_ros_init_successful = false;

void setup() 
{
    pinMode(LED_PIN, OUTPUT);
    micro_ros_init_successful = false;
    set_microros_transports();
}

void loop() 
{
    static unsigned long prev_connect_test_time;
    // check if the agent got disconnected at 10Hz
    if(millis() - prev_connect_test_time >= 100)
    {
        prev_connect_test_time = millis();
        // check if the agent is connected
        if(RMW_RET_OK == rmw_uros_ping_agent(10, 2))
        {
            // reconnect if agent got disconnected or haven't at all
            if (!micro_ros_init_successful) 
            {
                createEntities();
            } 
        } 
        else if(micro_ros_init_successful)
        {
            // clean up micro-ROS components
            destroyEntities();
        }
    }
    
    if(micro_ros_init_successful)
    {
        rclc_executor_spin_some(&executor, RCL_MS_TO_NS(10));
    }
}

void controlCallback(rcl_timer_t * timer, int64_t last_call_time) 
{
    publishData();
}

void createEntities()
{
    allocator = rcl_get_default_allocator();
    //create init_options
    RCCHECK(rclc_support_init(&support, 0, NULL, &allocator));
    // create node
    RCCHECK(rclc_node_init_default(&node, "odometry_test", "", &support));
    // create odometry publisher
    RCCHECK(rclc_publisher_init_best_effort( 
        &odom_publisher, 
        &node,
        ROSIDL_GET_MSG_TYPE_SUPPORT(nav_msgs, msg, Odometry),
        "odom/unfiltered"
    ));

    const unsigned int control_timeout = 20;
    RCCHECK(rclc_timer_init_default( 
        &control_timer, 
        &support,
        RCL_MS_TO_NS(control_timeout),
        controlCallback
    ));
    executor = rclc_executor_get_zero_initialized_executor();
    RCCHECK(rclc_executor_init(&executor, &support.context, 1, & allocator));
    RCCHECK(rclc_executor_add_timer(&executor, &control_timer));
    // synchronize time with the agent
    syncTime();
    digitalWrite(LED_PIN, HIGH);
    micro_ros_init_successful = true;
}

void destroyEntities()
{
    digitalWrite(LED_PIN, LOW);

    rcl_publisher_fini(&odom_publisher, &node);
    rcl_node_fini(&node);
    rcl_timer_fini(&control_timer);
    rclc_executor_fini(&executor);
    rclc_support_fini(&support);

    micro_ros_init_successful = false;
}

void publishData()
{
    struct timespec time_stamp = getTime();

    odom_msg.header.stamp.sec = time_stamp.tv_sec;
    odom_msg.header.stamp.nanosec = time_stamp.tv_nsec;
    odom_msg.pose.pose.orientation.w = 1.;

    RCSOFTCHECK(rcl_publish(&odom_publisher, &odom_msg, NULL));
}

void syncTime()
{
    // get the current time from the agent
    unsigned long now = millis();
    RCCHECK(rmw_uros_sync_session(10));
    unsigned long long ros_time_ms = rmw_uros_epoch_millis(); 
    // now we can find the difference between ROS time and uC time
    time_offset = ros_time_ms - now;
}

struct timespec getTime()
{
    struct timespec tp = {0};
    // add time difference between uC time and ROS time to
    // synchronize time with ROS
    unsigned long long now = millis() + time_offset;
    tp.tv_sec = now / 1000;
    tp.tv_nsec = (now % 1000) * 1000000;

    return tp;
}

void rclErrorLoop() 
{
    while(true)
    {
        flashLED(2);
    }
}

void flashLED(int n_times)
{
    for(int i=0; i<n_times; i++)
    {
        digitalWrite(LED_PIN, HIGH);
        delay(150);
        digitalWrite(LED_PIN, LOW);
        delay(150);
    }
    delay(1000);
}
```

#### Expected behavior

Receive data on `odom/unfiltered` topic

#### Actual behavior

Nothing is received

#### Additional information

Odometry msg is not published although other messages (IMU) are. The above code is a minimal version of my code from Linorobot project to solely publish odometry data. The same code used to work from previous commits of micro_ros_arduino.

Thank you!

grassjelly, 2022-03-24:
I haven't fully investigated yet but could this be related to this? https://github.com/micro-ROS/micro_ros_arduino/issues/48

pablogs9, 2022-03-24:
Try to fill `odom_msg.header.frame_id` and `odom_msg.child_frame_id` both are strings and thery are not being initialized.

https://micro.ros.org/docs/tutorials/advanced/handling_type_memory/

pablogs9, 2022-03-24:
Be careful because you are allocating memory each time `micro_ros_utilities_create_message_memory` is being called in `createEntities`.

Can you try a basic approach with something like the below code and share the micro-ROS Agent output using `-v6`?
Make sure that the agent is open before connecting the board.

```c
#include <Arduino.h>
#include <micro_ros_arduino.h>
#include <stdio.h>

#include <rcl/rcl.h>
#include <rcl/error_handling.h>
#include <rclc/rclc.h>
#include <rclc/executor.h>
#include <micro_ros_utilities/type_utilities.h>
#include <micro_ros_utilities/string_utilities.h>
#include <nav_msgs/msg/odometry.h>

#define LED_PIN 13 //used for debugging status
#define RCCHECK(fn) { rcl_ret_t temp_rc = fn; if((temp_rc != RCL_RET_OK)){rclErrorLoop();}}
#define RCSOFTCHECK(fn) { rcl_ret_t temp_rc = fn; if((temp_rc != RCL_RET_OK)){}}

rcl_publisher_t odom_publisher;
nav_msgs__msg__Odometry odom_msg;

rclc_executor_t executor;
rclc_support_t support;
rcl_allocator_t allocator;
rcl_node_t node;
rcl_timer_t control_timer;

unsigned long long time_offset = 0;
bool micro_ros_init_successful = false;

void setup() 
{
    pinMode(LED_PIN, OUTPUT);
    micro_ros_init_successful = false;
    set_microros_transports();

    micro_ros_utilities_create_message_memory(
        ROSIDL_GET_MSG_TYPE_SUPPORT(nav_msgs, msg, Odometry),
        &odom_msg,
        (micro_ros_utilities_memory_conf_t) {});

    odom_msg.header.frame_id = micro_ros_string_utilities_set(odom_msg.header.frame_id, "odom");
    odom_msg.child_frame_id = micro_ros_string_utilities_set(odom_msg.child_frame_id, "base_footprint");

    createEntities();
}

void loop() 
{
    struct timespec time_stamp = getTime();

    odom_msg.header.stamp.sec = time_stamp.tv_sec;
    odom_msg.header.stamp.nanosec = time_stamp.tv_nsec;
    odom_msg.pose.pose.orientation.w = 1.;

    RCSOFTCHECK(rcl_publish(&odom_publisher, &odom_msg, NULL));
}

....
```

grassjelly, 2022-03-24:
I think it's crashing..here's the output:

```

[1648134162.581297] info     | TermiosAgentLinux.cpp | init                     | Serial port not found. | device: /ttyACM0, error 2, waiting for connection...
[1648134163.587304] info     | TermiosAgentLinux.cpp | init                     | Serial port not found. | device: /ttyACM0, error 2, waiting for connection...
[1648134164.592930] info     | TermiosAgentLinux.cpp | init                     | Serial port not found. | device: /ttyACM0, error 2, waiting for connection...
[1648134165.598368] info     | TermiosAgentLinux.cpp | init                     | Serial port not found. | device: /ttyACM0, error 2, waiting for connection...
[1648134166.604641] info     | TermiosAgentLinux.cpp | init                     | Serial port not found. | device: /ttyACM0, error 2, waiting for connection...

```

It says Serial Port not found but I can see the board when running:
`ls /dev/ttyACM*`

pablogs9, 2022-03-24:
This output is with the code I shared or with the one you provided?

grassjelly, 2022-03-24:
Hey Pablo. Thanks for helping, it works now. Earlier you've mentioned to make sure that the agent must be open before connecting the board. Does it always have to be this way? Realized that it only works when the agent is open first.

pablogs9, 2022-03-25:
You need to instantiate the micro-ROS agent before connecting the board because the client will try to connect to the agent in rclc_support_init() for a [certain amount of times](https://github.com/eProsima/Micro-XRCE-DDS-Client/blob/243b16b659f1b46e6678db21c30f246910a3b697/CMakeLists.txt#L57) with a [certain timeout](https://github.com/eProsima/Micro-XRCE-DDS-Client/blob/243b16b659f1b46e6678db21c30f246910a3b697/CMakeLists.txt#L58). You can check the return value of this function and retry until the agent is reached. Also, you can [ping the agent](https://github.com/micro-ROS/micro_ros_arduino/blob/e85b9b47142601a72c742e1af71a0ff399b03c07/examples/micro-ros_reconnection_example/micro-ros_reconnection_example.ino#L93) in order to wait until it is available. But in general, you should handle the client-agent connection.

Here you have a good example of how to handle this reconnection: https://github.com/micro-ROS/micro_ros_tivac_launchpad_app/blob/053ff64f635ccf8cef558e1639725abbd3c24db3/src/microros.c#L104-L135

If you don't mind I'm closing. Please reopen if you have further questions.

closed 2022-03-25 by pablogs9

grassjelly, 2022-03-25:
Got it. Thanks again @pablogs9

kjrusscher, 2024-07-01:
I had this same problem. For others with the same problem: solution is to use the default initialization function for the odometry publisher. So replacing `rclc_publisher_init_best_effort` with `rclc_publisher_init_default`.