micro-ROS/micro_ros_arduino #885: NativeEthernet library not found when using PlatformIO

DaniGarciaLopez, 2022-03-24:
## Issue template

- Hardware description: Teensy 4.0 with PlatformIO
- Version or commit hash: galactic

#### Steps to reproduce the issue

Create an empty project in platformIO with the configuration suggested in the README for Teensy 4.0 in patformio.ini:
```
[env:teensy40]
platform = teensy
board = teensy40
framework = arduino

lib_deps =
    https://github.com/micro-ROS/micro_ros_arduino

build_flags =
    -L ./.pio/libdeps/teensy40/micro_ros_arduino/src/imxrt1062/fpv5-d16-hard
    -l microros
    -D ARDUINO_TEENSY41
```
And a simple main program importing microros:
```
#include <Arduino.h>
#include <micro_ros_arduino.h>

void setup() {
  // put your setup code here, to run once:
}

void loop() {
  // put your main code here, to run repeatedly:
}
```

#### Expected behavior
Successfull Compilation 

#### Actual behavior
Following error appears:
```
.pio/build/teensy40/libFrameworkArduino.a(startup.c.o): In function `configure_external_ram':
startup.c:(.flashmem+0x4a0): undefined reference to `_extram_end'
startup.c:(.flashmem+0x4a4): undefined reference to `_extram_start'
collect2: error: ld returned 1 exit status
*** [.pio/build/teensy40/firmware.elf] Error 1
```

Without the build flag `-D ARDUINO_TEENSY41`, PlatformIO throws errors from the NativeEthernet library:
```
/home/user/.platformio/packages/framework-arduinoteensy/libraries/NativeEthernet/src/NativeEthernet.cpp: In static member function 'static int EthernetClass::begin(uint8_t*, long unsigned int, long unsigned int)':
/home/user/.platformio/packages/framework-arduinoteensy/libraries/NativeEthernet/src/NativeEthernet.cpp:126:30: error: 'FNET_CPU_ETH0_IF' was not declared in this scope
           if(fnet_netif_init(FNET_CPU_ETH0_IF, mac, 6) != FNET_ERR){
                              ^
/home/user/.platformio/packages/framework-arduinoteensy/libraries/NativeEthernet/src/NativeEthernet.cpp: In static member function 'static void EthernetClass::begin(uint8_t*, IPAddress, IPAddress, IPAddress, IPAddress)':
/home/user/.platformio/packages/framework-arduinoteensy/libraries/NativeEthernet/src/NativeEthernet.cpp:258:30: error: 'FNET_CPU_ETH0_IF' was not declared in this scope
           if(fnet_netif_init(FNET_CPU_ETH0_IF, mac, 6) != FNET_ERR){
                              ^
Compiling .pio/build/teensy40/lib2fb/micro_ros_arduino/native_ethernet_transport.cpp.o
*** [.pio/build/teensy40/lib6bd/NativeEthernet/NativeEthernet.cpp.o] Error 1
```
#### Additional information
We were previously using an older precompiled version in galactic with PlatformIO without any problems. It compiles without errors using Arduino IDE. We also tried with a Teensy 3.2 with the same behaviour.

Thanks!

grassjelly, 2022-03-24:
you can ignore NativeEthernet library. Here's a reference: https://github.com/linorobot/linorobot2_hardware/blob/master/firmware/platformio.ini#L59-L66

DaniGarciaLopez, 2022-03-25:
Thanks @grassjelly! It works as you suggest. Closing the issue.

closed 2022-03-25 by DaniGarciaLopez