micro-ROS/micro_ros_arduino #892: Client callback does not get called

Johannap1, 2022-03-28:
- Hardware description: ESP32
- RTOS: microros arduino with platformio
- Version or commit hash: ROS2 foxy

Hello, 

I would like a client trigger to initiate a system shutdown, but currently I have no idea if the service call was successful on the microros side, because the client callback does not get called in my implementation. Right now, it gets stuck with sending the shutdown request (see code below). On the main device side the callback is succesful, so that works, but I cannot get the system response back. Unfortunately, I could also not find a client example for microros arduino that I could test, so any help what could cause this would be appreciated. I followed the RTOS example [here](https://github.com/micro-ROS/micro-ROS-demos/blob/foxy/rclc/addtwoints_client/main.c).

Code on micro-ros side:
`#include <Arduino.h>

#include <micro_ros_arduino.h>

#include <stdio.h>
#include <rcl/rcl.h>
#include <rcl/error_handling.h>
#include <rclc/rclc.h>
#include <rclc/executor.h>

#include <std_msgs/msg/bool.h>
#include <std_msgs/msg/string.h>
#include <std_msgs/msg/int32.h>
#include <std_msgs/msg/float32.h>
#include <rcl_interfaces/msg/log.h>
#include <std_srvs/srv/trigger.h>

#include "MessageLogger.h"

#define LED_PIN 13 //TODO: define error LED

rcl_node_t node;
rclc_support_t support;
rcl_allocator_t allocator;
rclc_executor_t executor;

//Logging publishers
rcl_publisher_t log_publisher;
rcl_interfaces__msg__Log msgLog;
rcl_publisher_t publisher;
std_msgs__msg__Bool test;

//Shutdown client
rcl_client_t shutdown_client;
std_srvs__srv__Trigger_Request shutdown_request;
std_srvs__srv__Trigger_Response shutdown_response;
int64_t shutdown_sequence_number;

bool shutdown_;

#define RCCHECK(fn) { rcl_ret_t temp_rc = fn; if((temp_rc != RCL_RET_OK)){error_loop();}}
#define RCSOFTCHECK(fn) { rcl_ret_t temp_rc = fn; if((temp_rc != RCL_RET_OK)){}}

void error_loop(){
  while(1){
    digitalWrite(LED_PIN, !digitalRead(LED_PIN));
    delay(100);
  }
}

//TODO: this does not work for some reason
void shutdown_client_callback(const void * shutdown_response){
  std_srvs__srv__Trigger_Response * shutdown_response_in = (std_srvs__srv__Trigger_Response *) shutdown_response;
  send_log_message("Shutdown service callback called", 1);
  if(shutdown_response_in->success){
    shutdown_ = true;
  }
}

void send_log_message(std::string log_message, uint32_t severity){
  switch (severity){
    case 0:
      msgLog.level = rcl_interfaces__msg__Log__DEBUG;
    case 1:
      msgLog.level = rcl_interfaces__msg__Log__INFO;
    case 2:
      msgLog.level = rcl_interfaces__msg__Log__WARN;
  }
  msgLog.level = rcl_interfaces__msg__Log__INFO;
  msgLog.name.data = (char *)"ESP32 ";
  msgLog.name.size = strlen(msgLog.name.data);
  msgLog.msg.data = (char*)log_message.c_str();
  msgLog.msg.size = strlen(msgLog.msg.data);
  RCSOFTCHECK(rcl_publish(&log_publisher, &msgLog, NULL));
}

void setup() {

  //microros setup
  set_microros_transports();
  delay(1000); 

  allocator = rcl_get_default_allocator();

  // create init_options
  RCCHECK(rclc_support_init(&support, 0, NULL, &allocator));

  // create node
  RCCHECK(rclc_node_init_default(&node, "user_interface", "", &support));

  // create logging publisher
  RCCHECK(rclc_publisher_init_default(
    &log_publisher,
    &node,
    ROSIDL_GET_MSG_TYPE_SUPPORT(rcl_interfaces, msg, Log),
    "/rosout"));

  // create shutdown client
  RCCHECK(rclc_client_init_default(&shutdown_client, &node, ROSIDL_GET_SRV_TYPE_SUPPORT(std_srvs, srv, Trigger), "full_system_shutdown"));

  // create executor
  RCCHECK(rclc_executor_init(&executor, &support.context, 1, &allocator));
  RCCHECK(rclc_executor_add_client(&executor, &shutdown_client, &shutdown_response, &shutdown_client_callback)); //TODO: use pointer instead

}

void loop() {
  RCSOFTCHECK(rclc_executor_spin_some(&executor, RCL_MS_TO_NS(10))); 

  delay(1000);

  
  if(!shutdown_){
    std_srvs__srv__Trigger_Request__init(&shutdown_request);
    send_log_message("Calling shutdown service.", 1);
    shutdown_request.structure_needs_at_least_one_member = 0;
    RCCHECK(rcl_send_request(&shutdown_client, &shutdown_request, &shutdown_sequence_number));
  }
  else if(shutdown_){
    send_log_message("Shutdown succesful.", 1);
  }

}`

Code on main device
`   self.shutdown_system_service = self.create_service(Trigger, 'full_system_shutdown', self.shutdown_system_callback)

    def shutdown_system_callback(self, request, response):
        response.success = True
        #response.message = 'system shutdown successful'
        self.get_logger().info('system shutdown successful')
        return response
'

Thanks!

pablogs9, 2022-03-28:
Try to init memory in the response callback:

```c
void setup() {
shutdown_response.message.data = (char *) malloc(100);
shutdown_response.message.size = 0;
shutdown_response.message.capacity = 100;
...
```

More info here: https://micro.ros.org/docs/tutorials/advanced/handling_type_memory/

Johannap1, 2022-03-28:
That solved it! Thanks so much for the quick answer.

closed 2022-03-28 by Johannap1