micro-ROS/micro_ros_arduino #907: Undefined reference to rclc_lifecycle functions

tnagyzambo, 2022-04-01:
## Issue template

- Hardware description: Arduino Portenta H7
- RTOS: mbedOS
- Installation type: platformio
- Version or commit hash: galactic

#### Steps to reproduce the issue
Attempt to compile minimal lifecycle example:
```c
#include <micro_ros_arduino.h>

#include <rcl/error_handling.h>
#include <rcutils/logging_macros.h>

#include <lifecycle_msgs/msg/transition_description.h>
#include <lifecycle_msgs/msg/transition_event.h>
#include <lifecycle_msgs/srv/change_state.h>
#include <lifecycle_msgs/srv/get_available_states.h>
#include <lifecycle_msgs/srv/get_available_transitions.h>
#include <lifecycle_msgs/srv/get_state.h>

#include <rcl/error_handling.h>
#include <rcl/rcl.h>
#include <rclc/executor.h>
#include <rclc/rclc.h>

#include <rclc_lifecycle/rclc_lifecycle.h>

#define LED_PIN 13

#define RCCHECK(fn)                                                                                                    \
    {                                                                                                                  \
        rcl_ret_t temp_rc = fn;                                                                                        \
        if ((temp_rc != RCL_RET_OK)) {                                                                                 \
            printf("Failed status on line %d: %d. Aborting.\n", __LINE__, (int)temp_rc);                               \
            return 1;                                                                                                  \
        }                                                                                                              \
    }

#define RCSOFTCHECK(fn)                                                                                                \
    {                                                                                                                  \
        rcl_ret_t temp_rc = fn;                                                                                        \
        if ((temp_rc != RCL_RET_OK)) {                                                                                 \
            printf("Failed status on line %d: %d. Continuing.\n", __LINE__, (int)temp_rc);                             \
        }                                                                                                              \
    }

void error_loop() {
    while (1) {
        digitalWrite(LED_PIN, !digitalRead(LED_PIN));
        delay(100);
    }
}

int my_on_configure() {
    printf("  >>> lifecycle_node: on_configure() callback called.\n");
    return RCL_RET_OK;
}

int my_on_activate() {
    printf("  >>> lifecycle_node: on_activate() callback called.\n");
    return RCL_RET_OK;
}

int my_on_deactivate() {
    printf("  >>> lifecycle_node: on_deactivate() callback called.\n");
    return RCL_RET_OK;
}

[... 120 lines cut ...]
main.cpp:(.text.startup.main+0x1b2): undefined reference to `rclc_lifecycle_node_fini(rclc_lifecycle_node_t*, rcutils_allocator_t*)'
collect2: error: ld returned 1 exit status
*** [.pio/build/portenta_h7_m7/firmware.elf] Error 1
=============================================================================== [FAILED] Took 2.92 seconds ===============================================================================
```

#### Additional information
Not sure if it is helpful but running `nm` on `libmicroros.a` results in:
```
00000000 T rclc_lifecycle_change_state
00000000 T rclc_lifecycle_change_state_callback
00000000 T rclc_lifecycle_execute_callback
00000000 T rclc_lifecycle_get_available_states_callback
00000000 T rclc_lifecycle_get_state_callback
00000000 T rclc_lifecycle_init_change_state_server
00000000 T rclc_lifecycle_init_get_available_states_server
00000000 T rclc_lifecycle_init_get_state_server
00000000 T rclc_lifecycle_node_fini
00000000 T rclc_lifecycle_register_callback
00000000 T rclc_lifecycle_register_on_activate
00000000 T rclc_lifecycle_register_on_cleanup
00000000 T rclc_lifecycle_register_on_configure
00000000 T rclc_lifecycle_register_on_deactivate
00000000 T rclc_make_node_a_lifecycle_node
```

pablogs9, 2022-04-01:
rclc_lifecycle is not built in the micro-ROS library, check [here](https://github.com/micro-ROS/micro_ros_setup/blob/5a2eb5788a5ba0b44b61b2939139e9ebde60ca07/config/host/generic/create.sh#L12)

You can copy it to `extra_packages` folder and try to build it, but it is not tested nor validated in MCU environments. Maybe the maintainer @JanStaschulat can tell us more details.

tnagyzambo, 2022-04-03:
Thanks for the reply.

Unfortunately your suggestion does not work as there are package name conflicts.
```
ERROR:colcon:colcon build: Duplicate package names not supported:
- rclc_lifecycle:
  - extra_packages/rclc/rclc_lifecycle
  - uros/rclc/rclc_lifecycle
  ```
 From my reading of the micro ros project website it suggests that this feature should be ready to use. Maybe there is a misunderstanding on my part. Since it seems that this is not limited to the Arduino library, Is there a more appropriate place to continue this discussion?

pablogs9, 2022-04-04:
> From my reading of the micro ros project website it suggests that this feature should be ready to use. Maybe there is a misunderstanding on my part. Since it seems that this is not limited to the Arduino library, is there a more appropriate place to continue this discussion?

The rclc repo

pablogs9, 2022-04-05:
Reopening until https://github.com/ros2/rclc/pull/279 is merged and library is supported under C++ build

pablogs9, 2022-04-08:
https://github.com/ros2/rclc/pull/279 merged

closed 2022-04-08 by pablogs9