micro-ROS/micro_ros_arduino #909: [Custom Transport] micro_ros_agent not generating any messages in response to packets sent from arduino via UDP

SpMakris, 2022-04-02:
- Hardware description: Arduino DUE, ENC28J60 SPI ethernet peripheral
- Installation type: Micro_ros publisher example, micro_ros_agent udp4 mode
- Version or commit hash: ROS2 galactic

I am trying to implement a UDP transport layer, by modifying the arduino_default transport, which is set up to use the Serial port by default.
I have modified the default open, read, write and close functions according to the ethernet.h documentation [here ](https://www.arduino.cc/en/Reference/EthernetUDPWrite)and [here ](https://www.arduino.cc/en/Reference/EthernetUDPRead)

>  byte mac[] = {
>     0xDE, 0xAD, 0xBE, 0xEF, 0xFE, 0xED};
> IPAddress ip(10, 42, 0, 10);
> IPAddress remoteHost(10, 42, 0, 1);
> unsigned int remotePort = 11349;
> unsigned int localPort = 11345; // local port to listen on
> EthernetUDP Udp;

> bool arduino_transport_open(struct uxrCustomTransport *transport)
>   {
>     Ethernet.init(41);
>     Ethernet.begin(mac, ip);
>     Udp.begin(localPort);
>     return true;
>   }

> bool arduino_transport_close(struct uxrCustomTransport *transport)
>   { 
>     return true;
>   }
>

(Includes printing the packet buffer for debugging purposes)
>   size_t arduino_transport_write(struct uxrCustomTransport *transport, uint8_t *buf, size_t len, uint8_t *errcode)
>   {
>     Udp.beginPacket(remoteHost, remotePort);
> 
>     int temp = Udp.write((char *)buf, len);
>     for(int i = 0; i<len; i++)
>     {
>       SerialUSB.print(buf[i], HEX);
>       SerialUSB.print(" ");
>     }
>     SerialUSB.println("");
>     Udp.endPacket();
>     SerialUSB.println("send packet");
>     return temp;
>   }
> 
>   size_t arduino_transport_read(struct uxrCustomTransport *transport, uint8_t *buf, size_t len, int timeout, uint8_t *errcode)
>   {
>     int packetSize = Udp.parsePacket();
>     if (packetSize)
>     {
>       Udp.read((char *)buf, len);
>     }
>     return packetSize;
>   }

I start the micro ros agent with the command
>ros2 run micro_ros_agent micro_ros_agent udp4 --port 11349 -v6 ROS_DOMAIN_ID=0

Upon starting the arduino due I can see UDP packets arriving on the terminal, and can verify the correct packet is received by comparing it to the output of the serial console. I don't see any response being generated from the micro_ros agent, and no new topics show up when executing 
>ros2 topic list .

Sample output from the micro_ros agent
[image]

I'm not sure where to look. Maybe someone has tried a similar thing and can help me out?

SpMakris, 2022-04-03:
After digging around, I tried setting XRCE framing to false in file micro_ros_arduino.h, in the function set_microros_transports, and now it seems to connect as it should. I am a beginner in the ROS2 and microros environments so I don't really understand the implications of that change or why it makes it work, so I'm leaving this issue open for now, until a more experienced developer can chime in (and close it if they consider this to be a proper solution).

pablogs9, 2022-04-04:
You need to disable framing when using packet-oriented transports. Framing is only used in stream-like transport. More information here: https://micro-xrce-dds.docs.eprosima.com/en/latest/transport.html#stream-framing-label

closed 2022-04-04 by pablogs9