micro-ROS/micro_ros_arduino #910: RP2040 cortex-m0plus/libmicroros.a not linking functions

Kotochleb, 2022-04-02:
- Hardware description: Maker Pi RP2040 - Cytron
- RTOS: None
- Installation type: PlatformIO
- Version or commit hash: Galactic on Ubuntu 20.04

Hi. I tried to build micro_ros for RP2040 with Arduino support on a board with RP2040 that is not [Arduino Nano RP2040 Connect](https://docs.arduino.cc/hardware/nano-rp2040-connect). I would like not to use custom buildchains and use as much mainstrem code as possible.
I want to connect with my SBC over UART, preferably UART 0, but right now serial over USB is fine. My code is modyfied [micro-ros_publisher.ino](https://github.com/micro-ROS/micro_ros_arduino/blob/galactic/examples/micro-ros_publisher/micro-ros_publisher.ino) without including `#include <micro_ros_arduino.h>` and with `set_microros_transports()` changed to:
``` C
extern "C" bool arduino_transport_open(struct uxrCustomTransport * transport);
extern "C" bool arduino_transport_close(struct uxrCustomTransport * transport);
extern "C" size_t arduino_transport_write(struct uxrCustomTransport* transport, const uint8_t * buf, size_t len, uint8_t * err);
extern "C" size_t arduino_transport_read(struct uxrCustomTransport* transport, uint8_t* buf, size_t len, int timeout, uint8_t* err);

rmw_uros_set_custom_transport(
	true,
	NULL,
	arduino_transport_open,
	arduino_transport_close,
	arduino_transport_write,
	arduino_transport_read
);
```

My `platform.ini` looks as follows:
```
[env:maker_pi_cytron]
platform = raspberrypi
board = pico
framework = arduino
board_build.mcu = rp2040
board_build.f_cpu = 133000000L

build_flags = 
    -L ./.pio/libdeps/maker_pi_cytron/micro_ros_arduino/src/cortex-m0plus/
    -l microros
    -D LIB_PICO_STDIO_USB
    -D PICO_UART_ENABLE_CRLF_SUPPORT=0
    -D PICO_STDIO_ENABLE_CRLF_SUPPORT=0
    -D PICO_STDIO_DEFAULT_CRLF=0

platform_packages =
  toolchain-gccarmnoneeabi @ ~1.70201.0
  framework-arduino-mbed @ ~2.4.1

lib_deps =
    https://github.com/micro-ROS/micro_ros_arduino
```

While trying to build this code I get following warnings and finally build fails. I tried to rebuild the `libmicroros.a` from within the docker but it didn't fix the issue. Previously when I was experimenting with [wizio-pico](https://github.com/Wiz-IO/wizio-pico) the code was building, but the SDK in this repo is too old and I had to move to someting more modern.

Logs below show that the problem might be realted to linking to functions in `cortex-m0plus/libmicroros.a` file. (I had to clip part of the logs)

```
.pio/libdeps/pi_pico/micro_ros_arduino/src/cortex-m0plus/libmicroros.a(libactionlib_msgs__rosidl_typesupport_introspection_c-goal_status__type_support.c.obj): In function `GoalStatus__rosidl_typesupport_introspection_c__GoalStatus_init_function':
goal_status__type_support.c:(.text.GoalStatus__rosidl_typesupport_introspection_c__GoalStatus_init_function+0x0): multiple definition of `GoalStatus__rosidl_typesupport_introspection_c__GoalStatus_init_function'
.pio/libdeps/pi_pico/micro_ros_arduino/src/cortex-m0plus/libmicroros.a(libaction_msgs__rosidl_typesupport_introspection_c-goal_status__type_support.c.obj):goal_status__type_support.c:(.text.GoalStatus__rosidl_typesupport_introspection_c__GoalStatus_init_function+0x0): first defined here
.pio/libdeps/pi_pico/micro_ros_arduino/src/cortex-m0plus/libmicroros.a(libactionlib_msgs__rosidl_typesupport_introspection_c-goal_status__type_support.c.obj): In function `GoalStatus__rosidl_typesupport_introspection_c__GoalStatus_fini_function':
goal_status__type_support.c:(.text.GoalStatus__rosidl_typesupport_introspection_c__GoalStatus_fini_function+0x0): multiple definition of `GoalStatus__rosidl_typesupport_introspection_c__GoalStatus_fini_function'
.pio/libdeps/pi_pico/micro_ros_arduino/src/cortex-m0plus/libmicroros.a(libaction_msgs__rosidl_typesupport_introspection_c-goal_status__type_support.c.obj):goal_status__type_support.c:(.text.GoalStatus__rosidl_typesupport_introspection_c__GoalStatus_fini_function+0x0): first defined here
.pio/libdeps/pi_pico/micro_ros_arduino/src/cortex-m0plus/libmicroros.a(libactionlib_msgs__rosidl_typesupport_introspection_c-goal_status_array__type_support.c.obj): In function `GoalStatusArray__rosidl_typesupport_introspection_c__size_function__GoalStatus__status_list':
goal_status_array__type_support.c:(.text.GoalStatusArray__rosidl_typesupport_introspection_c__size_function__GoalStatus__status_list+0x0): multiple definition of `GoalStatusArray__rosidl_typesupport_introspection_c__size_function__GoalStatus__status_list'
.pio/libdeps/pi_pico/micro_ros_arduino/src/cortex-m0plus/libmicroros.a(libaction_msgs__rosidl_typesupport_introspection_c-goal_status_array__type_support.c.obj):goal_status_array__type_support.c:(.text.GoalStatusArray__rosidl_typesupport_introspection_c__size_function__GoalStatus__status_list+0x0): first defined here
.pio/libdeps/pi_pico/micro_ros_arduino/src/cortex-m0plus/libmicroros.a(libactionlib_msgs__rosidl_typesupport_introspection_c-goal_status_array__type_support.c.obj): In function `GoalStatusArray__rosidl_typesupport_introspection_c__get_const_function__GoalStatus__status_list':
goal_status_array__type_support.c:(.text.GoalStatusArray__rosidl_typesupport_introspection_c__get_const_function__GoalStatus__status_list+0x0): multiple definition of `GoalStatusArray__rosidl_typesupport_introspection_c__get_const_function__GoalStatus__status_list'
.pio/libdeps/pi_pico/micro_ros_arduino/src/cortex-m0plus/libmicroros.a(libaction_msgs__rosidl_typesupport_introspection_c-goal_status_array__type_support.c.obj):goal_status_array__type_support.c:(.text.GoalStatusArray__rosidl_typesupport_introspection_c__get_const_function__GoalStatus__status_list+0x0): first defined here
.pio/libdeps/pi_pico/micro_ros_arduino/src/cortex-m0plus/libmicroros.a(libactionlib_msgs__rosidl_typesupport_introspection_c-goal_status_array__type_support.c.obj): In function `GoalStatusArray__rosidl_typesupport_introspection_c__GoalStatusArray_init_function':
goal_status_array__type_support.c:(.text.GoalStatusArray__rosidl_typesupport_introspection_c__GoalStatusArray_init_function+0x0): multiple definition of `GoalStatusArray__rosidl_typesupport_introspection_c__GoalStatusArray_init_function'
.pio/libdeps/pi_pico/micro_ros_arduino/src/cortex-m0plus/libmicroros.a(libaction_msgs__rosidl_typesupport_introspection_c-goal_status_array__type_support.c.obj):goal_status_array__type_support.c:(.text.GoalStatusArray__rosidl_typesupport_introspection_c__GoalStatusArray_init_function+0x0): first defined here
.pio/libdeps/pi_pico/micro_ros_arduino/src/cortex-m0plus/libmicroros.a(libactionlib_msgs__rosidl_typesupport_introspection_c-goal_status_array__type_support.c.obj): In function `GoalStatusArray__rosidl_typesupport_introspection_c__GoalStatusArray_fini_function':
goal_status_array__type_support.c:(.text.GoalStatusArray__rosidl_typesupport_introspection_c__GoalStatusArray_fini_function+0x0): multiple definition of `GoalStatusArray__rosidl_typesupport_introspection_c__GoalStatusArray_fini_function'
[... 198 lines cut ...]
multi_nested__type_support.c:(.text.rosidl_typesupport_introspection_c__get_message_type_support_handle__test_msgs__msg__MultiNested+0x30): undefined reference to `rosidl_typesupport_introspection_c__get_message_type_support_handle__test_msgs__msg__Arrays'
.pio/libdeps/pi_pico/micro_ros_arduino/src/cortex-m0plus/libmicroros.a(libtest_msgs__rosidl_typesupport_introspection_c-nested__type_support.c.obj): In function `rosidl_typesupport_introspection_c__get_message_type_support_handle__test_msgs__msg__Nested':
nested__type_support.c:(.text.rosidl_typesupport_introspection_c__get_message_type_support_handle__test_msgs__msg__Nested+0x2): undefined reference to `rosidl_typesupport_introspection_c__get_message_type_support_handle__test_msgs__msg__BasicTypes'
.pio/libdeps/pi_pico/micro_ros_arduino/src/cortex-m0plus/libmicroros.a(libtest_msgs__rosidl_typesupport_introspection_c-nested_message__type_support.c.obj): In function `rosidl_typesupport_introspection_c__get_message_type_support_handle__test_msgs__action__NestedMessage_Goal':
nested_message__type_support.c:(.text.rosidl_typesupport_introspection_c__get_message_type_support_handle__test_msgs__action__NestedMessage_Goal+0xa): undefined reference to `rosidl_typesupport_introspection_c__get_message_type_support_handle__test_msgs__msg__BasicTypes'
[... cut ...]
unbounded_sequences__type_support_c.c:(.text._UnboundedSequences__cdr_deserialize+0x1ec): undefined reference to `rosidl_typesupport_microxrcedds_c__get_message_type_support_handle__test_msgs__msg__BasicTypes'
.pio/libdeps/pi_pico/micro_ros_arduino/src/cortex-m0plus/libmicroros.a(libtest_msgs__rosidl_typesupport_microxrcedds_c-unbounded_sequences__type_support_c.c.obj): In function `_UnboundedSequences__cdr_serialize':
unbounded_sequences__type_support_c.c:(.text._UnboundedSequences__cdr_serialize+0x12a): undefined reference to `rosidl_typesupport_microxrcedds_c__get_message_type_support_handle__test_msgs__msg__BasicTypes'
```

pablogs9, 2022-04-04:
Please copy here the whole build output log.

Can you share your whole code?
Why are you removing `#include <micro_ros_arduino.h>`?

Kotochleb, 2022-04-04:
Here is `main.cpp` file:
https://github.com/Kotochleb/microros_rp2040_robot/blob/main/src/main.cpp

I am not including `micro_ros_arduino.h` since it does not add anything to the code. In case of specifying Raspberry Pi Pico it will force transport layer to be over wifi. If I don't define any target architecture, from what I understand from source code, it won't do anything. That is why I am defining transport layer myself.

In case of build log, the problem is that I exceed a boundary of how long my message on GitHub can be, and I tried to clip is as little as I could. Most of the logs are repetitive.

pablogs9, 2022-04-04:
In some Arduino environments the include makes the build system to resolve the library: 

```bash
Using board 'teensy41' from platform in folder: /home/pgarrido/.arduino15/packages/teensy/avr
Using core 'teensy4' from platform in folder: /home/pgarrido/.arduino15/packages/teensy/avr
Detecting libraries used...
/home/pgarrido/.arduino15/packages/teensy/../tools/arm/bin/arm-none-eabi-g++ -E -CC -x c++ -w -g -Wall -ffunction-sections -fdata-sections -nostdlib -std=gnu++14 -fno-exceptions -fpermissive -fno-rtti -fno-threadsafe-statics -felide-constructors -Wno-error=narrowing -mthumb -mcpu=cortex-m7 -mfloat-abi=hard -mfpu=fpv5-d16 -D__IMXRT1062__ -DTEENSYDUINO=153 -DARDUINO=10607 -DARDUINO_TEENSY41 -DF_CPU=600000000 -DUSB_SERIAL -DLAYOUT_US_ENGLISH -I/home/pgarrido/.arduino15/packages/teensy/avr/cores/teensy4 /tmp/arduino-sketch-776FACD3F9198D5291708649B4419502/sketch/micro-ros_reconnection_example.ino.cpp -o /dev/null

CHECK HERE:

Alternatives for micro_ros_arduino.h: [micro_ros_arduino-0.0.1@2.0.4-rolling]
ResolveLibrary(micro_ros_arduino.h)
  -> candidates: [micro_ros_arduino-0.0.1@2.0.4-rolling]
```

Maybe if you remove this line, the build system is not including the library and that's why the linker does not find the symbols.

Kotochleb, 2022-04-04:
I included the `micro_ros_arduino.h` and I got error messages related to wifi.
I also added `-U ARDUINO_NANO_RP2040_CONNECT` to ensure that this platform won't be compiled. But eventually, it still wants to include wifi anyway.

```
<command-line>:0:1: error: macro names must be identifiers
<command-line>:0:2: warning: extra tokens at end of #undef directive
Compiling .pio/build/maker_pi_cytron/lib0cf/SocketWrapper/MbedSSLClient.cpp.o
<command-line>:0:1: error: macro names must be identifiers
<command-line>:0:2: warning: extra tokens at end of #undef directive
Compiling .pio/build/maker_pi_cytron/lib0cf/SocketWrapper/MbedServer.cpp.o
<command-line>:0:1: error: macro names must be identifiers
<command-line>:0:2: warning: extra tokens at end of #undef directive
Compiling .pio/build/maker_pi_cytron/lib0cf/SocketWrapper/MbedUdp.cpp.o
<command-line>:0:1: error: macro names must be identifiers
<command-line>:0:2: warning: extra tokens at end of #undef directive
Compiling .pio/build/maker_pi_cytron/lib0cf/SocketWrapper/SocketHelpers.cpp.o
<command-line>:0:1: error: macro names must be identifiers
<command-line>:0:2: warning: extra tokens at end of #undef directive
Compiling .pio/build/maker_pi_cytron/lib0cf/SocketWrapper/utility/http_parser/http_parser.c.o
Compiling .pio/build/maker_pi_cytron/liba84/Ethernet/Ethernet.cpp.o
<command-line>:0:1: error: macro names must be identifiers
<command-line>:0:1: error: macro names must be identifiers
<command-line>:0:2: warning: extra tokens at end of #undef directive
<command-line>:0:2: warning: extra tokens at end of #undef directive
<command-line>:0:1: error: macro names must be identifiers
<command-line>:0:2: warning: extra tokens at end of #undef directive
/home/user/.platformio/packages/framework-arduino-mbed@2.4.1/libraries/SocketWrapper/src/utility/http_parser/http_parser.c: In function 'http_parser_execute':
/home/user/.platformio/packages/framework-arduino-mbed@2.4.1/libraries/SocketWrapper/src/utility/http_parser/http_parser.c:1817:31: warning: this statement may fall through [-Wimplicit-fallthrough=]
               parser->upgrade = 1;
               ~~~~~~~~~~~~~~~~^~~
/home/user/.platformio/packages/framework-arduino-mbed@2.4.1/libraries/SocketWrapper/src/utility/http_parser/http_parser.c:1819:13: note: here
             case 1:
             ^~~~
/home/user/.platformio/packages/framework-arduino-mbed@2.4.1/libraries/SocketWrapper/src/utility/http_parser/http_parser.c: In function 'http_parser_parse_url':
/home/user/.platformio/packages/framework-arduino-mbed@2.4.1/libraries/SocketWrapper/src/utility/http_parser/http_parser.c:2376:18: warning: this statement may fall through [-Wimplicit-fallthrough=]
         found_at = 1;
         ~~~~~~~~~^~~
/home/user/.platformio/packages/framework-arduino-mbed@2.4.1/libraries/SocketWrapper/src/utility/http_parser/http_parser.c:2379:7: note: here
       case s_req_server:
       ^~~~
*** [.pio/build/maker_pi_cytron/lib0cf/SocketWrapper/utility/http_parser/http_parser.c.o] Error 1
In file included from /home/user/.platformio/packages/framework-arduino-mbed@2.4.1/libraries/SocketWrapper/src/MbedSSLClient.cpp:1:0:
/home/user/.platformio/packages/framework-arduino-mbed@2.4.1/libraries/SocketWrapper/src/MbedSSLClient.h: In member function 'int arduino::MbedSSLClient::setRootCA()':
/home/user/.platformio/packages/framework-arduino-mbed@2.4.1/libraries/SocketWrapper/src/MbedSSLClient.h:46:14: error: 'TLSSocket' was not declared in this scope
     return ((TLSSocket*)sock)->set_root_ca_cert_path("/wlan/");
              ^~~~~~~~~
/home/user/.platformio/packages/framework-arduino-mbed@2.4.1/libraries/SocketWrapper/src/MbedSSLClient.h:46:14: note: suggested alternative: 'DTLSSocket'
     return ((TLSSocket*)sock)->set_root_ca_cert_path("/wlan/");
              ^~~~~~~~~
              DTLSSocket
/home/user/.platformio/packages/framework-arduino-mbed@2.4.1/libraries/SocketWrapper/src/MbedSSLClient.h:46:24: error: expected primary-expression before ')' token
     return ((TLSSocket*)sock)->set_root_ca_cert_path("/wlan/");
                        ^
/home/user/.platformio/packages/framework-arduino-mbed@2.4.1/libraries/SocketWrapper/src/MbedSSLClient.h:46:25: error: expected ')' before 'sock'
     return ((TLSSocket*)sock)->set_root_ca_cert_path("/wlan/");
                         ^~~~
*** [.pio/build/maker_pi_cytron/src/main.cpp.o] Error 1
*** [.pio/build/maker_pi_cytron/lib0cf/SocketWrapper/MbedSSLClient.cpp.o] Error 1
/home/user/.platformio/packages/framework-arduino-mbed@2.4.1/libraries/SocketWrapper/src/MbedClient.cpp: In member function 'int arduino::MbedClient::connectSSL(SocketAddress)':
/home/user/.platformio/packages/framework-arduino-mbed@2.4.1/libraries/SocketWrapper/src/MbedClient.cpp:123:16: error: expected type-specifier before 'TLSSocket'
     sock = new TLSSocket();
                ^~~~~~~~~
/home/user/.platformio/packages/framework-arduino-mbed@2.4.1/libraries/SocketWrapper/src/MbedClient.cpp:134:19: error: 'TLSSocket' does not name a type; did you mean 'DTLSSocket'?
   if (static_cast<TLSSocket *>(sock)->open(getNetwork()) != NSAPI_ERROR_OK) {
                   ^~~~~~~~~
                   DTLSSocket
/home/user/.platformio/packages/framework-arduino-mbed@2.4.1/libraries/SocketWrapper/src/MbedClient.cpp:134:29: error: expected '>' before '*' token
   if (static_cast<TLSSocket *>(sock)->open(getNetwork()) != NSAPI_ERROR_OK) {
                             ^
[... cut ...]
         return ((TLSSocket*)_socket)->connect(socketAddress);
                            ^
/home/user/.platformio/packages/framework-arduino-mbed@2.4.1/libraries/SocketWrapper/src/utility/https_request.h:103:29: error: expected ')' before '_socket'
         return ((TLSSocket*)_socket)->connect(socketAddress);
                             ^~~~~~~
*** [.pio/build/maker_pi_cytron/liba84/Ethernet/Ethernet.cpp.o] Error 1
*** [.pio/build/maker_pi_cytron/lib0cf/SocketWrapper/SocketHelpers.cpp.o] Error 1
```

pablogs9, 2022-04-04:
Maybe you should add another target that is a bare RP2040 and is not related to RP2040 Nano. If you make those modifications in `micro_ros_arduino.h` do not hesitate to contribute them via PR.

Kotochleb, 2022-04-04:
Sure. I will fork it and try to implement it.

Kotochleb, 2022-04-07:
Ok. I made it work over USB.
So far `micro_ros_arduino.h` includes anything with wifi even though preprocessor should omit those includes. I will take a deeper look into that on the weekend.
It turned out the linker problem was solved in https://github.com/micro-ROS/micro_ros_arduino/issues/774.

With all that, I got nice 100 Hz published over USB. Right now, I will clean up all the stuff, make the code look better, do some more testing and hopefully the PR will be open in two weeks.

pablogs9, 2022-04-08:
Nice, I'm closing. Please reopen if you have any related issue,

closed 2022-04-08 by pablogs9