micro-ROS/micro_ros_arduino #911: micro_ros esp32

Sarwan1494, 2022-04-03:
## Issue template

I have installed micro_ros_arduino using the include library option for version foxy. I am getting the following error while uploading the arduino code.  kindly inform how to solve this issue. 

[image]

- Hardware description: esp32
- Installation type: micro_ros_arduino 
- Version or commit hash: foxy

pablogs9, 2022-04-04:
@Acuadros95 can you take a look?

Acuadros95, 2022-04-05:
Latest release `v2.0.4-foxy` worked right ahead for me.

Could you check your esp32 board package version? micro-ROS is supported on the latest version 2.0.2. 
Related: https://github.com/micro-ROS/micro_ros_arduino/issues/814

Sarwan1494, 2022-04-05:
I updated the Arduino esp32 board package and it works fine. Thanks a lot

closed 2022-04-05 by pablogs9