micro-ROS/micro_ros_arduino #912: Unexpected behavior of reconnection example

Johannap1, 2022-04-03:
- Hardware description: ESP32dev
- microros arduino + platformio
- Version or commit hash: ROS2 foxy

Hello, 

I tried out the[ reconnection example](https://github.com/micro-ROS/micro_ros_arduino/blob/galactic/examples/micro-ros_reconnection_example/micro-ros_reconnection_example.ino)  because I want to use something similar for my application, but it is not working as I would expect it to work. 

What is happening for me is that rmw_uros_ping_agent only returns RMW_RET_OK when the entities have not yet been created. As soon as I create the entities,  rmw_uros_ping_agent returns an error, even thought the agent is still running. Running the example this way causes just a continuous creation and destruction of the entities and the system never starts spinning the executor.

What I would expect is that rmw_uros_ping_agent returns a true value as long as the agent is connected, then creates the entities once and at each later iteration spins the executor. The entities should only be destroyed once the connection to the agent is actually lost. 

Please let me know if I understand correct that this behavior is not desired and if you have any ideas on how to fix this issue. I already tried increasing the timeout and number of connection tries but that does not change the behavior. I also tried modifying the colcon.meta as suggested [here ](https://github.com/micro-ROS/micro_ros_setup/issues/370), but that also did not change anything. 

Thanks!

pablogs9, 2022-04-04:
Which version of the library and the agentt are you using?
Make sure that you are using the latest commit of this library and also that you are using the latest version of the agent.

Johannap1, 2022-04-04:
I just updated my version of the agent. For the library, I just import it using platformio as it is also stated on the README of this repository:

```
[env:esp32dev]
platform = https://github.com/platformio/platform-espressif32.git#feature/arduino-upstream
board = esp32dev
framework = arduino
lib_deps =
    https://github.com/micro-ROS/micro_ros_arduino.git
build_flags =
    -L ./.pio/libdeps/esp32dev/micro_ros_arduino/src/esp32/
    -l microros
    -D ESP32

platform_packages =
  toolchain-xtensa32 @ ~2.80400.0
  framework-arduinoespressif32@https://github.com/espressif/arduino-esp32.git#2.0.2
```

I also just tried creating a new blank project and then adding the reconnection example code, but the issue remains.

pablogs9, 2022-04-04:
no, but make sure that you are using the latest commit of the foxy branch. 

@Acuadros95 does the last release (2.0.2) includes the modifications on the XRCE ping behavior?

Johannap1, 2022-04-04:
Yes, I checked that we are using the latest commit so that should not be the issue, see below:

[image]

pablogs9, 2022-04-04:
Can you give a try this reimplementation of the reconnection example? https://github.com/micro-ROS/micro_ros_arduino/pull/913

I have tried both (old and #913) using serial transport, a Teensy 4 and launching the agent with:

```bash

Johannap1, 2022-04-04:
Thanks, this works now for me. Also with both the old and new version of it. So I think it is the way that I started up the agent 

normally I just run it without the docker using 

```
ros2 run micro_ros_agent micro_ros_agent serial --dev /dev/ttyUSB0
```
Can you tell me what the difference is, please?

pablogs9, 2022-04-04:
Well, using this command you are using a local workspace where the agent is built. Probably you need to recreate the micro-ROS Agent and rebuild it in order to have the latest version.

The docker usually has the latest version.

I'm closing please reopen if you have any other questions.

closed 2022-04-04 by pablogs9