micro-ROS/micro_ros_arduino #920: micro ros - esp32 wifi connection

Sarwan1494, 2022-04-05:
hello, 

I followed the tutorials and installed the necessary packages. Using serial connection the micro ros Arduino library is working fine, but I cannot connect to the wifi. I have listed the commands i used and the results. Kindly inform what am I doing wrong or if I missed any installation process

command: **ros2 run micro_ros_setup create_firmware_ws.sh freertos esp32**
Firmware already created. Please delete /home/sarwan/micro_ros_foxy/firmware folder if you want a fresh installation.

command: **ros2 run micro_ros_setup build_firmware.sh menuconfig**
Build process proceeds and no window is opened for wifi configuration.

command: **ros2 run micro_ros_setup flash_firmware.sh**
Flashing firmware for host platform generic
Nothing to flash: running in host mode
Use 'RMW_IMPLEMENTATION=rmw_microxrcedds ros2 run micro_ros_demos_rcl int32_subscriber' to start with examples

Thanks

Acuadros95, 2022-04-06:
Hi @Sarwan1494,

This issue is open on the `micro_ros_arduino` repo, but you are using the [micro_ros_setup](https://github.com/micro-ROS/micro_ros_setup) tool. You can also try this recommended alternatives:
- Use `micro_ros_arduino` wifi example: [link](https://github.com/micro-ROS/micro_ros_arduino/blob/galactic/examples/micro-ros_publisher_wifi/micro-ros_publisher_wifi.ino)
- Use [micro-ROS component for ESP-IDF](https://github.com/micro-ROS/micro_ros_espidf_component), which integrates micro-ROS on Espressif ESP32 IDF projects

Anyway, let me check the `micro_ros_setup` commands.

Acuadros95, 2022-04-06:
Check the following messages on your build proccess:
> Firmware already created. Please delete /home/sarwan/micro_ros_foxy/firmware folder if you want a fresh installation.
...
> Nothing to flash: running in host mode.

- This means that you had a firmware folder already created, I guess you used the same environment to test a micro-ROS example on Linux. 
- You need to delete the `firmware` folder before calling `ros2 run micro_ros_setup create_firmware_ws.sh freertos esp32`, otherwise the build system will build for linux again.

Once you have deleted the firmware folder, you can run this commands to build and flash a wifi example:

```bash
# Prepare environment for esp32 build
ros2 run micro_ros_setup create_firmware_ws.sh freertos esp32

# Configure APP and transport: Modify add_two_ints_service with your APP name
ros2 run micro_ros_setup configure_firmware.sh ping_pong -t udp

# Launch menu for wifi and general configuration and build
ros2 run micro_ros_setup build_firmware.sh menuconfig

# Flash the binary
ros2 run micro_ros_setup flash_firmware.sh

```

Notice that you were missing the `configure` step, which selects the app to compile and the transport kind to use (serial or udp)

Sarwan1494, 2022-04-10:
Hello,

Thanks a lot for the prompt reply. It works perfectly now.

closed 2022-04-11 by Acuadros95