micro-ROS/micro_ros_arduino #922: Micro-ros client connected to agent but stop publishing after several reconnections.

NoyZuberi, 2022-04-06:
Hi,

I created topic and published it from the client, I saw that the client succeed to connect to the agent and on `ros2 topic echo` I got the topic data as I should. But, after I restarted the client several times it stopped publish any data.

From the client view, it's look like there is no problem. The API `rcl_publish` keep being called on a loop and returned `RCL_RET_OK` on each publish.

Do you know maybe what might be the issue here?
Thanks.

pablogs9, 2022-04-06:
Please provide the code and details about your platform/transport in order to allow us to replicate the issue.

pablogs9, 2022-04-07:
Sorry but without a replicable environment we cannot do much more. Please provide an example (out of your project if needed) and steps for replicating the problem. 

Also explain in detail the situation, in this case I do not fully understand what do you mean by "after I restarted the client several times", that means connecting and disconecting the MCU power? What is the micro-ROS Agent behaviour?

NoyZuberi, 2022-04-07:
Of course, I understand, I will try to elaborate a little more.

Our setup is an MCU which is connected using UART to a Raspberry Pi. The micro-ROS agent is running constantly using systemd. That means, that once the MCU is reset, the agent will automatically reconnect itself once the port is available once more.

I think that the replicable environment, is pretty much an ARM M7 MCU running the micro-ROS client, with a serial transport over UART at a baudrate of 921600. In case the connection with the micro-ROS agent is valid, it will call the rclc_executor_spin_some function, and publish a ROS message. Like the one you can see in the files attached in the previous post. It is basically an array of 12 UInt16 members.

NoyZuberi, 2022-04-14:
It's look like the issue was with the timers been defined as signed int64 and not unsigned int64.
I ran with cherry-pick to [fujitatomoya](https://github.com/fujitatomoya/rcl/commits?author=fujitatomoya) [fix](https://github.com/ros2/rcl/pull/974) for this issue https://github.com/ros2/rcl/issues/967, and it solved the issue.

Is there any chance you can merge the following fixes to micro-ROS/rcl?
1. [Add additional null check for timer argument](https://github.com/ros2/rcl/pull/973)
2. [use uint64_t for timer period](https://github.com/ros2/rcl/pull/974)

Thanks.

pablogs9, 2022-04-18:
[micro-ROS/rcl](https://github.com/micro-ROS/rcl) is always on top of ros2/rcl, so 1. is already merged in micro-ROS and 2. will be after it is merged into ros2/rcl.

closed 2022-04-18 by pablogs9