micro-ROS/micro_ros_arduino #930: ESP32 WiFi connection problem

WhateverKatzenfutter, 2022-04-10:
## Issue template

- Hardware description: [ESP32](https://learn.adafruit.com/adafruit-huzzah32-esp32-feather)
- Installation type: Arduino IDE precompiled library
- Version or commit hash: [v2.0.4-galactic](https://github.com/micro-ROS/micro_ros_arduino/releases/tag/v2.0.4-galactic)  

#### Steps to reproduce the issue
- Follow the "How to use the precompiled library" instruction.
- Flash the micro-ros_publisher_wifi.ino on to the ESP32.
- Start micro ros agent via docker: docker run -it --net=host microros/micro-ros-agent:galactic udp4 -p 8888

#### Expected behavior
- ESP32 connects with the WiFi and establishes a connection with the micro ros agent.

#### Actual behavior
- ESP32 is connecting to the WiFi: 
[image]

- No connection with the micro ros agent gets established:
[image]

#### Additional information
Hey guys, Iam pretty new to ROS at all and therefore not really sure if I got the installation steps right here, but I hope you can help me.

In the micro-ros_publisher_wifi.ino example for the set_microros_wifi_transports function Iam using the IP adress of my local machine where micro-ros agent is running via the docker and the port 8888, this should be right or?

[image]

Acuadros95, 2022-04-11:
> PS: Iam using Windows 10, should I better switch to Linux for this kind of work?

Yes, micro-ROS agent doesnt support serial transport on Windows, the wifi problems may also be related to docker issues with windows. Can you try on Linux?

Anyway, check the network traffic with wireshark or similar. The esp32 should send a periodic message trying to init a session to the configured Agent IP and port.

WhateverKatzenfutter, 2022-04-11:
Thanks for the idea to check the network traffic, but I think I will just switch to Linux and report back, if there are still some problems.

closed 2022-04-11 by WhateverKatzenfutter