micro-ROS/micro_ros_arduino #932: line order of generated files

amilcarlucas, 2022-04-11:
## Issue template

- Hardware description: Arduino Portenta
- RTOS: Mbed
- Installation type: node, using micro_ros_arduino directly
- Version or commit hash: latest galactic

#### Steps to reproduce the issue
Just compile it using the instructions on the README.md file

#### Expected behavior
The generated `available_ros2_types` and `build_packages` files should a bit bit similar to the ones in the checked in at the git repository.

#### Actual behavior
The lines in the new generated files are out of order, comparing with the repository files is very hard

#### Additional information
It should either:
* produce a reproducible output order,
* or sort the files (during or after generation)

This way comparing with the repo is straightforward

pablogs9, 2022-04-11:
That's a known issue, we do not have yet a way of producing a reproducible order. Please PR if you have any solution in mind.

amilcarlucas, 2022-04-11:
done and tested

PR #934 (2022-04-11): sort available_ros2_types and built_packages so that the build result order is reproducible
Fixes #932
closed 2022-04-11 by pablogs9