micro-ROS/micro_ros_arduino #940: micro_ros_arduino + Teensy4.1 + CAN

scch1092, 2022-04-12:
Hello everyone, 

I currently have the problem that when I upload my code, an error occurs. The programs for CAN and micro_ros work independently. Together in one code, however, no longer. For my CAN library I get the error "undefined reference for ...".

The CAN library of tonton81 "FlexCAN_4T" (https://github.com/tonton81/FlexCAN_T4) is used. 

- Hardware description: teensy41;  imxrt1062/fpv5-d16-hard (PlatformIO)
- Installation type: build_flags =
    -L ./.pio/libdeps/teensy41/micro_ros_arduino/src/imxrt1062/fpv5-d16-hard/
    -l microros
    -D ARDUINO_TEENSY41
- Version or commit hash: 2.0.4-galactic+sha.3df8bba

#### Steps to reproduce the issue
Load a microros example and add the CAN library:
#include <micro_ros_arduino.h>
#include <stdio.h>
#include <rcl/rcl.h>
#include <rcl/error_handling.h>
#include <rclc/rclc.h>
#include <rclc/executor.h>
#include <std_msgs/msg/int32.h>
#include <FlexCAN_T4.h>

#### Actual behavior
 In function FlexCAN_T4<...>::flexcan_interrupt() [....] undefined reference to ext_output1(CAN_message_t const&) 

Thanks in advance

pablogs9, 2022-04-12:
Not related to micro-ROS. Please reopen if a relation between this issue and micro-ROS for Arduino library is found.

pablogs9, 2022-04-12:
For some platforms, our library redefines the `__attribute()__`. Check here: https://github.com/micro-ROS/micro_ros_arduino/blob/3df8bba318a7e19fc6077c23bb42f0cf9eb6852e/src/micro_ros_arduino.h#L7

The error mentions a function that is defined as `__attribute__((weak))`: https://github.com/tonton81/FlexCAN_T4/blob/98654c7844408b536a44c3f3693a9c9a7335b21a/FlexCAN_T4.tpp#L1804

Try to just remove [this line](https://github.com/micro-ROS/micro_ros_arduino/blob/3df8bba318a7e19fc6077c23bb42f0cf9eb6852e/src/micro_ros_arduino.h#L7) and recompile.

pablogs9, 2022-04-12:
Also is possible to include this header before micro-ROS for Arduino:

```c
#include <FlexCAN_T4.h>
#include <micro_ros_arduino.h>
#include <stdio.h>
#include <rcl/rcl.h>
#include <rcl/error_handling.h>
#include <rclc/rclc.h>
#include <rclc/executor.h>
#include <std_msgs/msg/int32.h>
```

In any case, I'm working in a solution

scch1092, 2022-04-12:
> Also is possible to include this header before micro-ROS for Arduino:
>
> ```c
> #include <FlexCAN_T4.h>
> #include <micro_ros_arduino.h>
> #include <stdio.h>
> #include <rcl/rcl.h>
> #include [...]

wait a minute... somehow it does work

pablogs9, 2022-04-12:
This is a possible solution, I have fixed this here: https://github.com/micro-ROS/micro_ros_arduino/pull/941

closed 2022-04-12 by pablogs9