micro-ROS/micro_ros_arduino #945: transform_broadcaster.h library

v-mifass, 2022-04-13:
Hello I am working to move a project from ROS1 to ROS2. Part of this is moving from rosserial to micro-ros. I have a couple sensors implemented and visible in rviz2 but have been stuck at adding the transform for the range sensors.

Similar examples show "#include <tf2_ros transform_broadcaster.h="">" but there is no tf2_ros in the imported micro_ros_arduino library. It is in geometry2 (tf2_ros/include/tf2_ros) but I get other include/dependency errors when I manually add this.

I am trying to locate the origin and orientation of a range sensor, I can do this easily with a marker display (to represent a switch) as it has 'pose' contained within it.

I put this on ros answers: https://answers.ros.org/question/398967/transform_broadcasterh-library-on-micro-ros-arduino/ and was directed to github so trying here.  Not sure if I am missing something obvious or of this is something not yet implemented...

## Issue template
- Hardware description:  Teensy4.1
- RTOS: micro-ros arduino
- Installation type: PlatformIO within VisualStudioCode
- Version or commit hash: galactic 2.0.4

Acuadros95, 2022-04-18:
Hi @v-mifass,

>  It is in geometry2 (tf2_ros/include/tf2_ros) but I get other include/dependency errors when I manually add this.
Could you give more insight on this? Did you include this package on the extra repos and the build fails?
- If you didn't include it, check this section of the Readme:  [How to build the precompiled library](https://github.com/micro-ROS/micro_ros_arduino#how-to-build-the-precompiled-library)
- If the geometry2 package build is failing, you can just add the colcon package with the necessary messages, avoiding unnecessary packages which may have unmet dependencies.

v-mifass, 2022-04-19:
I have tried adding the library to the extra repos in platformio but don't see the files show up in the src folder so I will try to do what I want on the PC side instead.  

This issue can be closed

Acuadros95, 2022-04-20:
How did you try? This should be doable.

You can also test our new platformIO approach: [micro_ros_platformio](https://github.com/micro-ROS/micro_ros_platformio). It also includes the `extra_packages` functionality. 
Please give it a go and open a issue if needed, help will be granted.

v-mifass, 2022-04-20:
I tried adding geometry2 by both copying the downloaded files into extras/library_generation/extra_packages and adding the link to extras/library_generation/extra_packages/extra_packages.repos.  I never saw the tf2_ros2 folder show along with the compiled src files nor did my code didn't find the transfer_broadcaster files (*.h or *.cpp).

I also tried micro_ros_platformio but am getting a library building error.  I appreciate the offer for help but worry my software experience is bit too low for working with such fresh code;)

Acuadros95, 2022-04-21:
Did you rebuild the library as explained in this section: [How to build the precompiled library](https://github.com/micro-ROS/micro_ros_arduino#how-to-build-the-precompiled-library)(

> I also tried micro_ros_platformio but am getting a library building error. I appreciate the offer for help but worry my software experience is bit too low for working with such fresh code;)
I remember having problems building the complete `geometry2` package, maybe you can try only adding the folders with the needed msgs, instead of the whole repository.

pablogs9, 2022-05-17:
Any update on this? @v-mifass

pablogs9, 2022-06-14:
Closing due to inactivity

closed 2022-06-14 by pablogs9