micro-ROS/micro_ros_arduino #952: Compiling the library with docker image fails. Cannot delete /project/src: Directory not empty

sejego, 2022-04-20:
## Issue template

- Hardware description: Teensy 4.0
- Installation type: Docker
- Version or commit hash: foxy

I am trying to compile the library to use with custom messages. Following the tutorials I did the following:
1. Cloned the repository of the micro_ros_arduino
2. cd into micro_ros_arduino and added my custom message package into /extras
3. Used following docker commands to start the compilation
`sudo docker run -it --rm -v $(pwd):/project --env MICROROS_LIBRARY_FOLDER=extras microros/micro_ros_static_library_builder:foxy -p teensy40`

`sudo docker pull microros/micro_ros_static_library_builder:foxy`

However, it seems that it cannot overwrite files present in /src folder. I tried to manually removing files inside the /src, the whole folder altogether, but it seems to not work at all. Am I doing something wrong, or it is a bug in the compilation commands inside the docker image?
Here is the output from the terminal:
```
sejego@pop-os:~/dev_ws/microcontroller/micro_ros_arduino$ sudo docker run -it --rm -v $(pwd):/project --env MICROROS_LIBRARY_FOLDER=extras microros/micro_ros_static_library_builder:foxy -p teensy40
Get:1 http://archive.ubuntu.com/ubuntu focal InRelease [265 kB]
Get:2 http://security.ubuntu.com/ubuntu focal-security InRelease [114 kB]      
Get:3 http://archive.ubuntu.com/ubuntu focal-updates InRelease [114 kB]        
Get:4 http://archive.ubuntu.com/ubuntu focal-backports InRelease [108 kB]      
Get:5 http://security.ubuntu.com/ubuntu focal-security/restricted amd64 Packages [1139 kB]
Get:6 http://archive.ubuntu.com/ubuntu focal/universe amd64 Packages [11.3 MB]
Get:7 http://security.ubuntu.com/ubuntu focal-security/universe amd64 Packages [869 kB]
Get:8 http://packages.ros.org/ros2/ubuntu focal InRelease [4676 B]             
Get:9 http://security.ubuntu.com/ubuntu focal-security/multiverse amd64 Packages [25.8 kB]
Get:10 http://security.ubuntu.com/ubuntu focal-security/main amd64 Packages [1769 kB]
Get:11 http://packages.ros.org/ros2/ubuntu focal/main amd64 Packages [1012 kB] 
Get:12 http://archive.ubuntu.com/ubuntu focal/multiverse amd64 Packages [177 kB]
Get:13 http://archive.ubuntu.com/ubuntu focal/restricted amd64 Packages [33.4 kB]
Get:14 http://archive.ubuntu.com/ubuntu focal/main amd64 Packages [1275 kB]    
Get:15 http://archive.ubuntu.com/ubuntu focal-updates/main amd64 Packages [2186 kB]
Get:16 http://archive.ubuntu.com/ubuntu focal-updates/universe amd64 Packages [1153 kB]
Get:17 http://archive.ubuntu.com/ubuntu focal-updates/restricted amd64 Packages [1214 kB]
Get:18 http://archive.ubuntu.com/ubuntu focal-updates/multiverse amd64 Packages [30.3 kB]
Get:19 http://archive.ubuntu.com/ubuntu focal-backports/universe amd64 Packages [26.0 kB]
Get:20 http://archive.ubuntu.com/ubuntu focal-backports/main amd64 Packages [51.2 kB]
Fetched 22.9 MB in 4s (5877 kB/s)                                              
Reading package lists... Done
Building dependency tree       
Reading state information... Done
184 packages can be upgraded. Run 'apt list --upgradable' to see them.
Creating firmware for generate_lib platform generic
.......
=== ./ament/ament_cmake (git) ===
Cloning into '.'...
=== ./ament/ament_index (git) ===
Cloning into '.'...
=== ./ament/ament_lint (git) ===
Cloning into '.'...
=== ./ament/ament_package (git) ===
Cloning into '.'...
=== ./ament/googletest (git) ===
Cloning into '.'...
=== ./ament/uncrustify_vendor (git) ===
Cloning into '.'...
=== ./ros2/ament_cmake_ros (git) ===
Cloning into '.'...
List of repositories is empty
#All required rosdeps installed successfully
..............
=== ./ros2/common_interfaces (git) ===
Cloning into '.'...
=== ./ros2/example_interfaces (git) ===
Cloning into '.'...
=== ./ros2/libyaml_vendor (git) ===
Cloning into '.'...
=== ./ros2/rcl (git) ===
Cloning into '.'...
[... 156 lines cut ...]
Starting >>> ament_cmake_pyflakes
Finished <<< ament_cmake_mypy [0.96s]                                     
Starting >>> ament_cmake_ros
Finished <<< ament_cmake_pclint [0.93s]                                     
Starting >>> ament_index_cpp
Finished <<< ament_cmake_pycodestyle [0.94s]
Finished <<< ament_cmake_pyflakes [0.82s]                                     
Finished <<< ament_cmake_ros [0.94s]                                     
Finished <<< ament_index_cpp [8.84s]                        

Summary: 60 packages finished [25.2s]
#All required rosdeps installed successfully
#All required rosdeps installed successfully
Cloning into 'geometry2'...
remote: Enumerating objects: 14646, done.
remote: Counting objects: 100% (1805/1805), done.
remote: Compressing objects: 100% (350/350), done.
remote: Total 14646 (delta 1624), reused 1494 (delta 1454), pack-reused 12841
Receiving objects: 100% (14646/14646), 3.29 MiB | 5.67 MiB/s, done.
Resolving deltas: 100% (10160/10160), done.
=== ./control_msgs (git) ===
Cloning into '.'...
find: cannot delete ‘/project/src/’: Directory not empty

```

Acuadros95, 2022-04-20:
The error can be actually ignored, the problem is the docker builder argument `-p teensy40`.
You should use `-p teensy4` instead:
https://github.com/micro-ROS/micro_ros_arduino/blob/cace9cc291359e6b2221bac3b0d11c250c885c2e/extras/library_generation/library_generation.sh#L14

sejego, 2022-04-20:
@Acuadros95 Thank you and sorry for such a silly mistake. Everything works fine! Issue can be closed now.