micro-ROS/micro_ros_arduino #955: Several publishers of same message type won't work

SimonGodon, 2022-04-21:
I have a sketch that creates a node that includes:
3 publishers (2 of them have same message type "measurementMessage"')
1 subscriber

The problem is that there is an error every time I try to add the 2 publishers that have the measurementMessage, the second one will not get created.  All the rest works and if I remove one of the two measurementMessage, it works. The 2 publishers are identical besides the topic name so it's not a typo of anything like that. 

The problem is at this line, for the second publisher of the same message type: 
rcl_ret_t rc = rclc_publisher_init_default(
  &publisher, &node,
  &type_support, &topic_name);

If I invert the order in which I create the publishers, it's still the second one created that causes an error.

Is it that it's impossible to create 2 publishers with same message type with micro-ros?

Thanks!

Acuadros95, 2022-04-21:
Can you create 3 publishers with different types?

I think your board is using the [colcon_verylowmem.meta](https://github.com/micro-ROS/micro_ros_arduino/blob/galactic/extras/library_generation/colcon_verylowmem.meta) meta file, which only allows 2 publishers.
Modify this file and rebuild the library, check the [How to build the precompiled library](https://github.com/micro-ROS/micro_ros_arduino#how-to-build-the-precompiled-library) readme section for more info.

SimonGodon, 2022-04-21:
Thanks,
I am using arduino nano rp2040 connect
I tried to change the maximum number of publishers to 3 in the [colcon_verylowmem.meta](https://github.com/micro-ROS/micro_ros_arduino/blob/galactic/extras/library_generation/colcon_verylowmem.meta) file, and then built the library, but now when I try to execute the code (which compiles correctly), the line
RCCHECK(rclc_support_init(&support, 0, NULL, &allocator));
not working and brings me to the error_loop
If I try to remove this line, then the next one:
RCCHECK(rclc_node_init_default(&node, "node_name", "", &support));
throws me in the error loop also

Is that because other things should be modified in the [colcon_verylowmem.meta](https://github.com/micro-ROS/micro_ros_arduino/blob/galactic/extras/library_generation/colcon_verylowmem.meta) file?

SimonGodon, 2022-04-21:
> Please:
> 
>     * use the issue template, do not delete it
> 
>     * share sample code in order to reproduce the issue
> 
>     * indicate which board are you using
> 
>     * provide steps for replicating

Ok, I edited my original post

pablogs9, 2022-04-21:
If `rclc_support_init` fails maybe your board is not being connected to the micro-ROS agent.

Can you share the output of the micro-ROS agent using the flag `-v6`?
Is your board correctly connected to the wifi?

SimonGodon, 2022-04-21:
You are right, the problem was that I started the node (arduino) before the agent (PC), now it is solved and I can have 3 publishers.
For people who would encounter the same error, the steps for solving it are:

-Change the maximum number of publishers to 3 in the [colcon_verylowmem.meta](https://github.com/micro-ROS/micro_ros_arduino/blob/galactic/extras/library_generation/colcon_verylowmem.meta) file
-Build the library ([How to build the precompiled library](https://github.com/micro-ROS/micro_ros_arduino#how-to-build-the-precompiled-library))
-Retry, do not forget to start the agent first.

Thanks @pablogs9  and @Acuadros95  for your help!

closed 2022-04-21 by pablogs9