micro-ROS/micro_ros_arduino #956: No service response

scch1092, 2022-04-21:
Hi, it's me again. 

I tried to create a service on my teensy. I used the sample program to create my service. Unfortunately I do not get any response to my request. In the terminal the service is displayed (ros2 service list -t).

- Hardware description: teensy41; imxrt1062/fpv5-d16-hard (PlatformIO)
- Installation type: 
  build_flags =
    -L ./.pio/libdeps/teensy41/micro_ros_arduino/src/imxrt1062/fpv5-d16-hard/
    -l microros
    -D ARDUINO_TEENSY41
- Version or commit hash: 2.0.4-galactic+sha.3df8bba

#### Steps to reproduce the issue
1. Load the example for microROS server (micro-ros_addtwoints_service.ino)
2. Flash on device (in my case teensy 4.1)
3. Start miroROS agent
4. send request via terminal (  ros2 service call /addtwoints example_interfaces/srv/AddTwoInts '{a: 2, b: 3}'   ) 

#### Expected behavior
Get response

#### Actual behavior
terminal output:

   waiting for service to become available...
   requester: making request: example_interfaces.srv.AddTwoInts_Request(a=2, b=3)

nothing happens on terminal...

pablogs9, 2022-04-21:
Are you using ROS 2 galactic? If so, have you changed the default middleware to Fast DDS? micro-ROS won't work with Cyclone DDS (the default implementation in ROS 2 Galactic).

https://docs.ros.org/en/galactic/Installation/DDS-Implementations/Working-with-eProsima-Fast-DDS.html

scch1092, 2022-04-21:
I use ROS2 galactic. I followed the instructions to install and use fastDDS, thanks for the tip! But apparently I had already installed fastDDS. Nothing changed even after switching to rmw_fastrtps.

[image]

pablogs9, 2022-04-21:
Make sure that you use Fast DDS in the console where you are calling `ros2 service call...`.

As you can see the micro-ROS Agent on the micro-ROS side is working properly because the service request is being received in the agent side (see `====> DDS <====`) and after a while, the service response is being sent to the ROS 2 dataspace (see `<==== DDS ====>`)

scch1092, 2022-04-21:
oh wow, such a simple solution :smile: 
Thanks!

closed 2022-04-21 by scch1092

k7g03z, 2022-06-20:
Hi @pablogs9 
In your post on [April 21](https://github.com/micro-ROS/micro_ros_arduino/issues/956#issuecomment-1104963947), you wrote: 

> micro-ROS won't work with Cyclone DDS 

Can you elaborate on why this is the case?

k7g03z, 2022-07-14:
> Hi @pablogs9 In your post on [April 21](https://github.com/micro-ROS/micro_ros_arduino/issues/956#issuecomment-1104963947), you wrote:
> 
> > micro-ROS won't work with Cyclone DDS
> 
> Can you elaborate on why this is the case?

Hi @pablogs9, we have dug deep into this issue and believe this [issue](https://github.com/eclipse-cyclonedds/cyclonedds/issues/1328) describes the root cause.

A fix for FastDDS is also proposed [here](https://github.com/eProsima/Fast-DDS/pull/2837).

cc @sumanth-nirmal @miguelcompany