micro-ROS/micro_ros_arduino #964: Agent cannot Handle more than a specific amount of client connections

kalaajiahmad, 2022-04-22:
- Hardware description: ESP32 x 9
- Installation type: Micro_ros_arduino Library
- Version or commit hash: Foxy 2.0.4

#### Steps to reproduce the issue
By Connecting 9 Esps to one Agent and observing the Connection, some Esps might disconnect and re-connect randomly and it gets worse over time.
[image]

#### Expected behavior
Stable connection for a long period of time of multiple clients

#### Actual behavior
Un-stable and un-reliable connection.

#### Additional information
For now was able to have a more stable connection by fixing the client_key of each esp
using the following:
[image]

pablogs9, 2022-04-25:
Could you provide code to replicate this issue, please do not share code in the image, just paste here your micro-ROS application to allow us to replicate the use case.

pablogs9, 2022-05-17:
Any update on this?

pablogs9, 2022-06-14:
Closing due to inactivity

closed 2022-06-14 by pablogs9