micro-ROS/micro_ros_arduino #974: Symlinks to extra packages in the /extra_packages folder are skipped when building custom library via the docker container

safdark, 2022-05-04:
## Issue template

- Hardware description: N/A
- RTOS: N/A
- Installation type: N/A
- Version or commit hash: galactic

#### Steps to reproduce the issue
1. Clone this repository to ~/micro_ros_arduino

2. Create a ***symlink*** in the extras/library_generation/extra_packages folder, pointing to an external ```my_custom_msgs``` package
```
extra_packages/ $> ln -s <path-to-my-custom-msg> ./my_custom_msgs
```

3. Invoke the docker build
```
~/micro_ros_arduino/ $> sudo docker run -it --rm -v $(pwd):/project --env MICROROS_LIBRARY_FOLDER=extras microros/micro_ros_static_library_builder:galactic -p esp32
```

4. Query the generated static library for the ```my_custom_msg``` package, using the ```nm``` tool:
```
$> nm src/esp32/libmicroros.a | grep "my_custom_msgs"
```

#### Expected behavior

Some type information of messages/services in the ```my_custom_msgs``` should show up. Being able to symlink to a package avoids the extra step of having to *copy* over the files each time the external package is updated.

#### Actual behavior

None of those types have been included.

#### Additional information

If the same steps are performed, but with the ```my_custom_msgs``` package being ***copied*** to extra_packages/ instead of creating a symlink, that package gets picked up into the custom build and the corresponding types will show up in the libmicroros.a file.

bjv-capra, 2022-05-04:
AFAIK, this is a docker-related issue, unfortunately, mounting symlinks doesn't quite work, you would have to mount both directories, your project, and where the symlink is, and they should have the same path. Otherwise, you could mount the packages directly to the same place. i.e.
docker run -it --rm \
     --env MICROROS_LIBRARY_FOLDER=extras \
     -v $(pwd):/project \
     -v path_to_my_msgs1:/project/my_msgs1 \
     microros/micro_ros_static_library_builder:galactic -p esp32

pablogs9, 2022-05-09:
Thanks for the explanation @bjv-capra.

If you want to keep synced your extra packages, please use git repos and [`extra_packages.repos`](https://github.com/micro-ROS/micro_ros_arduino/blob/galactic/extras/library_generation/extra_packages/extra_packages.repos) files.

closed 2022-05-09 by pablogs9