micro-ROS/micro_ros_arduino #977: subscription callback is ignored when message type is a string

RubenCarreira99, 2022-05-04:
## Specs

- Hardware description: esp32dev
- PlatformIO
- Installation type: micro_ros_arduino
- Version:  galactic

#### Steps to reproduce the issue
I'm trying to create a simple topic to send/receive strings. As you can see in the code below, when the esp32 receives a message it should publish another one in response. If I use `std_msgs__msg__Int32` message type, the subscription callback actually publishes the message. The same doesn't happen if I use `std_msgs__msg__String` or a custom interface message type that has `std_msgs__msg__String` attributes. 
When I flash the code to my esp32, the esp32 connects to the agent and the topic is created. If I create a subscriber through the terminal, I can see the messages of the `std_msgs__msg__String` type published from another terminal.
For debugging purposes I tried to publish a custom message inside the loop() function and it worked, so I'm assuming the problem is in the subscription callback. 

```
#include <Arduino.h>
#include "WiFi.h"

// microros ===================================================================
#include <micro_ros_arduino.h>

#include <stdio.h>
#include <rcl/rcl.h>
#include <rcl/error_handling.h>
#include <rclc/rclc.h>
#include <rclc/executor.h>
#include <std_msgs/msg/int32.h>
//#include <cenario4_interfaces/msg/service_handling.h>

#define RCCHECK(fn) { rcl_ret_t temp_rc = fn; if((temp_rc != RCL_RET_OK)){printf("Failed status on line %d: %d. Aborting.\n",__LINE__,(int)temp_rc); vTaskDelete(NULL);}}
#define RCSOFTCHECK(fn) { rcl_ret_t temp_rc = fn; if((temp_rc != RCL_RET_OK)){printf("Failed status on line %d: %d. Continuing.\n",__LINE__,(int)temp_rc);}}

rclc_executor_t executor;
rcl_subscription_t subscriber;
rcl_publisher_t publisher;
rclc_support_t support;
rcl_allocator_t allocator;
rcl_node_t node;
rcl_timer_t timer;
std_msgs__msg__Int32 sub_msg;
std_msgs__msg__Int32 publish_msg;

void subscription_callback(const void * msgin)
{
	//const std_msgs__msg__Int32 * msg = (const std_msgs__msg__Int32 *)msgin;
    
    publish_msg.data = 2;
    rcl_publish(&publisher, (const void*)&publish_msg, NULL);
}

void createEntities(){
    int agent_port = 8888;

    set_microros_wifi_transports("SSID", "PASSWORD", "192.168.X.X", agent_port);

    delay(200);

    allocator = rcl_get_default_allocator();

    rcl_init_options_t init_options = rcl_get_zero_initialized_init_options();
    rcl_init_options_init(&init_options, allocator);
    rcl_init_options_set_domain_id(&init_options, 30);

    RCCHECK(rclc_support_init_with_options(&support, 0, NULL, &init_options, &allocator));
    
    rclc_node_init_default(&node, "micro_ros_node", "", &support);
    
    RCCHECK(rclc_publisher_init_default(
        &publisher,
        &node,
        ROSIDL_GET_MSG_TYPE_SUPPORT(std_msgs, msg, Int32),
        "tasks_topic"));

    RCCHECK(rclc_subscription_init_default(
        &subscriber,
        &node,
        ROSIDL_GET_MSG_TYPE_SUPPORT(std_msgs, msg, Int32),
        "tasks_topic"));

    //create executor and add a subscriber to it
    RCCHECK(rclc_executor_init(&executor, &support.context, 1, &allocator));
    RCCHECK(rclc_executor_add_subscription(&executor, &subscriber, &sub_msg, &subscription_callback, ON_NEW_DATA));
}

void setup()
{
  createEntities();
}

void loop()
{
  RCCHECK(rclc_executor_spin_some(&executor, RCL_MS_TO_NS(10)));
}
```

#### Additional information
Custom message type used

```
string service_name
string action
string task_type
int64 id
```

tbcky, 2022-05-08:
I also met with difficulties, esp32 information on how to subscribe to the ros2

Acuadros95, 2022-05-09:
Could you provide the code with the custom type?

Your issue is probably related to string memory init, check this tutorial: [Handling messages memory in micro-ROS](https://micro.ros.org/docs/tutorials/advanced/handling_type_memory/)

RubenCarreira99, 2022-05-09:
> Could you provide the code with the custom type?
> 
> Your issue is probably related to string memory init, check this tutorial: [Handling messages memory in micro-ROS](https://micro.ros.org/docs/tutorials/advanced/handling_type_memory/)

Thank you for your tip, but after looking into that link I still couldn´t find the issue here. The code below has the custom type as you requested. (The way I assign pub_msg it's values probably isn't the best. I presume it should be done with the snprintf(), but when I tested the rcl_publish() in the loop() it worked fine, so I left it like that )

```
#include <Arduino.h>
#include "WiFi.h"

// microros ===================================================================
#include <micro_ros_arduino.h>

#include <stdio.h>
#include <rcl/rcl.h>
#include <rcl/error_handling.h>
#include <rclc/rclc.h>
#include <rclc/executor.h>
#include <cenario4_interfaces/msg/service_handling.h>

#define RCCHECK(fn) { rcl_ret_t temp_rc = fn; if((temp_rc != RCL_RET_OK)){printf("Failed status on line %d: %d. Aborting.\n",__LINE__,(int)temp_rc); vTaskDelete(NULL);}}
#define RCSOFTCHECK(fn) { rcl_ret_t temp_rc = fn; if((temp_rc != RCL_RET_OK)){printf("Failed status on line %d: %d. Continuing.\n",__LINE__,(int)temp_rc);}}

rclc_executor_t executor;
rcl_subscription_t subscriber;
rcl_publisher_t publisher;
rclc_support_t support;
rcl_allocator_t allocator;
rcl_node_t node;
rcl_timer_t timer;
cenario4_interfaces__msg__ServiceHandling sub_msg;
cenario4_interfaces__msg__ServiceHandling publish_msg;

void subscription_callback(const void * msgin)
{
    const cenario4_interfaces__msg__ServiceHandling * sub_msg = (const cenario4_interfaces__msg__ServiceHandling *)msgin;
    
    //booking process between task_supplier and task_performer
 
    if (strlen(sub_msg->action.data) < 1 || strcmp(sub_msg->action.data, "B")==0)
    {
        return;

    } else if (strcmp(sub_msg->action.data, "A")==0)
    {
        //snprintf(publish_msg.action, 1, "%s", book);
        publish_msg.action.data = "B";
        publish_msg.id = 1;
        publish_msg.service_name.data = "qwerty";
        publish_msg.task_type.data = "qwerty";
        rcl_publish(&publisher, (const void*)&publish_msg, NULL);

    } else if (strcmp(sub_msg->action.data, "S")==0)
    {
        printf("calls task_performer");
    }

}

void createEntities(){
    int agent_port = 8888; //port
 
    set_microros_wifi_transports("SSID", "PASSWORD", "X.X.X.X", agent_port);

    delay(200);

    allocator = rcl_get_default_allocator();

    rcl_init_options_t init_options = rcl_get_zero_initialized_init_options();
    rcl_init_options_init(&init_options, allocator);
    rcl_init_options_set_domain_id(&init_options, 30);

    // create support_options
    RCCHECK(rclc_support_init_with_options(&support, 0, NULL, &init_options, &allocator));
    
    //create node
    rclc_node_init_default(&node, "micro_ros_node", "", &support);

    RCCHECK(rclc_publisher_init_default(
        &publisher,
        &node,
        ROSIDL_GET_MSG_TYPE_SUPPORT(cenario4_interfaces, msg, ServiceHandling),
        "tasks_topic"));

    //create subscriber
    RCCHECK(rclc_subscription_init_default(
        &subscriber,
        &node,
        ROSIDL_GET_MSG_TYPE_SUPPORT(cenario4_interfaces, msg, ServiceHandling),
        "tasks_topic"));

    //create executor and add a subscriber to it
    RCCHECK(rclc_executor_init(&executor, &support.context, 1, &allocator));
    RCCHECK(rclc_executor_add_subscription(&executor, &subscriber, &sub_msg, &subscription_callback, ON_NEW_DATA));
}

void setup()
{
  createEntities();

}

void loop()
{
  RCCHECK(rclc_executor_spin_some(&executor, RCL_MS_TO_NS(10)));

}
```

Acuadros95, 2022-05-13:
You need to initialize `sub_msg` memory to store the received data.
Also, you need to set the `publish_msg` size and capacity values of each string (Capacity = size + 1).

Check the `Compound types in micro-ROS` section of the tutorial.

RubenCarreira99, 2022-05-13:
Thank you @Acuadros95, it's just like you said. I had to allocate memory to store the data of type string. 

https://micro.ros.org/docs/tutorials/advanced/handling_type_memory/  -- The 'compound types' chapter was super helpful for me.

This page has good documentation that I couldn't understand at first. As I'm still learning the whole microROS thing, I come across these kind of issues. I'm sorry if it seemed too basic.

Closing the issue. Once again, thank you @Acuadros95

closed 2022-05-13 by RubenCarreira99