micro-ROS/micro_ros_arduino #983: Message dropping for subscribers

ajaykumar666, 2022-05-10:
## Issue template

- Hardware description: STM32F412VGT custom board
- Version or commit hash: microROS Arduino foxy

I have 4 publishers for sensors and 2 subscribers for motors. When running the bot there are messages dropping for subscribers causing the bot to move irregularly. I tried different QOS profiles in qos_profile.h but things didn't improve. Is there any way to solve this?

pablogs9, 2022-05-10:
Please provide instructions for replicating this issue: procedure for building the application for STM32F412VGT (as it is not a micro-ROS for Arduino supported platform) code, ROS 2 types, configuration, meta files...

ajaykumar666, 2022-05-10:
the controller uses Arduino/libraries/micro_ros_arduino-foxy/src/cortex-m4/fpv4-sp-d16-hard precomplied library. all that is needed is to use STM32duino board manager in arduino.

`#include "config.h"
#include "BotState.h"
#include "Motors.h"
//#include "usSensor.h"
#include "Clutch.h"
#include "imuSensor.h"

#include <micro_ros_arduino.h>
#include <stdio.h>
#include <rcl/rcl.h>
#include <rcl/error_handling.h>
#include <rclc/rclc.h>
#include <rclc/executor.h>
#include <rmw_microros/rmw_microros.h>

#include <sensor_msgs/msg/range.h>
#include <std_msgs/msg/int16.h>
#include <std_msgs/msg/int32.h>
#include <std_msgs/msg/u_int8.h>
#include <std_msgs/msg/float32_multi_array.h>
#include <std_msgs/msg/u_int8_multi_array.h>

rcl_publisher_t imu;
std_msgs__msg__Float32MultiArray msgImu;

//rcl_publisher_t us;
//std_msgs__msg__UInt8MultiArray uscan;

rcl_publisher_t lwheel_encoder;
std_msgs__msg__Int32 lwheel_int;

rcl_publisher_t rwheel_encoder;
std_msgs__msg__Int32 rwheel_int;

//rcl_publisher_t clutch_state_fb;
//std_msgs__msg__UInt8 clutch_state_int;

rcl_subscription_t lmotor_cmd;
std_msgs__msg__Int16 ltspeed;

rcl_subscription_t rmotor_cmd;
std_msgs__msg__Int16 rtspeed;

//rcl_subscription_t clutch_state;
//std_msgs__msg__UInt8 cstate;

rclc_executor_t executor;
rclc_support_t support;
rcl_allocator_t allocator;
rcl_node_t node;
rcl_timer_t timer;
rcl_timer_t us_timer;
rcl_ret_t temp_rc;

#define RCCHECK(fn) {  temp_rc = fn; if((temp_rc != RCL_RET_OK)){error_loop();}}

int break_count = 0;

bool micro_ros_init_successful;

void error_loop() {
  while (1) {
    delay(100);
    Serial.println("ERROR!!!");
    Motors::breakMotors();
    HAL_NVIC_SystemReset();
  }
}
[... 171 lines cut ...]
  micro_ros_init_successful = false;
//    ct = millis();
}

void loop() {
  //  Serial.println("Loop");

  if (RMW_RET_OK == rmw_uros_ping_agent(50, 2))
  {
    //    Serial.println("Agent found");
    if (!micro_ros_init_successful) {
      create_entities();
    } else {
      //      Serial.println("Spinning");
      rclc_executor_spin_some(&executor, RCL_MS_TO_NS(2));
    }
  }
  else if (micro_ros_init_successful)
  {
    //    Serial.println("Agent lost");
    Motors::breakMotors();
    destroy_entities();
  }
  //  Serial.println(millis()-ct);
}`

ajaykumar666, 2022-05-10:
> Which are the rates of the subscriptions?

Imu and encoders are being subscribed at 50hz
While motor commands are being published at 20 hz on Ros side

pablogs9, 2022-05-11:
Could you check if this works as expected not pinging the agent, something like:

```c
void setup() {
...
      create_entities();
...
}

void loop() {
rclc_executor_spin_some(&executor, RCL_MS_TO_NS(2));
}
```

Another interesting check would be to create the entities in best effort. This is using `rclc_*_init_best_effort` instead of `rclc_*_init_default`.

Those two changes will reduce the number of messages on the wire (in this case a serial port), allowing the transfer of more data payloads. This is a 115200 bauds serial port, probably the loss of some packages is due to the bandwidth available.

Finally, if the two options above do not work, I would try reducing the rate of the publications and subscriptions to fit the bandwithd of the transport layer (aprox 115 kbps),

ajaykumar666, 2022-05-12:
As for the baudrate, I have already increased it to 460800 so as to create a bigger bandwidth. I'll try to reduce the frequency of publishing and rcl_init_best_effort.

ajaykumar666, 2022-05-16:
thanks for the help. it was fixed by changing some frequencies for publishers and subscribers.

closed 2022-05-16 by ajaykumar666