micro-ROS/micro_ros_arduino #984: can not create multiple publishers for the same node

roman-rs, 2022-05-11:
## Issue template

- Hardware description: arduino nano rp2040 connect
- RTOS: 
- Installation type: compiled from sources using 'colcon build' ROS2 is galactic branch
- Version or commit hash: galactic

#### Steps to reproduce the issue
create more than two publishers

#### Expected behavior
several publishers are created and sending data to micro ros agent

#### Actual behavior
program failed at the 3rd call to rclc_publisher_init_default() with generic error RCL_RET_ERROR

#### Additional information
At first I successfully created two publishers 'encoders' and 'ina' to publish data from motor encoders and data from ina260 module (amp, volt, watts).
Then I decided to add data from the on-board IMU sensor. created everything the same way as I did for ina, but program stopped at that call.

the code snippets below:

the macro I'm using to check return codes:
#define RCCHECK(fn) {               \
    rcl_ret_t temp_rc = fn;         \
    if ((temp_rc != RCL_RET_OK))    \
    {                               \
        error_loop();               \
    }                               \
}

...
some global vars:
rclc_executor_t executor;
rcl_subscription_t subscriber;
rcl_publisher_t enc_pub;
rcl_publisher_t ina_pub;
rcl_publisher_t imu_pub;
rclc_support_t support;
rcl_allocator_t allocator;
rcl_node_t node;

Adafruit_INA260                    ina260       = Adafruit_INA260();

std_msgs__msg__Float32MultiArray  *enc_msg, *ina_msg, *imu_msg;
...
at the setup() function I have the following initializations:
    Serial.begin(9600);

    while (!Serial) {
        delay(1000);
    }

    pinMode(LED_PIN, OUTPUT);
    digitalWrite(LED_PIN, HIGH);

    Serial.println("Starting");

    odom_timer = millis();

    
    if (!ina260.begin()) {
        Serial.println("Couldn't find INA260 chip");
        error_loop();
    }   

    if (!IMU.begin()) {
        Serial.println("Failed to initialize IMU!");
        error_loop();
    }
[... 91 lines cut ...]
[1652238712.715798] debug    | UDPv4AgentLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0x0292AA28, len: 84
[1652238712.715914] debug    | UDPv4AgentLinux.cpp | send_message             | [** <<UDP>> **]        | client_key: 0x0292AA28, len: 13
[1652238712.731681] debug    | UDPv4AgentLinux.cpp | send_message             | [** <<UDP>> **]        | client_key: 0x0292AA28, len: 13
[1652238712.736140] debug    | UDPv4AgentLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0x0292AA28, len: 13
[1652238712.736201] debug    | UDPv4AgentLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0x0292AA28, len: 24
[1652238712.736295] info     | ProxyClient.cpp    | create_publisher         | publisher created      | client_key: 0x0292AA28, publisher_id: 0x001(3), participant_id: 0x000(1)
[1652238712.736422] debug    | UDPv4AgentLinux.cpp | send_message             | [** <<UDP>> **]        | client_key: 0x0292AA28, len: 14
[1652238712.736465] debug    | UDPv4AgentLinux.cpp | send_message             | [** <<UDP>> **]        | client_key: 0x0292AA28, len: 13
[1652238712.795743] debug    | UDPv4AgentLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0x0292AA28, len: 13
[1652238712.822689] debug    | UDPv4AgentLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0x0292AA28, len: 13
[1652238712.822724] debug    | UDPv4AgentLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0x0292AA28, len: 36
[1652238712.823283] info     | ProxyClient.cpp    | create_datawriter        | datawriter created     | client_key: 0x0292AA28, datawriter_id: 0x001(5), publisher_id: 0x001(3)
[1652238712.823402] debug    | UDPv4AgentLinux.cpp | send_message             | [** <<UDP>> **]        | client_key: 0x0292AA28, len: 14
[1652238712.823470] debug    | UDPv4AgentLinux.cpp | send_message             | [** <<UDP>> **]        | client_key: 0x0292AA28, len: 13
[1652238712.853808] debug    | UDPv4AgentLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0x0292AA28, len: 13

I tried to debug with gdb-multiarch, but as soon as I reach the corresponding line in my code, 's' doesn't jump into the 
    RCCHECK(rclc_publisher_init_default(
                &imu_pub,
                &node,
                ROSIDL_GET_MSG_TYPE_SUPPORT(std_msgs, msg, Float32MultiArray),
                "imu"));
    delay(100);
function call. instead it immediately place me into error_loop(). although 'step' in gdb supposed to make step in.
sorry this is the first time I'm trying to use gdb to troubleshood arduino.

pablogs9, 2022-05-11:
RP2040 nano in micro-ROS for Arduino is prebuild using `colcon_verylowmem.meta` and as you can see [here](https://github.com/micro-ROS/micro_ros_arduino/blob/46454377f82cc21fa02ca56ead718f37f85d1779/extras/library_generation/colcon_verylowmem.meta#L44) it only has preconfigured slots for 2 publishers.

You need to [rebuild](https://github.com/micro-ROS/micro_ros_arduino#how-to-build-the-precompiled-library) the precompiled library, modifying the values to fit your requirements.

roman-rs, 2022-05-11:
> RP2040 nano in micro-ROS for Arduino is prebuild using `colcon_verylowmem.meta` and as you can see [...]

Thank you for the reply.

Do you see any potential issues/problems increasing these default values?
e.g. if I want to have 5 publishers? 15 publishers? and 5 subscriptions?

rp2040 seems to be powerful enough to handle such values? is it?
I'll give it a try for sure (recompiling micro-ros with new values) and update the thread.

pablogs9, 2022-05-11:
It highly depends on the amount of memory available in the MCU. Each entity consumes an amount of RAM and you won't be able to link de binary if the memory overflows.

roman-rs, 2022-05-12:
great, I modified the default values in colcon_verylowmem.meta and now 3 publishers are running and pushing data from microros.

Thanks again!