micro-ROS/micro_ros_arduino #1014: Have to disconnect/reconnect USB for micro_ros_agent to work

joncamp, 2022-05-24:
## Issue template

- Hardware description: Teensy 4.1, host system is Jetson Xavier running Jetpack 5.0 DP, with foxy running in docker container
- RTOS: Sample code from https://micro.ros.org/docs/tutorials/core/teensy_with_arduino/
- Installation type: Steps from https://micro.ros.org/docs/tutorials/core/teensy_with_arduino/
- Version or commit hash: micro-ros foxy

#### Steps to reproduce the issue
1. Follow steps in tutorial to flash teensy with sample code.
2. Follow steps in tutorial to setup micro-ROS in the docker container
3. Run `ros2 run micro_ros_agent micro_ros_agent serial --dev /dev/ttyACM0`

#### Expected behavior
Agent is able to connect to Teensy and run accordingly
```
[1653426265.766437] info     | TermiosAgentLinux.cpp | init                     | running...             | fd: 3
[1653426265.767956] info     | Root.cpp           | set_verbose_level        | logger setup           | verbose_level: 4
[1653427559.637736] info     | TermiosAgentLinux.cpp | fini                     | server stopped         | fd: 3
[1653427559.644223] error    | TermiosAgentLinux.cpp | init                     | open device error      | device: /dev/ttyACM0, errno: 6
[1653427559.650021] error    | TermiosAgentLinux.cpp | init                     | open device error      | device: /dev/ttyACM0, errno: 6
[1653427560.155846] error    | TermiosAgentLinux.cpp | init                     | open device error      | device: /dev/ttyACM0, errno: 6
[1653427560.661702] error    | TermiosAgentLinux.cpp | init                     | open device error      | device: /dev/ttyACM0, errno: 6
[1653427561.166965] error    | TermiosAgentLinux.cpp | init                     | open device error      | device: /dev/ttyACM0, errno: 6
[1653427561.672720] error    | TermiosAgentLinux.cpp | init                     | open device error      | device: /dev/ttyACM0, errno: 6
[1653427562.179331] error    | TermiosAgentLinux.cpp | init                     | open device error      | device: /dev/ttyACM0, errno: 6
[1653427562.684482] error    | TermiosAgentLinux.cpp | init                     | open device error      | device: /dev/ttyACM0, errno: 6
[1653427563.185248] info     | TermiosAgentLinux.cpp | init                     | running...             | fd: 3
[1653427564.919978] info     | Root.cpp           | create_client            | create                 | client_key: 0x4A0AE1BE, session_id: 0x81
[1653427564.920292] info     | SessionManager.hpp | establish_session        | session established    | client_key: 0x4A0AE1BE, address: 0
[1653427564.971293] info     | ProxyClient.cpp    | create_participant       | participant created    | client_key: 0x4A0AE1BE, participant_id: 0x000(1)
[1653427564.973236] info     | ProxyClient.cpp    | create_topic             | topic created          | client_key: 0x4A0AE1BE, topic_id: 0x000(2), participant_id: 0x000(1)
[1653427564.974489] info     | ProxyClient.cpp    | create_publisher         | publisher created      | client_key: 0x4A0AE1BE, publisher_id: 0x000(3), participant_id: 0x000(1)
[1653427564.976416] info     | ProxyClient.cpp    | create_datawriter        | datawriter created     | client_key: 0x4A0AE1BE, datawriter_id: 0x000(5), publisher_id: 0x000(3)
```

#### Actual behavior
Agent waits forever
```
/ws/src# ros2 run micro_ros_agent micro_ros_agent serial --dev /dev/ttyACM0
[1653426265.766437] info     | TermiosAgentLinux.cpp | init                     | running...             | fd: 3
[1653426265.767956] info     | Root.cpp           | set_verbose_level        | logger setup           | verbose_level: 4
```

#### Additional information
Unplugging and replugging USB allows the Teensy to communicate properly. This is problematic because we need to be able to connect to the teensy after reboot, without having to unplug/replug.

pablogs9, 2022-05-25:
Hello @joncamp, indeed this is related to the issue you mention. I copy here my answer for reference:

> Hello, the client will try to connect to the agent in rclc_support_init() for a [certain amount of times](https://github.com/eProsima/Micro-XRCE-DDS-Client/blob/243b16b659f1b46e6678db21c30f246910a3b697/CMakeLists.txt#L57) with a [certain timeout](https://github.com/eProsima/Micro-XRCE-DDS-Client/blob/243b16b659f1b46e6678db21c30f246910a3b697/CMakeLists.txt#L58).

> If this attempt fails, in this example, the app will fail. You can check the return value of this function and retry until the agent is reached. You can [ping the agent](https://github.com/micro-ROS/micro_ros_arduino/blob/645c73e5e2929a28b16ed5192e31a9e1483bb2db/examples/micro-ros_reconnection_example/micro-ros_reconnection_example.ino#L104) in order to wait until it is available. But in general, you should handle the client-agent connection.

> In your use case, if you open the agent and later the client, the client might lose all the opportunities to connect. If you open the agent and then the client, it will succeed.

In the same way you need to implement client's logic to check the agent availability and run `rclc_support_init()`. In the updated example [here](https://github.com/micro-ROS/micro_ros_arduino/blob/galactic/examples/micro-ros_reconnection_example/micro-ros_reconnection_example.ino#L104) you can see an example implementation using a state machine.

Let us know if you solve your issue.

joncamp, 2022-06-03:
Yes - the updated sample seems to resolve the issue. Immediately after plugging in the proxyclient seems to pick up the teensy as expected. Thanks!

```
# ros2 run micro_ros_agent micro_ros_agent serial --dev /dev/ttyACM0
[1654278742.124495] info     | TermiosAgentLinux.cpp | init                     | running...             | fd: 3
[1654278742.125380] info     | Root.cpp           | set_verbose_level        | logger setup           | verbose_level: 4
[1654278742.140311] info     | Root.cpp           | create_client            | create                 | client_key: 0x40BE7F97, session_id: 0x81
[1654278742.140764] info     | SessionManager.hpp | establish_session        | session established    | client_key: 0x40BE7F97, address: 0
[1654278742.188601] info     | ProxyClient.cpp    | create_participant       | participant created    | client_key: 0x40BE7F97, participant_id: 0x000(1)
[1654278742.190297] info     | ProxyClient.cpp    | create_topic             | topic created          | client_key: 0x40BE7F97, topic_id: 0x000(2), participant_id: 0x000(1)
[1654278742.191348] info     | ProxyClient.cpp    | create_publisher         | publisher created      | client_key: 0x40BE7F97, publisher_id: 0x000(3), participant_id: 0x000(1)
[1654278742.192796] info     | ProxyClient.cpp    | create_datawriter        | datawriter created     | client_key: 0x40BE7F97, datawriter_id: 0x000(5), publisher_id: 0x000(3)
```

closed 2022-06-03 by joncamp