micro-ROS/micro_ros_arduino #1062: How can i use Float32 Multi Array in publisher (Micro_ros_teensy)

KiBots, 2022-06-30:
## Issue template

- Hardware description:  teensy4.1
- RTOS: Micro_ros_arduino for teensy4.1
- Version or commit hash: galactic on Ubuntu 20.04

#### Steps to reproduce the issue
#include <AS5045.h>
#include <Servo.h>
#include <micro_ros_arduino.h>

#include <stdio.h>
#include <rcl/rcl.h>
#include <rcl/error_handling.h>
#include <rclc/rclc.h>
#include <rclc/executor.h>

#include <std_msgs/msg/int32.h>
#include <std_msgs/msg/float32_multi_array.h>

rcl_publisher_t publisher;
rcl_publisher_t publisher2;
rcl_subscription_t subscriber;
rcl_subscription_t subscriber2;
std_msgs__msg__Int32 msg4;
std_msgs__msg__Int32 msg3;
std_msgs__msg__Float32MultiArray msg2;
std_msgs__msg__Int32 msg0;
std_msgs__msg__Int32 msg;
rclc_executor_t executor;
rclc_executor_t executor2;
rclc_support_t support;
rcl_allocator_t allocator;
rcl_node_t node;
rcl_timer_t timer;

#define LED_PIN 13

#define RCCHECK(fn) { rcl_ret_t temp_rc = fn; if((temp_rc != RCL_RET_OK)){error_loop();}}
#define RCSOFTCHECK(fn) { rcl_ret_t temp_rc = fn; if((temp_rc != RCL_RET_OK)){}}

#define CSpin   2
#define CLKpin  3
#define DOpin   4

Servo myservo;
Servo driver;
int ang = 1500;//1890-max_left//1590-middle//1110-right_max
int vel = 1545;

AS5045 enc (CSpin, CLKpin, DOpin) ;
int flag = 0, timet, flag_t = 1, flag_net = 0, flag_vel = 1;
byte period = 100, temp_odom;
float odom_p;
float odom;
int value_12bit = 360;
float wh_round_length = 0.23079;
float velocity;
long  start_time;
float start, end1;

void error_loop(){
  while(1){
    digitalWrite(LED_PIN, !digitalRead(LED_PIN));
    delay(100);
  }
}

void subscription_callback(const void * msgin)
{  
  const std_msgs__msg__Int32 * msg0 = (const std_msgs__msg__Int32 *)msgin;
}
void subscription_callback2(const void * msgin)
{  
  const std_msgs__msg__Int32 * msg4 = (const std_msgs__msg__Int32 *)msgin;
}

void setup() {
  set_microros_transports();
  
  pinMode(LED_PIN, OUTPUT);
  digitalWrite(LED_PIN, HIGH);  
  
  delay(2000);

  allocator = rcl_get_default_allocator();

  //create init_options
  RCCHECK(rclc_support_init(&support, 0, NULL, &allocator));

  // create node
  RCCHECK(rclc_node_init_default(&node, "micro_ros_arduino_node", "", &support));

  // create publisher
  RCCHECK(rclc_publisher_init_default(
    &publisher,
    &node,
    ROSIDL_GET_MSG_TYPE_SUPPORT(std_msgs, msg, Int32),
    "odom"));
  RCCHECK(rclc_publisher_init_default(
    &publisher2,
    &node,
    ROSIDL_GET_MSG_TYPE_SUPPORT(std_msgs, msg, Int32),
    "vova_pub"));

  // create subscriber
  RCCHECK(rclc_subscription_init_default(
    &subscriber,
    &node,
    ROSIDL_GET_MSG_TYPE_SUPPORT(std_msgs, msg, Int32),
    "Angle"));
    
  RCCHECK(rclc_subscription_init_default(
    &subscriber2,
    &node,
    ROSIDL_GET_MSG_TYPE_SUPPORT(std_msgs, msg, Int32),
    "Velocity"));

  // create executor
  RCCHECK(rclc_executor_init(&executor, &support.context, 1, &allocator));
  RCCHECK(rclc_executor_init(&executor2, &support.context, 1, &allocator));
  RCCHECK(rclc_executor_add_subscription(&executor, &subscriber, &msg0, &subscription_callback, ON_NEW_DATA));
  RCCHECK(rclc_executor_add_subscription(&executor2, &subscriber2, &msg4, &subscription_callback2, ON_NEW_DATA));
  msg2.data[0] = 0.1;
  msg2.data[1] = 0.1;
  msg3.data = 0;
  msg0.data=0;
  msg4.data=0;
  
  enc.begin ();
  myservo.attach(9);
  driver.attach(11);
}

void loop() {
  //delay(100);
  start = enc.read () * 0.08789;
  delay(period);

  end1 = enc.read () * 0.08789;
  if (start > end1)
  {
    if (abs(start - end1) >= 16)
    {
      temp_odom = (end1 + value_12bit) - start;
    }
  }
  else if (start < end1)
  {
    if (abs(start - end1) >= 16)
    {
      temp_odom = end1 - start;
    }
  }

  odom += temp_odom / 1531.91489;
  velocity = float(temp_odom * 0.0063672) ;
  //Serial.println(velocity);
  vel = 1506;
  temp_odom = 0;
  
  //######______SUBSCRIBER__________###########
  RCCHECK(rclc_executor_spin_some(&executor, RCL_MS_TO_NS(100)));
  RCCHECK(rclc_executor_spin_some(&executor2, RCL_MS_TO_NS(100)));
  /////////////////////////////////////////////

  //######______PUBLISHER__________###########
  RCSOFTCHECK(rcl_publish(&publisher, &msg2, NULL));
  //RCSOFTCHECK(rcl_publish(&publisher2, &msg3, NULL));
  ////////////////////////////////////////////

  msg2.data[0] = velocity;
  msg2.data[1] = odom;
  
  if (vel<=1504 and velocity<0.05)
  {
    vel=1000;
  }
  driver.writeMicroseconds(vel);
  myservo.writeMicroseconds(ang);
}
#### Expected behavior
Publish array like this ([1.123, 4.567])
#### Actual behavior
[image]

#### Additional information

Acuadros95, 2022-06-30:
Try with:

```c
  msg2.data.data[0] = 0.1;
  msg2.data.data[1] = 0.1;
```

KiBots, 2022-07-01:
It's working. Thanks

closed 2022-07-01 by KiBots

Genozen, 2022-07-25:
> Try with:
> 
> ```c
>   msg2.data.data[0] = 0.1;
>   msg2.data.data[1] = 0.1;
> ```

Hello, I tried following this `msg2.data.data[0] = 0.1;`
but it keeps giving me wifi:Association refused temporarily, comback time 299 mSec...
Any idea on this?