micro-ROS/micro_ros_arduino #1065: Teensy 4.1 with Ethernet does not work

adityakamath, 2022-07-03:
I'm trying to run the micro-ros_publisher_ethernet example on the Teensy 4.1 but it does not work. I'm able to compile and upload the code, but the Teensy just does not connect and eventually times out. I've tried it with the Teensy Ethernet Kit and a separate breakout board with ethernet, but neither of them are able to connect to the agent running on a Raspberry Pi. In comparison, the same example works perfectly fine on the Arduino Portenta with its breakout board. I also have no problems using the serial transport with the Teensy. 

I'm making this issue because I cannot find relevant documentation anywhere, or any working examples with the Teensy and Ethernet. Has anyone tried the ethernet example with the Teensy? Or faced this issue?

- Hardware description: Teensy 4.1
- Version or commit hash: galactic

Steps to reproduce the issue:
examples/micro-ros_publisher_ethernet with Teensy 4.1 and Teensy Ethernet Kit

pablogs9, 2022-07-04:
Ethernet transport for Teensy 4.1 was contributed and is not tested officially.

adityakamath, 2022-07-09:
@pablogs9 good to know, I thought I had an issue in my devices or code somewhere...

I'm not sure when I can spend time on this, but I'll leave this issue open for now.

pablogs9, 2022-07-11:
Does #1072 solve this?

adityakamath, 2022-07-11:
No it doesn't, but I figured out the issue... Unlike the Arduino Portenta,  Teensy 4.1 has a pre-defined mac address which can be retrieved with this function:

```
void get_teensy_mac(uint8_t *mac) {
    for(uint8_t by=0; by<2; by++) mac[by]=(HW_OCOTP_MAC1 >> ((1-by)*8)) & 0xFF;
    for(uint8_t by=0; by<4; by++) mac[by+2]=(HW_OCOTP_MAC0 >> ((3-by)*8)) & 0xFF;
}
```

Then from within the setup function:

```
uint8_t mac[6];
get_teensy_mac(mac)
byte teensy_mac[] = {mac[0], mac[1], mac[2], mac[3], mac[4], mac[5]};
...
set_microros_native_ethernet_udp_transports(teensy_mac, teensy_ip, agent_ip, port);
```

This does the trick and Ethernet works. So does the Reconnection example mentioned in #1072, but it does not work for the Portenta, so it might only be an Arduino issue. 

Note: on the ethernet kit, only the green LED turns on, but not the orange LED. But seems like this is a known issue that other users have also faced. I am not sure of the root cause of this, or the fix, if there is any. But it works. 

I will make a PR with these changes in the Ethernet publisher example. Should it also be documented somewhere, the readme perhaps?

closed 2022-07-11 by adityakamath